micro-ROS/micro_ros_setup #530: Static library creation: Wrong folder structure in firmware/build/include

dr-mh, 2022-05-29:
## Issue template

- Hardware description:
- RTOS:
- Installation type: Custom static micro-ROS library
- Version or commit hash: humble

#### Steps to reproduce the issue
Follow the steps described in https://micro.ros.org/docs/tutorials/advanced/create_custom_static_library/

#### Expected behavior
After the building process the folder tree in firmware/build/include should look like:
├── rcl
        |── allocator.h
       ├── arguments.h
       ├── client.h
       ├── context.h
       ├── domain_id.h

#### Actual behavior
The folder tree in firmware/build/include has the wrong structure with an additional sublevel in many subfolders, for example the rcl-path looks like:
├── rcl
│   └── rcl
│       ├── allocator.h
│       ├── arguments.h
│       ├── client.h
│       ├── context.h
│       ├── domain_id.h

The compilation process of a program which includes the generated static library fails, because a lot of files in the include path assumes the first mentioned folder structure.

#### Additional information

pablogs9, 2022-05-30:
This is expected behavior in ROS 2 and micro-ROS as you can see here: https://github.com/ros2/ros2/issues/1150

As far as some platforms are easier just to include one folder and have all the headers inside, what we do is iterate the folder structure and copy "back" folders that are repeated, for example, you can take a look here: https://github.com/micro-ROS/micro_ros_arduino/blob/1c9c40a85986cf16aceb899a6d2ebebfe6620e87/extras/library_generation/library_generation.sh#L233-L242

Could you check if this patch works for you: https://github.com/micro-ROS/micro_ros_setup/pull/531 ?

dr-mh, 2022-05-30:
Thank you for your comment and your help.
Based on your link, I executed the following script in the microros workspace to fix this:

`#!/bin/bash`

`LIBRARY_PATH=$(pwd)/firmware/build`

`######## Fix include paths  ########`
`pushd firmware/mcu_ws > /dev/null`
`    INCLUDE_ROS2_PACKAGES=$(colcon list | awk '{print $1}' | awk -v d=" " '{s=(NR==1?s:s d)$0}END{print s}')`
`popd > /dev/null`

`for var in ${INCLUDE_ROS2_PACKAGES}; do`
`    if [ -d "$LIBRARY_PATH/include/${var}/${var}" ]; then`
`        rsync -r $LIBRARY_PATH/include/${var}/${var}/* $LIBRARY_PATH/include/${var}`
`        rm -rf $LIBRARY_PATH/include/${var}/${var}`
`    fi`
`done`

And now it works, as in the previous microros versions.

closed 2022-05-30 by dr-mh