micro-ROS/micro_ros_setup #533: High CPU load when subscriber is added

gseqBE, 2022-06-08:
## Issue template

- Hardware description: STM32H743zi Nucleo board, over udp
- RTOS: Zephyr
- Installation type: micro_ros_setop. Docker containers
- Version or commit hash:

#### Steps to reproduce the issue

1.  Attached code has idle loop calculation. It has a publisher(/counter) at 100Hz and a bool subscriber /sensors/led
2. If only publisher is enabled,(remove rclc_executor_add_subscription() and reduce executor by 1),the idle and uros task output is below:
       ```
       IDLE load 91.2%
       UROS load 10.5%
       ``` 
   This seems close to correct as the CPU spends most of its time in the idle loop and the UROS load is at 10%
  
3. if publisher and subscriber is enabled, the idle and uros task output is below:
       ```
       IDLE load 11.9%
       UROS load 89.7%
       ``` 

This seems off. The CPU spends ~90% of its time in the uros task and ~10% in idle.
5. uros agent on host is active and running and a connection to the board has been made.

#### Expected behavior
When publisher and subscriber are enabled, the uros task load is unusually high. Should be close to 10% or a little more

#### Actual behavior
When publisher and subscriber are enabled, the uros task load is 90%

#### Additional information
Basic publishing and subscription works.

```
#include <zephyr.h>
#include <stdio.h>
#include <string.h>

#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <std_msgs/msg/int32.h>

#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <rmw_microros/rmw_microros.h>
#include <microros_transports.h>
#include <std_msgs/msg/bool.h>

#include <logging/log.h>

#define MICRO_ROS_APP_STACK 16000
#define MICRO_ROS_APP_TASK_PRIO 5

LOG_MODULE_REGISTER(main, LOG_LEVEL_DBG);

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printk("Failed status on line %d: %d. Aborting.\n",__LINE__,(int)temp_rc);}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printk("Failed status on line %d: %d. Continuing.\n",__LINE__,(int)temp_rc);}}

rcl_publisher_t publisher;
rcl_subscription_t led_subscription;
std_msgs__msg__Int32 msg;
std_msgs__msg__Bool led_msg;
void uros_transport_task(void);

K_THREAD_DEFINE(uros_id, MICRO_ROS_APP_STACK, uros_transport_task, NULL, NULL, NULL,
		MICRO_ROS_APP_TASK_PRIO, 0, 0);

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{	
	RCLC_UNUSED(last_call_time);
	if (timer != NULL) {
		RCSOFTCHECK(rcl_publish(&publisher, &msg, NULL));
		msg.data++;
	}
}

void led_subscription_callback(const void * msgin)
{
	static int32_t data = 1;
	const std_msgs__msg__Bool * msg = (const std_msgs__msg__Bool *)msgin;
	printk("Received subscription data from Control %d", data++);
	
}

static k_tid_t idle_tid[CONFIG_MP_NUM_CPUS];

#define GET_IDLE_TID(i, tid) do {\
	if (strcmp(tname, "idle 0" STRINGIFY(i)) == 0) { \
		idle_tid[i] = tid; \
	} \
} while (0);

static void thread_cb(const struct k_thread *cthread, void *user_data)
{
	const char *tname = k_thread_name_get((struct k_thread *)cthread);

	UTIL_LISTIFY(CONFIG_MP_NUM_CPUS, GET_IDLE_TID, (k_tid_t)cthread)
}

static void timeout_cb(struct k_timer *timer)
{
	static uint64_t prev_idle_cycles;
	static uint64_t prev_urostask_cycles;
	static uint64_t total_cycles;
	k_thread_runtime_stats_t rt_stats_all;
	uint64_t idle_cycles = 0;
	int err = 0;

	for (int i = 0; i < CONFIG_MP_NUM_CPUS; i++) {
		k_thread_runtime_stats_t thread_stats;

		err = k_thread_runtime_stats_get(idle_tid[i], &thread_stats);
		if (err < 0) {
			return;
		}

		idle_cycles += thread_stats.execution_cycles;
	}

	k_thread_runtime_stats_t thread_stats_urostask;
	uint64_t urostask_cycles = 0;

	err = k_thread_runtime_stats_get(uros_id, &thread_stats_urostask);
	if (err < 0) {
		return;
	}

	urostask_cycles += thread_stats_urostask.execution_cycles;
	
	err = k_thread_runtime_stats_all_get(&rt_stats_all);
	if (err < 0) {
		return;
	}

	int idle_load = 1000 - (1000 * (idle_cycles - prev_idle_cycles) /
			(rt_stats_all.execution_cycles - total_cycles));

	prev_idle_cycles = idle_cycles;

	int urostask_load = 1000 - (1000 * (urostask_cycles - prev_urostask_cycles) /
			(rt_stats_all.execution_cycles - total_cycles));

	prev_urostask_cycles = urostask_cycles;

	total_cycles = rt_stats_all.execution_cycles;

	printk("IDLE load %d.%d%%\r\n", idle_load / 10, idle_load % 10);
	printk("UROS load %d.%d%%\r\n", urostask_load / 10, urostask_load % 10);
}

K_TIMER_DEFINE(timer, timeout_cb, NULL);

void main(void)
{	

	k_thread_foreach(thread_cb, NULL);

	k_timer_start(&timer, K_MSEC(1000), K_MSEC(1000));

	for (int i = 0; i < CONFIG_MP_NUM_CPUS; i++) {
		printk("\t%s\n", k_thread_name_get(idle_tid[i]));
	}
}

void uros_transport_task(void)
{
    // Set custom transports
	rmw_uros_set_custom_transport(
		MICRO_ROS_FRAMING_REQUIRED,
		(void *) &default_params,
		zephyr_transport_open,
		zephyr_transport_close,
		zephyr_transport_write,
		zephyr_transport_read
	);

	// Init micro-ROS
	rcl_allocator_t allocator = rcl_get_default_allocator();
	rclc_support_t support;

	// create init_options
	RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

	// create node
	rcl_node_t node;
	RCCHECK(rclc_node_init_default(&node, "ursos_transport_node", "", &support));

	// create publisher
	RCCHECK(rclc_publisher_init_default(&publisher, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32), "/counter"));

	// Creating subscriber
	RCCHECK(rclc_subscription_init_default(&led_subscription, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Bool), "/sensors/led"));

	// create timer, 10ms, 100Hz
	rcl_timer_t timer;
	const unsigned int timer_timeout_ms = 10;
	RCCHECK(rclc_timer_init_default(
		&timer,
		&support,
		RCL_MS_TO_NS(timer_timeout_ms),
		timer_callback));

	// create executor
	rclc_executor_t executor;
	RCCHECK(rclc_executor_init(&executor, &support.context, 2, &allocator));
	RCCHECK(rclc_executor_add_timer(&executor, &timer));
	RCCHECK(rclc_executor_add_subscription(&executor, &led_subscription, &led_msg, &led_subscription_callback, ON_NEW_DATA));

	while(1){
    	rclc_executor_spin_some(&executor, RCL_MS_TO_NS(1000));
	}

	RCCHECK(rcl_publisher_fini(&publisher, &node))
	RCCHECK(rcl_node_fini(&node))
}
```

pablogs9, 2022-06-09:
Well, you are in preemptive priority space and your task's main loop is not making any syscall, so your micro-ROS loop is blocking every <5 priority. I guess that IDLE task is priority `CONFIG_NUM_PREEMPT_PRIORITIES -1`.

You need to handle the execution load of your application, in this case, having a real-time executor that guaranteed the timeout operation if you want a 10% CPU load in the micro-ROS application (assuming this task is your higher priority task):

```c
while(1){
    	rclc_executor_spin_some(&executor, RCL_MS_TO_NS(10));
        // Sleep 90 ms, use Zephyr API. i.e. usleep(90000); 
}
```

cmorganBE, 2022-06-09:
@pablogs9 the rclc_executor_spin_some() isn't blocking with the given timeout, waiting for events to occur?

To the OPs point, even at 100Hz shouldn't most of the time be spent in rclc_executor_spin_some() waiting for data to process (idle waiting for the OS to resume execution when data arrives?)

pablogs9, 2022-06-10:
`rclc_executor_spin_some` will block until some data is available in the executor or the timeout happens.

You can't assume that `rclc_executor_spin_some` will call OS to idle waiting while data arrives. It depends on the transport implementation.

Have you tried the proposed main loop?

cmorganBE, 2022-06-10:
@pablogs9 I think your proposal will work but has the side effect of introducing latency into the system.

I'd imagine we'd want to look at rclc_executor_spin_some() and following that down until we find the interactions with the OS (or looking at the transport shim code directly, I'm not familiar with this system yet) and then seeing how to make that properly blocking. That would seem to give us both low latency and sleep during idle (pending what the OS is doing) no?

pablogs9, 2022-06-10:
> I'd imagine we'd want to look at rclc_executor_spin_some() and following that down until we find the interactions with the OS (or looking at the transport shim code directly, I'm not familiar with [...]

That's right, check here: https://github.com/micro-ROS/zephyr_apps/blob/5929a3a3ea4ca2859e7eb5a18a7b00c29552898f/microros_extensions/microros_transports.c#L89

cmorganBE, 2022-06-10:
@pablogs9  oh perfect, we can improve that to resolve this issue.

Is there any problem with blocking rclc_executor_spin_some()? The docs indicated that it could block until the timeout occurred (which is entirely fine) so my read was that the api promise was "may block until timeout".

pablogs9, 2022-06-10:
It may block until timeout if there is no event triggered in the executor (data on the sub, timer...). If an event occurs, it won't block the timeout time.

cmorganBE, 2022-06-10:
@pablogs9 I don't see anywhere where zephyr primitives are used to wait/sleep along the spin_some path. Is that something not supported/implemented for micro-ros due to limitations of some of the rtoses being supported? Like we know zephyr can do it, but there isn't any blocking/waitset stuff in the microros zephyr transport code for example.

So it seems like along the path once we get to rmw_wait() on the microros side there is no way for rmw_wait() to actually wait, so it runs and returns immediately?

And we appreciate the input here, we may end up helping to improve this once we understand it completely.

pablogs9, 2022-06-10:
The transport is the only point where micro-ROS relies on the OS to wait for incoming data. The middleware controls the timeout time and calls the transport until the timeout is reached or a message is received. So you should be able to tune the RTOS calls in the receive method.

cmorganBE, 2022-06-10:
So rmw_wait() is calling transport_read() and using transport_read() to throttle its wakeup? We had speculated that this was one way to avoid processing since if you've got data you need to handle it, whether or not its for your subscriber etc.

We still can't see why, when using the UDP transport and a subscriber subscribing on an event triggered once a second (the subscriber prints and we can see it only print once a second), that the wait wouldn't be sleeping via the blocking call at https://github.com/micro-ROS/zephyr_apps/blob/5929a3a3ea4ca2859e7eb5a18a7b00c29552898f/microros_extensions/microros_transports.c#L164

cmorganBE, 2022-06-10:
Maybe the poll() isn't properly sleeping? https://github.com/micro-ROS/zephyr_apps/blob/5929a3a3ea4ca2859e7eb5a18a7b00c29552898f/microros_extensions/microros_transports.c#L168

pablogs9, 2022-06-13:
Not sure about the poll, maybe is misbehaving. Have you reached any conclusion?

cmorganBE, 2022-06-13:
@pablogs9 I'll discuss with @gseqBE, I'm thinking we should put some prints around the rmw_wait() call and confirm its exiting, look at network traffic etc, basically confirm that the poll is or isn't waiting.

closed 2022-06-15 by pablogs9