micro-ROS/micro_ros_setup #537: Enabling the multithreading capability at usage of a custom build static library

dr-mh, 2022-07-07:
- Hardware description:
- RTOS: FreeRTOS
- Installation type: static library
- Version or commit hash: ROS2 humble

#### Steps to reproduce the issue
I want to enable the multithreading capability of micro-ROS with FreeRTOS when I use a custom build static library.
I added in the colcon.meta file from https://micro.ros.org/docs/tutorials/advanced/create_custom_static_library/ the following lines at "microxrcedds_client":
`"-DUCLIENT_PROFILE_MULTITHREAD=ON",`
`"-DUCLIENT_PROFILE_SHARED_MEMORY=ON",`
`"-DUCLIENT_PLATFORM_FREERTOS_PLUS_TCP=ON"`

#### Expected behavior
The compilation process finishes without errors and the multithreading capability is enabled.

#### Actual behavior
Without "-DUCLIENT_PLATFORM_FREERTOS_PLUS_TCP=ON" I get the error: `#error XRCE multithreading not supported for this platform.` If I include the option I get the error: `fatal error: FreeRTOS.h: No such file or directory`

#### Additional information
Do you have any suggestions how the example files "my_custom_toolchain.cmake" and "my_custom_colcon.meta" from [https://micro.ros.org/docs/tutorials/...](https://micro.ros.org/docs/tutorials/advanced/create_custom_static_library/) should be extended to solve that issue? Should I include FreeRTOS into the build process? Thank you very much for your help!

Acuadros95, 2022-07-07:
Hi @dr-mh,

When using `UCLIENT_PLATFORM_FREERTOS_PLUS_TCP` and `UCLIENT_PROFILE_MULTITHREAD`, micro-ROS will use FreeRTOS Semaphores to handle multi threading: [detail](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/master/src/c/profile/multithread/multithread.c#L14).

So, Yes! You will need to include FreeRTOS on your build process.

dr-mh, 2022-07-09:
Thank you for comment. I included FreeRTOS into the build process and now it works.

closed 2022-07-09 by dr-mh

JohannKa, 2023-06-20:
> Thank you for answer. I included FreeRTOS into the build process and now it works.

Hi
Sorry to ask but how had you included FreeRTOS into the build process please ?
(I'm in the same case than you but don't find how to add FreeRTOS source in the build...)

markushehn, 2023-06-20:
Hi,
I used the FreeRTOS files from my application (alternatively you can download them) and included them in the .cmake-file with the following lines:
`include_directories(<path to FreeRTOS>/FreeRTOS/FreeRTOS-Kernel/include)`
`include_directories(<path to FreeRTOS>/FreeRTOS/FreeRTOS-Kernel/portable/GCC/ARM_CM4F)`
Replace <path to FreeRTOS> with the path of FreeRTOS in your setup. The path portable/GCC/ARM_CM4F depends on your target-CPU and could be different (in my case it is a Cortex-M4F-core).
You must also include the same FreeRTOSConfig.h header file in the build process for the static library as for your target application with FreeRTOS. If the config file would be a different one, it could come to errors.
I hope that helped.

JohannKa, 2023-06-20:
Thanks it helped a lot !
Now no complain about FreeRTOS anymore but microxrcedds_client fails with "XRCE multithreading not supported for this platform".
I added set(PLATFORM_NAME_FREERTOS ON) in toolchain.cmake and "-DPLATFORM_NAME_FREERTOS=ON" in colcon.meta but no luck so far...

JohannKa, 2023-06-21:
Wow I finally got it but this PLATFORM_NAME_FREERTOS is definitely tricky to maintain all along the toolchain...
I worked around adding it in extract_flags.py and I got rid of the "not supported" error.

Thanks a lot again @markushehn you anyway saved me a lot of time and I may submit an issue to the team with details about my specifics if any use.

markushehn, 2023-06-21:
I am glad that it works!
For the sake of completeness, here are the other things I added in the build files compared to the static-library example from https://micro.ros.org/docs/tutorials/advanced/create_custom_static_library/
In the .cmake-file I added the include_directories as described above and the FREERTOS-flag in the line (and a few other things like FPU usage):
`set(FLAGS "-O3 -DPLATFORM_NAME_FREERTOS -ffunction-sections -fdata-sections -fno-exceptions -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 --param max-inline-insns-single=500 -D'RCUTILS_LOG_MIN_SEVERITY=RCUTILS_LOG_MIN_SEVERITY_NONE'" CACHE STRING "" FORCE)`
The only changes I made in the colcon.meta file compared to the static library example is in the microxrcedds_client section:
`"microxrcedds_client": {`
`            "cmake-args": [`
`                "-DUCLIENT_PIC=OFF",`
`                "-DUCLIENT_PROFILE_UDP=OFF",`
`                "-DUCLIENT_PROFILE_TCP=OFF",`
`                "-DUCLIENT_PROFILE_DISCOVERY=OFF",`
`                "-DUCLIENT_PROFILE_SERIAL=OFF",`
`                "-DUCLIENT_PROFILE_STREAM_FRAMING=ON",`
`                "-DUCLIENT_PROFILE_CUSTOM_TRANSPORT=ON",`
`                "-DUCLIENT_PROFILE_MULTITHREAD=ON",`
`                "-DUCLIENT_PROFILE_SHARED_MEMORY=ON"`
`            ]`
`        },`

Sorokonog, 2023-11-15:
Hello everyone, can you advice me what to do with makefile for CubeIDE. I'v got such standard includes:

```
# C includes
C_INCLUDES =  \
-ICore/Inc \
-IDrivers/STM32G4xx_HAL_Driver/Inc \
-IDrivers/STM32G4xx_HAL_Driver/Inc/Legacy \
-IMiddlewares/Third_Party/FreeRTOS/Source/include \
-IMiddlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2 \
-IMiddlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F \
```

 But while static lib is compiling multithread.h gives same error:  fatal error: FreeRTOS.h: No such file or directory

Where to add the path to get rid of this?