micro-ROS/micro_ros_setup #538: run the command: ros2 run micro_ros_setup build_agent.sh failed

z2010012286, 2022-07-11:
## Issue template

- Hardware description:  vmware Ubuntu20.04 
- RTOS: arduino_esp32 
- Installation type:  micro_ros_setup  
- Version or commit hash:  version of micro-ROS used: foxy  

#### Steps to reproduce the issue

Sorry for my poor English!
I followed the install tutorial for the micro_ros setup. https://manzurmurshid.medium.com/how-to-connect-teensy-3-2-with-micro-ros-and-ros2-foxy-6c8f99c9b66a and https://github.com/micro-ROS/micro_ros_setup/tree/foxy
All commands were performed perfect until this command:  ros2 run micro_ros_setup build_agent.sh

The terminal output as below:
[image]

#### Expected behavior

#### Actual behavior

#### Additional information

pablogs9, 2022-07-11:
Could you provide instructions for replicating this issue?

Acuadros95, 2022-07-11:
Just tested it and on a fresh `ros:foxy` docker and its working, Are you using a `ros:foxy` docker or how did you install ROS?
Could you also share your output for `apt list --installed | grep fastrtps`?

z2010012286, 2022-07-11:
> 
I tested it on a fresh ros:foxy docker and it works! 
But I'm still curious why it is not work in my ubuntu20.04...
The output for _apt list --installed | grep fastrtps_ as below.

[image]