micro-ROS/micro_ros_setup #539: Thrown assertion at lost agent connection with a multithread enabled micro-ROS

dr-mh, 2022-07-11:
## Issue template

- Hardware description: STM32F446RE
- RTOS: FreeRTOS
- Installation type: static library
- Version or commit hash: humble

#### Steps to reproduce the issue
I enabled the multithread capability of micro-ROS and initialized a publisher with rclc_publisher_init_default.
The function rcl_publish is executed in a separate task with a rate of 50 Hz.
If I start the microcontroller and the micro-ROS agent, everything works.
Now, the micro-ROS agent is stopped and the function rcl_publish returns as expected an error.

#### Expected behavior
The function rcl_publish just returns an error.

#### Actual behavior
After a few calls of the function rcl_publish (which returns because of the lost connection with an error), the program on the microcontroller freezes with the assertions:
Assert failed in file FreeRTOS/FreeRTOS-Kernel/queue.c, line 1167
Assert failed in file FreeRTOS/FreeRTOS-Kernel/portable/GCC/ARM_CM4F/port.c, line 420

#### Additional information
It seems, that if I initialize the publisher with rclc_publisher_init_best_effort, the assertions are not thrown. Is this the solution of that problem? Is there a reason why it could work with rclc_publisher_init_best_effort instead of rclc_publisher_init_default?

pablogs9, 2022-07-12:
- Could you provide code for replicating this and details about your setup (colcon.meta, environment)?
- Are you using serial transport or network transport?
- Could you provide a more complete stack trace or the assert?
- Does this happens without the multithread enabled? Why do you need multiple threads if you are creating your entities first and publishing later in sequential order?

dr-mh, 2022-07-12:
Hi, thank you for your quick reply.
I found the error. There was a stack overflow in that task. At normal operation, the stack is large enough. If the communication to the agent is lost, it seems that more memory is allocated, maybe through the not sent telegrams and a growing queue (and therefore the thrown assertion in queue.c).
Regarding to your question why I am using multithreading. I have one task, where the micro-ROS executor is running. Another task is triggered by an interrupt through a binary semaphore. In that task, data is processed and then sent by rcl_publish. It seems to me that this is the most elegant way to implement the desired functionality by a multithread enabled micro-ROS. If there are another, better ways, please let me know.

closed 2022-07-12 by dr-mh