micro-ROS/micro_ros_setup #542: Unable to  generate firmware for renesas RA6M5

pyro001, 2022-07-15:
## Issue template

- Hardware description: 
- RTOS: 
- Installation type: 
- Version or commit hash: 

#### Steps to reproduce the issue
using the docker from 
root@testrenesas-NUC11PAHi5:/renesas_uros# 
on the docker image I ran: 
ros2 run micro_ros_setup create_firmware_ws.sh renesas_ra ra6m5

#### Expected behavior
I was expecting this to straight up work for the given command and generate the  files , I got an error  instead: 
#### Actual behavior
Non valid RTOS/Platform: renesas_ra/ra6m5
Available platforms:
. freertos
+-- crazyflie21
+-- esp32
+-- nucleo_f446re
+-- nucleo_f446ze
+-- nucleo_f746zg
+-- nucleo_f767zi
+-- olimex-stm32-e407
. generate_lib
+-- generic
. host
+-- generic
. mbed
+-- disco_l475vg_iot01a
. raspbian
+-- stretch_v8
+-- buster_v7
+-- buster_v8
. zephyr
+-- discovery_l475_iot1
+-- olimex-stm32-e407
+-- nucleo_f401re
+-- nucleo_h743zi
+-- nucleo_f746zg

Acuadros95, 2022-07-18:
Are you running the commands on a `ros:galactic` docker?

I just tested the following commands on a fresh docker and it works OK:

```bash
# Source the ROS 2 installation
source /opt/ros/$ROS_DISTRO/setup.bash

# Create a workspace and download the micro-ROS tools
mkdir microros_ws
cd microros_ws
git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup

# Update dependencies using rosdep
sudo apt update && rosdep update
rosdep install --from-paths src --ignore-src -y

# Install pip
sudo apt-get install python3-pip -y

# Build micro-ROS tools and source them
colcon build
source install/local_setup.bash

# Create step
ros2 run micro_ros_setup create_firmware_ws.sh renesas_ra ra6m5
```

Anyway for the Renesas we recommend to use the [micro-ROS component for Renesas e2 studio and RA6M5](https://github.com/micro-ROS/micro_ros_renesas2estudio_component), please check it out.

pyro001, 2022-07-20:
Sorry for the late reply, it seems that the issue was in my previous installation of linux . I had already reset my computer, this seems to work straight up now!

Acuadros95, 2022-07-20:
Great!

I am closing, feel free to open an issue with your UDP transport issues here: [micro-ROS for Renesas e2 studio](https://github.com/micro-ROS/micro_ros_renesas2estudio_component).