micro-ROS/micro_ros_setup #543: Installing the micro-ROS build system failed on ROS2-rolling

Kay-Anonymous, 2022-07-17:
## Issue template

- Hardware description: 
- RTOS: 
- Installation type: 
- Version or commit hash: 

#### Steps to reproduce the issue

Hello,

I was going through the tutorials as is given in,  [First micro-ROS Application on Linux](https://micro.ros.org/docs/tutorials/core/first_application_linux/). However, I got a few compile error look likes about conflicting of rmw_microxrcedds package. I was installing ROS2-rolling in Ubuntu-20.04 by source code. Please have a look and if possible, please tell me what am I doing wrong.

```
# Build micro-ROS tools and source them
colcon build
source install/local_setup.bash
```

#### Expected behavior
Run micro-ROS demo apps 

#### Actual behavior
Compile error when build micro_ros_setup

#### Additional information
Ubuntu Linux - Focal (20.04) 64-bit
Source code installation of ROS2 rolling version

cmd history:
```
   24  cd ws/
   25  mkdir micro_ros
   26  cd micro_ros/
   27  git clone -b foxy https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
   28  ll src/
   29  source /opt/ros2_rolling/install/setup.sh 
   30  rosdep update && rosdep install --from-paths src --ignore-src -y
   31  colcon build
   32  source install/local_setup.bash
   33  ros2 run micro_ros_setup create_firmware_ws.sh host
   34  ros2 run micro_ros_setup build_firmware.sh
   35  colcon build --packages-select rosidl_typesupport_microxrcedds_cpp
   36  ros2 run micro_ros_setup build_firmware.sh
   37  colcon build --packages-select rmw_microxrcedds
   38  ros2 run micro_ros_setup build_firmware.sh
   39  vcs --help
   40  vcs diff
   41  vcs branch
   42  vcs statrs
   43  vcs status
   44  rm -rf build/ install/ log/
   45  colcon build --symlink-install --allow-overriding action_msgs builtin_interfaces composition_interfaces example_interfaces geometry_msgs lifecycle_msgs nav_msgs rcl_interfaces rosgraph_msgs sensor_msgs sensor_msgs_py statistics_msgs std_msgs std_srvs test_msgs unique_identifier_msgs visualization_msgs
   46  cd /opt/ros2_rolling/src/
   47  vcs status
   48  cd -
   49  colcon build --symlink-install --allow-overriding action_msgs builtin_interfaces composition_interfaces example_interfaces geometry_msgs lifecycle_msgs nav_msgs rcl_interfaces rosgraph_msgs sensor_msgs sensor_msgs_py statistics_msgs std_msgs std_srvs test_msgs unique_identifier_msgs visualization_msgs
   50  history 
```

compile log:
```
[Processing: action_msgs, complex_msgs, lifecycle_msgs, rcl_interfaces, rmw_microxrcedds, rosgraph_msgs, statistics_msgs, std_msgs]                                                                      
--- stderr: rmw_microxrcedds                                                                                                                                                   
In file included from /home/eldk/ws/micro_ros/src/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/./rmw_microros_internal/utils.h:20,
                 from /home/eldk/ws/micro_ros/src/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_client.c:26:
/home/eldk/ws/micro_ros/src/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/./rmw_microros_internal/types.h:103:3: error: conflicting types for ‘rmw_context_impl_t’
  103 | } rmw_context_impl_t;
      |   ^~~~~~~~~~~~~~~~~~
In file included from /opt/ros2_rolling/install/rmw/include/rmw/rmw/types.h:31,
                 from /opt/ros2_rolling/install/rmw/include/rmw/rmw/event.h:28,
                 from /opt/ros2_rolling/install/rmw/include/rmw/rmw/rmw.h:102,
                 from /home/eldk/ws/micro_ros/src/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_client.c:23:
/opt/ros2_rolling/install/rmw/include/rmw/rmw/init.h:34:35: note: previous declaration of ‘rmw_context_impl_t’ was here
   34 | typedef struct rmw_context_impl_s rmw_context_impl_t;
      |                                   ^~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/rmw_microxrcedds.dir/build.make:89：CMakeFiles/rmw_microxrcedds.dir/src/rmw_client.c.o] 错误 1
make[2]: *** 正在等待未完成的任务....
In file included from /home/eldk/ws/micro_ros/src/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_get_gid_for_publisher.c:18:
/home/eldk/ws/micro_ros/src/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/./rmw_microros_internal/types.h:103:3: error: conflicting types for ‘rmw_context_impl_t’
  103 | } rmw_context_impl_t;
      |   ^~~~~~~~~~~~~~~~~~
In file included from /opt/ros2_rolling/install/rmw/include/rmw/rmw/types.h:31,
                 from /opt/ros2_rolling/install/rmw/include/rmw/rmw/event.h:28,
                 from /opt/ros2_rolling/install/rmw/include/rmw/rmw/rmw.h:102,
                 from /home/eldk/ws/micro_ros/src/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_get_gid_for_publisher.c:15:
/opt/ros2_rolling/install/rmw/include/rmw/rmw/init.h:34:35: note: previous declaration of ‘rmw_context_impl_t’ was here
   34 | typedef struct rmw_context_impl_s rmw_context_impl_t;
      |                                   ^~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/rmw_microxrcedds.dir/build.make:141：CMakeFiles/rmw_microxrcedds.dir/src/rmw_get_gid_for_publisher.c.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:139：CMakeFiles/rmw_microxrcedds.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....
make: *** [Makefile:141：all] 错误 2
---
Failed   <<< rmw_microxrcedds [31.4s, exited with code 2]
Aborted  <<< lifecycle_msgs [1min 16s]                                                                                                                                                     
Aborted  <<< complex_msgs [33.2s]                                                                                                                                                     
Aborted  <<< rosgraph_msgs [36.3s]                                                                                                                                                      
Aborted  <<< statistics_msgs [37.2s]                                                                                                                                                  
Aborted  <<< action_msgs [40.7s]                                                                                                                    
Aborted  <<< rcl_interfaces [53.1s]                                                                                 
Aborted  <<< std_msgs [1min 5s]                                          

Summary: 9 packages finished [1min 49s]
  1 package failed: rmw_microxrcedds
  7 packages aborted: action_msgs complex_msgs lifecycle_msgs rcl_interfaces rosgraph_msgs statistics_msgs std_msgs
  7 packages had stderr output: action_msgs complex_msgs rcl_interfaces rmw_microxrcedds rosgraph_msgs statistics_msgs std_msgs
  21 packages not processed
```

pablogs9, 2022-07-18:
You are using micro_ros_setup Foxy's branch:

```
   27  git clone -b foxy https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
```

just download `main` branch for using Rolling

closed 2022-07-18 by pablogs9

Kay-Anonymous, 2022-07-18:
Think you very much ,I'll try it right away.