micro-ROS/micro_ros_setup #546: The humble version compiled generate_lib takes up more memory than foxy.

chrisjaxy, 2022-07-19:
## Issue template

- Hardware description: microros_olimex_e407_extensions
- RTOS: freertos
- Installation type: generate_lib
- Version or commit hash: humble

I want to switch the version of microros from foxy to humble in my project, so I recompile the static library of humble. Then I found that the `.data` in the project was 6-7K more than before. Then I followed the steps to compile ping_pong --transport serial, the most basic project of olimex-stm32-e407, with 400B more `.data`. I compared the .map files and found that the hubmle version seems to have the following content.
[image]
In my project, more ros messages are used, resulting in 6-7K more `.data`. I want to understand this part compared to foxy, is it necessary in humble? Is there a way to reduce the memory footprint of the project in the humble version?

pablogs9, 2022-07-19:
In Humble, micro-ROS integrates `rosidl_typesupport_introspection_c` so types can be introspected. This adds some static memory in order to store type information for making it available at runtime.

If you need to remove this extra size, just remove the folder `rosidl/rosidl_typesupport_introspection_c` from your micro-ROS source tree and this should remove all the extra built objects.

If I'm not wrong, that should be enough. Please report if this solves your problem and reopen if you have any other questions.

closed 2022-07-19 by pablogs9

chrisjaxy, 2022-07-19:
> rosidl_typesupport_introspection_c

Should I delete the `mcu_ws/ros2/rosidl/rosidl_typesupport_introspection_c` folder? I found that this folder also exists in the foxy version, and the compilation will report an error after I delete it.
[image]
In addition, I found that the static library of the humble version contains 67 packages, while the foxy version only contains 61, which should be caused by additional packages rather than public packages.

pablogs9, 2022-07-19:
Remove the folder or add an empty `COLCON_IGNORE` inside the folder and then build the environment.

chrisjaxy, 2022-07-19:
> Remove the folder or add an empty `COLCON_IGNORE` inside the folder and then build the environment.

Same error as above. I compared the COLCON_IGNORE files in humble and foxy and made them consistent, still the same problem. Can you experiment to get an effective method and tell me?

pablogs9, 2022-07-19:
Could you provide steps for replicating?
Make sure that you clean the environment (remove build, install, log) after removing the folder

pablogs9, 2022-07-19:
Sorry, I forgot to mention that micro_ros_utilities depend on the introspection type support. Just add another `COLCON_IGNORE` to this package.