micro-ROS/micro_ros_setup #548: How to cross compile package micro_ros_demos_rclc for Android

Kay-Anonymous, 2022-07-23:
Hello, great guys.
    I'm doing a research job. Could you help me with the cross-compilation issues I encountered when adding a custom platform into micro-ROS?

Best wishes!

## Issue template

- Hardware description:  armeabi-v7a
- RTOS: 
- Installation type: micro_ros_setup + custom config
- Version or commit hash: ROS2:rolling + micro_ros_setup :main

#### Steps to reproduce the issue

using android-ndk-r21e to  cross compile micro_ros_demos_rclc Failed~

#### Expected behavior
want to integrate the micro-ROS client into the self-developed Android platform

#### Actual behavior
`ros2 run micro_ros_setup create_firmware_ws.sh` by custom configs , but run `ros2 run micro_ros_setup build_firmware.sh` failed at:
```
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [CMakeFiles/fibonacci_action_client.dir/build.make:116：fibonacci_action_client] 错误 1
make[4]: *** [CMakeFiles/Makefile2:78：CMakeFiles/fibonacci_action_client.dir/all] 错误 2
make[3]: *** [Makefile:141：all] 错误 2
make[2]: *** [CMakeFiles/fibonacci_action_client.dir/build.make:115：fibonacci_action_client/src/fibonacci_action_client-stamp/fibonacci_action_client-build] 错误 2
make[1]: *** [CMakeFiles/Makefile2:133：CMakeFiles/fibonacci_action_client.dir/all] 错误 2
/home/eldk/ws/microros/e1e_client/firmware/mcu_ws/install/lib/libexample_interfaces__rosidl_typesupport_c.a(fibonacci__type_support.cpp.o):fibonacci__type_support.cpp:example_interfaces::action::rosidl_typesupport_c::_Fibonacci_Goal_message_typesupport_data: error: undefined reference to 'rosidl_typesupport_introspection_c__get_message_type_support_handle__example_interfaces__action__Fibonacci_Goal'

(Omit some identical undefined errors)

/home/eldk/ws/microros/e1e_client/firmware/mcu_ws/install/lib/libaction_msgs__rosidl_typesupport_c.a(cancel_goal__type_support.cpp.o):cancel_goal__type_support.cpp:action_msgs::srv::rosidl_typesupport_c::_CancelGoal_service_typesupport_data: error: undefined reference to 'rosidl_typesupport_microxrcedds_c__get_service_type_support_handle__action_msgs__srv__CancelGoal'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [CMakeFiles/fibonacci_action_server.dir/build.make:116：fibonacci_action_server] 错误 1
make[4]: *** [CMakeFiles/Makefile2:78：CMakeFiles/fibonacci_action_server.dir/all] 错误 2
make[3]: *** [Makefile:141：all] 错误 2
make[2]: *** [CMakeFiles/fibonacci_action_server.dir/build.make:115：fibonacci_action_server/src/fibonacci_action_server-stamp/fibonacci_action_server-build] 错误 2
make[1]: *** [CMakeFiles/Makefile2:160：CMakeFiles/fibonacci_action_server.dir/all] 错误 2
make: *** [Makefile:144：all] 错误 2
---
Failed   <<< micro_ros_demos_rclc [2.91s, exited with code 2]
                                   
Summary: 40 packages finished [51.7s]
  1 package failed: micro_ros_demos_rclc
  7 packages had stderr output: micro_ros_demos_rclc microxrcedds_client rcl rcl_action rclc rcutils rmw_microxrcedds
[ros2run]: Process exited with failure 2
```

I was find the symbol in someone *.a:
```
eldk@eldk-u20-04:~/ws/microros/xxx_client/firmware/mcu_ws/install/lib
$ grep -r rosidl_typesupport_microxrcedds_c__get_service_type_support_handle__action_msgs__srv__CancelGoal
匹配到二进制文件 libaction_msgs__rosidl_typesupport_c.a
匹配到二进制文件 libaction_msgs__rosidl_typesupport_microxrcedds_c.a
eldk@eldk-u20-04:~/ws/microros/xxx_client/firmware/mcu_ws/install/lib
$ nm libaction_msgs__rosidl_typesupport_c.a | grep rosidl_typesupport_microxrcedds_c__get_service_type_support_handle__action_msgs__srv__CancelGoal
         U rosidl_typesupport_microxrcedds_c__get_service_type_support_handle__action_msgs__srv__CancelGoal
eldk@eldk-u20-04:~/ws/microros/xxx_client/firmware/mcu_ws/install/lib
$ nm libaction_msgs__rosidl_typesupport_microxrcedds_c.a | grep rosidl_typesupport_microxrcedds_c__get_service_type_support_handle__action_msgs__srv__CancelGoal
00000001 T rosidl_typesupport_microxrcedds_c__get_service_type_support_handle__action_msgs__srv__CancelGoal
```

It look likes have some mistakes when link library

#### Additional information

config of `custom.toolschain.cmake` :
```
set(CMAKE_CXX_STANDARD 17)

set(ARCH ARCH_ARM)
add_definitions(-DARCH_ARM=1)

set(ANDROID_STL c++_static)

add_definitions(-DANDROID)

# Not support Fortran
set(CMAKE_Fortran_COMPILER CACHE STRING "" FORCE)

## __ANDROID__ will be defined in source in this case
include(/opt/android-ndk-r21e/build/cmake/android.toolchain.cmake)

set(CMAKE_CXX_STANDARD_LIBRARIES_INIT "/opt/android-ndk-r21e/sources/cxx-stl/llvm-libc++/libs/armeabi-v7a/libc++_static.a")

find_program(CCACHE "ccache")
if(CCACHE)
    set(CMAKE_C_COMPILER_LAUNCHER ${CCACHE} CACHE STRING "" FORCE)
    set(CMAKE_CXX_COMPILER_LAUNCHER ${CCACHE} CACHE STRING "" FORCE)
endif(CCACHE)

set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -funwind-tables -fasynchronous-unwind-tables -g")
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -funwind-tables -fasynchronous-unwind-tables -g")
set(CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -Wl,--no-merge-exidx-entries")
```

config of `client_uros_packages.repos` :
```
repositories:
  eProsima/Micro-CDR:
    type: git
    url: https://github.com/eProsima/Micro-CDR.git
    version: foxy
  eProsima/Micro-XRCE-DDS-Client:
    type: git
    url: https://github.com/eProsima/Micro-XRCE-DDS-Client.git
    version: foxy

# MicroROS
  uros/rcl:
    type: git
    url: https://github.com/micro-ROS/rcl
    version: master
  uros/rclc:
    type: git
    url: https://github.com/ros2/rclc
    version: master
  uros/micro_ros_utilities:
    type: git
    url: https://github.com/micro-ROS/micro_ros_utilities
    version: main
  uros/rcutils:
    type: git
    url: https://github.com/micro-ROS/rcutils
    version: master
  uros/micro_ros_msgs:
    type: git
    url: https://github.com/micro-ROS/micro_ros_msgs.git
    version: main
  uros/rmw_microxrcedds:
    type: git
    url: https://github.com/micro-ROS/rmw-microxrcedds.git
    version: main
  uros/rosidl_typesupport:
    type: git
    url: https://github.com/micro-ROS/rosidl_typesupport.git
    version: master
  uros/rosidl_typesupport_microxrcedds:
    type: git
    url: https://github.com/micro-ROS/rosidl_typesupport_microxrcedds.git
    version: main
  uros/tracetools:
    type: git
    url: https://gitlab.com/micro-ROS/ros_tracing/ros2_tracing.git/
    version: master
  uros/micro-ROS-demos:
    type: git
    url: https://github.com/micro-ROS/micro-ROS-demos.git
    version: main

# Required messages packages

```

custom `build.sh` : 
```
. $PREFIX/config/utils.sh

if [ $# -ge 1 ]; then
    TOOLCHAIN=$1
else
    TOOLCHAIN=${PREFIX}/config/custom/custom/custom.toolchain.cmake
fi

if [ $# -ge 2 ]; then
    COLCON_META=$2
	  echo "Using provided meta: $COLCON_META"

else
    COLCON_META=$FW_TARGETDIR/mcu_ws/colcon.meta
	  echo "Using default meta: $COLCON_META"
fi

BUILD_DIR=$FW_TARGETDIR/build

pushd $FW_TARGETDIR/mcu_ws >/dev/null

# Set these variables according to your own environment.
if [ -z ${ANDROID_ABI+x} ]; then
    ANDROID_ABI=armeabi-v7a
fi
if [ -z ${ANDROID_NATIVE_API_LEVEL+x} ]; then
    ANDROID_NATIVE_API_LEVEL=android-21
fi
if [ -z ${ANDROID_NDK+x} ]; then
    ANDROID_NDK=/opt/android-ndk-r21e
fi

# rm -rf build install log

    colcon build --packages-up-to micro_ros_demos_rclc \
        --merge-install \
        --packages-ignore-regex=.*_cpp \
        --metas $COLCON_META \
        --cmake-args \
        "--no-warn-unused-cli" \
        -DCMAKE_POSITION_INDEPENDENT_CODE:BOOL=OFF \
        -DCMAKE_TOOLCHAIN_FILE=${TOOLCHAIN} \
        -DCMAKE_FIND_ROOT_PATH_MODE_PACKAGE=BOTH \
        -DANDROID_FUNCTION_LEVEL_LINKING=OFF \
        -DANDROID_NATIVE_API_LEVEL=${ANDROID_NATIVE_API_LEVEL} \
        -DANDROID_ABI=${ANDROID_ABI} \
        -DANDROID_NDK=${ANDROID_NDK} \
        -DANDROID=ON \
        -DTHIRDPARTY=ON \
		-DBUILD_SHARED_LIBS=OFF \
        -DBUILD_TESTING=OFF \
        -DBUILD_MEMORY_TOOLS=OFF \
        -DBUILD_MEMORY_TESTS=OFF \
        -DCMAKE_BUILD_TYPE=Release \
        "-DCMAKE_EXPORT_COMPILE_COMMANDS=ON" \
        -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON

popd >/dev/null

```

Acuadros95, 2022-07-26:
It looks like you are missing the `example_interfaces` messages, could you check if that repository is being included in your build?

```repo
  ros2/example_interfaces:
    type: git
    url: https://github.com/ros2/example_interfaces
    version: humble
```

You can also check if the `Fibonacci_Goal` symbols are being included in your `.a`.

Kay-Anonymous, 2022-07-26:
Hi, @Acuadros95 . 
`example_interfaces` has already been build.

There is compile error : 
```
/home/eldk/ws/microros/e1e_client/firmware/mcu_ws/install/lib/libexample_interfaces__rosidl_typesupport_c.a(add_two_ints__type_support.cpp.o):add_two_ints__type_support.cpp:example_interfaces::srv::rosidl_typesupport_c::_AddTwoInts_Response_message_typesupport_data: error: undefined reference to 'rosidl_typesupport_microxrcedds_c__get_message_type_support_handle__example_interfaces__srv__AddTwoInts_Response'
```

And then, I find symbols `rosidl_typesupport_microxrcedds_c__get_message_type_support_handle__example_interfaces__srv__AddTwoInts_Response` in `libexample_interfaces__rosidl_typesupport_microxrcedds_c.a` instead of `libexample_interfaces__rosidl_typesupport_c.a`

[image]

Is this ERROR because micro_ros_demos_rclc only linked `libexample_interfaces__rosidl_typesupport_c.a` and not linked `libexample_interfaces__rosidl_typesupport_microxrcedds_c.a` ?

Kay-Anonymous, 2022-07-27:
Hi gurus,
  I tried to run `ros2 run micro_ros_setup create_firmware_ws.sh android generic` with only modify branchs to `main/master`. There are also some ERRORs:

```
/home/eldk/ws/microros/android/firmware/mcu_ws/uros/rcl/rcl/src/rcl/graph.c:106: error: undefined reference to 'rmw_get_publisher_names_and_types_by_node'
/home/eldk/ws/microros/android/firmware/mcu_ws/uros/rcl/rcl/src/rcl/graph.c:148: error: undefined reference to 'rmw_get_subscriber_names_and_types_by_node'
/home/eldk/ws/microros/android/firmware/mcu_ws/uros/rcl/rcl/src/rcl/graph.c:189: error: undefined reference to 'rmw_get_service_names_and_types_by_node'
/home/eldk/ws/microros/android/firmware/mcu_ws/uros/rcl/rcl/src/rcl/graph.c:229: error: undefined reference to 'rmw_get_client_names_and_types_by_node'
/home/eldk/ws/microros/android/firmware/mcu_ws/uros/rcl/rcl/src/rcl/graph.c:257: error: undefined reference to 'rmw_get_topic_names_and_types'
/home/eldk/ws/microros/android/firmware/mcu_ws/uros/rcl/rcl/src/rcl/graph.c:283: error: undefined reference to 'rmw_get_service_names_and_types'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [CMakeFiles/rcl.dir/build.make:510：librcl.so] 错误 1
```

How should I do if I want run micro_ros_demos in Android ( ndk-r21e ) on main branch?

Help me please~

Kay-Anonymous, 2022-08-14:
anymore help?