micro-ROS/micro_ros_setup #549: Micro ros agent compilation failed

DhevanVasu, 2022-07-27:
vvvv:~/Micro_Ros/uros_ws$ ros2 run micro_ros_setup build_agent.sh
Building micro-ROS Agent
Starting >>> micro_ros_msgs
Finished <<< micro_ros_msgs [2.61s]                     
Starting >>> micro_ros_agent
--- stderr: micro_ros_agent                             
/usr/bin/ld: warning: libfastrtps.so.2, needed by /opt/ros/foxy/lib/librmw_fastrtps_shared_cpp.so, may conflict with libfastrtps.so.2.4.1
/usr/bin/ld: CMakeFiles/micro_ros_agent.dir/src/agent/graph_manager/graph_manager.cpp.o: undefined reference to symbol '_ZN8eprosima8fastrtps5types12ReturnCode_t10RETCODE_OKE'
/usr/bin/ld: /opt/ros/foxy/lib/libfastrtps.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/micro_ros_agent.dir/build.make:172: micro_ros_agent] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/micro_ros_agent.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< micro_ros_agent [1.40s, exited with code 2]
                                
Summary: 1 package finished [4.14s]
  1 package failed: micro_ros_agent
  1 package had stderr output: micro_ros_agent

ubuntu - 20.04
Ros fox\y
While building micro ros agent am getting this error. Already fastrtps is available by default ros foxy.
Please help me to resolve issue.

pablogs9, 2022-07-27:
Please provide details for replicating this issue

DhevanVasu, 2022-07-27:
Ros2 foxty got installed after that followed below steps:

**Building:**
source /opt/ros/$ROS_DISTRO/setup.bash
mkdir uros_ws && cd uros_ws
git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
rosdep update && rosdep install --from-paths src --ignore-src -y
colcon build
source install/local_setup.bash

Building micro-ROS-Agent
ros2 run micro_ros_setup create_agent_ws.sh
ros2 run micro_ros_setup build_agent.sh -- Here throwing the error

pablogs9, 2022-07-27:
I have replicated your steps in a clean Foxy environment and it works well: https://asciinema.org/a/CNQQ6XqoUO8mQXBAzB1QLtVZz

Could you please recheck your Foxy installation and try in a fresh one?

DhevanVasu, 2022-07-27:
I had tried fresh installation many times but there is no use every same time error only.

pablogs9, 2022-07-27:
I'm not sure if I have understood you well. Do you also have this error in a fresh environment? If so, could you provide instructions for replicating it in a clean ROS 2 Foxy Docker?

pablogs9, 2022-08-04:
Any update on this?

pablogs9, 2022-09-19:
Closing due to inactivity

closed 2022-09-19 by pablogs9