micro-ROS/micro_ros_setup #552: Some custom ros messages on microros cannot be received

chrisjaxy, 2022-07-28:
## Issue template

- Hardware description: STM32F407
- RTOS: freeRTOS
- Installation type: generate_lib
- Version or commit hash: lastest

#### Steps to reproduce the issue
I tested the latest version of the microros static library on foxy, and added several custom ros messages. I found that the custom ros messages with strings were unacceptable. As shown below:
```
std_msgs/Header header
uint8 priority                      
uint32 package_size                 
string app_key                     
string version                    
string platform                   
string upgrade_service_name     
string updating_info_service_name  
string upgrade_progress_topic_name  
```
But some messages can be successfully received. 
```
float32 progress
uint8 upgradestatus
```
I added some logs and found that cdr_deserialize failed. I used the static library from the 11-month version a long time ago and found that there is no such problem. 
[image]
I am sure that my subscriber is initialized correctly, because I can list this topic on ubuntu20.04 and the serial port can also receive data correctly. I want to know what is the reason?

Steps to reproduce: Add a custom ros message to the `firmware/mcu_ws/ros2` file, and then compile the static library. Compile the same custom ros message on ros2, then use `ros2 topic pub` to send the message test
#### Expected behavior
custom ros messages on microros are correctly received.
#### Actual behavior

#### Additional information

pablogs9, 2022-07-28:
Are you initializing memory in messages? https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/

chrisjaxy, 2022-07-28:
> Are you initializing memory in messages? https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/

For the string type, I allocated char data[100] bytes of memory by default, and when sending the ros message, I did not fill the content of the string, but filled the data in the uint32. And I use a version I stored in 11 months here, and it can enter the callback function normally. Now the strange thing is that it does not go into the callback, but fails on the function `custom_subscription->type_support_callbacks->cdr_deserialize(
    &temp_buffer,
    ros_message)`

pablogs9, 2022-07-28:
Could you please share replication code?

chrisjaxy, 2022-07-28:
The referenced template is shown in my previous issue.
```
rclc_support_t test_support;
rcl_node_t test_node;
rclc_executor_t test_executor;
rcl_allocator_t test_allocator;
rcl_publisher_t test_publisher;
rcl_subscription_t test_subscription;

std_msgs__msg__UInt8 pile_heartbeat;
chassis_interfaces__msg__FirmwareInfo firmwareinfo;
bool micro_ros_init_successful = false;
bool micro_ros_destory_flag = false;

char app_key_string[100];
char version_string[100];
char platform_string[100];
char upgrade_service_name_string[100];
char updating_info_service_name[100];
char upgrade_progress_topic_name_string[100];

void firmwareinfo_callback(const void * msgin)
{
	const chassis_interfaces__msg__FirmwareInfo * msg = (const chassis_interfaces__msg__FirmwareInfo *)msgin;

	LOG_DEBUG("get message");
}

bool create_entities()
{	
	test_allocator = rcl_get_default_allocator();

	// create init_options
	RCCHECK(rclc_support_init(&test_support, 0, NULL, &test_allocator));

	// create node
	RCCHECK(rclc_node_init_default(&test_node, "test", "", &test_support));

	// create publisher
	RCCHECK(rclc_publisher_init_default(
		&test_publisher,
		&test_node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, UInt8),
		"std_msgs_msg_Int8"));

	RCCHECK(rclc_subscription_init_best_effort(
		&test_subscription,
		&test_node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(chassis_interfaces, msg, FirmwareInfo),
		"/firmwareinfo"));

	// create executor
	test_executor = rclc_executor_get_zero_initialized_executor();
	RCCHECK(rclc_executor_init(&test_executor, &test_support.context, 1, &test_allocator));
	RCCHECK(rclc_executor_add_subscription(&test_executor, &test_subscription, &firmwareinfo, firmwareinfo_callback, ON_NEW_DATA));

	micro_ros_init_successful = true;

	return true;
}

void destroy_entities()
{
	rcl_subscription_fini(&test_subscription, &test_node);
	rcl_publisher_fini(&test_publisher, &test_node);
	rcl_node_fini(&test_node);
	rclc_executor_fini(&test_executor);
	rclc_support_fini(&test_support);

	micro_ros_init_successful = false;
}

void StartDefaultTask(void *argument)
{
	bool pingFlag = false;
	MicroRosChannelConfig();

	firmwareinfo.app_key.data     = 0
	firmwareinfo.app_key.capacity = 100;
	firmwareinfo.app_key.size     = app_key_string;

	firmwareinfo.version.data     = 0
	firmwareinfo.version.capacity = 100;
	firmwareinfo.version.size     = version_string;

	firmwareinfo.platform.data     = 0
	firmwareinfo.platform.capacity = 100;
	firmwareinfo.platform.size     = platform_string;

	firmwareinfo.upgrade_service_name.data     = 0
	firmwareinfo.upgrade_service_name.capacity = 100;
	firmwareinfo.upgrade_service_name.size     = upgrade_service_name_string;

	firmwareinfo.updating_info_service.data     = 0
	firmwareinfo.updating_info_service.capacity = 100;
	firmwareinfo.updating_info_service.size     = updating_info_service_name;

	firmwareinfo.upgrade_progress_topic_name.data     = 0
	firmwareinfo.upgrade_progress_topic_name.capacity = 100;
	firmwareinfo.upgrade_progress_topic_name.size     = upgrade_progress_topic_name_string;

	while (1)
	{
		osDelay(5);
        if ((micro_ros_init_successful == false) && !pingFlag)
		{
			if (rmw_uros_ping_agent(50, 1) != RMW_RET_OK)
		 	{
				LOG_DEBUG("ping error!");
				osDelay(200);
				continue;
			}

			LOG_DEBUG("ping ok!");
			pingFlag = true;
		}

		if (!micro_ros_init_successful) {
			LOG_DEBUG("start create_entities!");
			if (create_entities() == false) {
				micro_ros_destory_flag = true;
				LOG_DEBUG("create_entities fail, destroy_entities!");
			} else {
				LOG_DEBUG("create_entities finish");
			}
		} else {
			rclc_executor_spin_some(&test_executor, RCL_MS_TO_NS(1));
		}

		if ((micro_ros_init_successful == true) && (micro_ros_destory_flag == true)){
			LOG_DEBUG("destroy_entities!");
			destroy_entities();
			micro_ros_destory_flag = false;
		}
    }
}
```
[image]

pablogs9, 2022-07-28:
Recheck your initialization approach, it should be:

```c
	firmwareinfo.app_key.data     = app_key_string
	firmwareinfo.app_key.capacity = 100;
	firmwareinfo.app_key.size     = 0 ;
```

chrisjaxy, 2022-07-28:
> Recheck your initialization approach, it should be:
> 
> ```c
> 	firmwareinfo.app_key.data     = app_key_string
> 	firmwareinfo.app_key.capacity = 100;
> 	firmwareinfo.app_key.size     = 0 ;
> ```

Sorry, there was my problem when I organized the code here. But I tested it as follows.
```
	firmwareinfo.app_key.data     = app_key_string
	firmwareinfo.app_key.capacity = 100;
	firmwareinfo.app_key.size     = 0 ;
```

Acuadros95, 2022-07-28:
You also need to initialize the message `header` string member memory.

What are you sending to the board?

chrisjaxy, 2022-07-29:
> You also need to initialize the message `header.frame_id` string member memory:
> 
> ```c
> typedef struct std_msgs__msg__Header
> {
>   builtin_interfaces__msg__Time stamp;
>   rosidl_runtime_c__String frame_id;
> } std_msgs__msg__Header;
> ```
> 
> What are you sending to the board?

> You also need to initialize the message `header.frame_id` string member memory:
> 
> ```c
> typedef struct std_msgs__msg__Header
> {
>   builtin_interfaces__msg__Time stamp;
>   rosidl_runtime_c__String frame_id;
> } std_msgs__msg__Header;
> ```
> 
> What are you sending to the board?

After that I initialized memory for `std_msgs__msg__Header`, but it still doesn't work. After that, I changed it to a `sensor_msgs_msg_Range` format, which has the type of `std_msgs__msg__Header`, and then I can still receive it without initializing the memory.
This topic I sent is a template I tested before, and it doesn't matter if I keep it.

pablogs9, 2022-07-29:
> I can still receive it without initializing

You need to initialize all the message structures that you use in micro-ROS. Please take a deep read of the tutorial we linked above.

Can we close this issue or do you have further questions beyond messages memory initialization?

chrisjaxy, 2022-07-29:
> After that I initialized memory for `std_msgs__msg__Header`, but it still doesn't work. 

Even if I initialize the memory in the custom message, I still cannot receive it. And the received data cannot enter the callback.

pablogs9, 2022-07-29:
Provide your latest code, so we can check the whole application with your modifications

chrisjaxy, 2022-07-29:
> I  After that, I changed it to a `sensor_msgs_msg_Range` format, which has the type of `std_msgs__msg__Header`, and then I can still receive it without initializing the memory. This topic I sent is a template I tested before, and it doesn't matter if I keep it.

I did another test, using a standard message, `sensor_msgs_msg_Range` with `std_msgs__msg__Header`, `std_msgs__msg__Header` with `std_msgs__msg__String`, and then tried not to initialize the memory. I found that the received data can enter the callback. With a custom message, the function `custom_subscription->type_support_callbacks->cdr_deserialize( &temp_buffer, ros_message)` fails and the callback cannot be entered.

chrisjaxy, 2022-07-29:
> Provide your latest code, so we can check the whole application with your modifications

The code I added is as follows, very simple.
```
char header_string[100];

firmwareinfo.header.frame_id.data        = header_string;
firmwareinfo.header.frame_id.capacity    = 100;
firmwareinfo.header.frame_id.size        = 0;
```

pablogs9, 2022-07-29:
Sorry, I need the application code (at least a replicable part that I can build in my environment) to check the issue.

chrisjaxy, 2022-07-29:
> Sorry, I need the application code (at least a replicable part that I can build in my environment) to check the issue.

Sorry, the problem is on my side. We only restarted the MCU during the test and kept the agent online. Perhaps the process of modifying the type caused this problem.
At the same time, I found a very interesting situation, for my custom message,
```
std_msgs/Header header
uint8 priority                      
uint32 package_size                 
string app_key                     
string version                    
string platform                   
string upgrade_service_name     
string updating_info_service_name  
string upgrade_progress_topic_name  
```
I only need to initialize the memory of the last upgrade_progress_topic_name, and I can enter the callback without initializing other types. And the same is true for other custom types of messages with strings. And the microros version I used before this did not need to do this. 
Can you tell me which version modification or commit has affected this?

closed 2022-07-29 by chrisjaxy