micro-ROS/micro_ros_setup #558: micro-ROS agent fails sometimes if ROS2 application is opened or closed

dr-mh, 2022-08-05:
## Issue template

- Hardware description: STM32F446
- RTOS: FreeRTOS
- Installation type: static library
- Version or commit hash: humble

#### Steps to reproduce the issue
The following code runs on a STM32F446:
```
void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{
    (void) last_call_time;
    if(timer != NULL)
    {
        msg.data++;
        rmw_uros_sync_session(1000);
        
        if((microros_rc = rcl_publish(&publisher, &msg, NULL)) != RCL_RET_OK)
            printf("Error at publishing\r\n");
    }
}

void microros_main(void *pvParameters)
{
    printf("microros_app start\r\n");
    
    rcl_allocator_t allocator = rcl_get_default_allocator();
    rcl_ret_t rc;
    
    msg.data = 0;
    
    while(1)
    {
        vTaskDelay(500);
        microros_agent_connection_rc = rmw_uros_ping_agent(100,1);
        printf("ping return val: %d\r\n", microros_agent_connection_rc);
        
        if(microros_agent_connection_rc == RCL_RET_OK)
        {
            printf("ping worked\r\n");
            
            // create init_options
            if(rclc_support_init(&support, 0, NULL, &allocator) != RCL_RET_OK)
                printf("Error rclc_support_init\r\n");
            
            printf("support created\r\n");
            
            // create node
            if(rclc_node_init_default(&node, "stm32_int32_publisher_rclc", "", &support) != RCL_RET_OK)
                printf("Error in rclc_node_init_default\r\n");
            
            printf("node created\r\n");
            
            // create publisher
            if(rclc_publisher_init_default(&publisher, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32), "stm32_std_msgs_msg_Int32") != RCL_RET_OK)
                printf("Error in rclc_publisher_init_default\r\n");
            
            printf("publisher created\r\n");
            
            // create timer
            const unsigned int timer_timeout_ms = 5;
            if(rclc_timer_init_default(&timer, &support, RCL_MS_TO_NS(timer_timeout_ms), timer_callback) != RCL_RET_OK)
                printf("Error in rcl_timer_init_default\r\n");
            
            printf("timer created\r\n");
            
            // create executor
            executor = rclc_executor_get_zero_initialized_executor();
            if(rclc_executor_init(&executor, &support.context, 1, &allocator) != RCL_RET_OK)
                printf("Error in rclc_executor_init\r\n");
            
            if(rclc_executor_add_timer(&executor, &timer) != RCL_RET_OK)
                printf("Error in rclc_executor_add_timer\r\n");
            
            printf("executor created\r\n");
            
            rclc_executor_spin(&executor);
        }
    }
    
    printf("microros_app finished\r\n");
    vTaskDelete(NULL);
}
```

Now, start for example rqt or another ROS2 program which subscribes the topic.
Start the micro-ROS agent to establish a connection.
If everything is connected, kill rqt or the other program.

#### Expected behavior
Everything works properly.

#### Actual behavior
The agent fails sometimes with the error:
```
terminate called after throwing an instance of 'std::system_error'
  what():  Invalid argument
[ros2run]: Aborted
```

#### Additional information
I found out, that the error occures often if the application (i.e. rqt) is running, before the micro-ROS connection is established, and then closed. Furthermore, this error occurs only, if the function rmw_uros_ping_agent is executed in the microcontroller program. If the ping function is never executed, everything works.
I observed in a more complex application, that the micro-ROS agent fails sometimes, if a new ROS2-program is started or closed, which subscribes the topic. But this behavior is very random and therefore hard to investigate.
Do you have any ideas why the micro-ROS agent fails sometimes with the mentioned error? The micro-ROS application on the microcontroller works properly and stable.
A solution could be to run the micro-ROS agent in a loop on a shell, if the agent fails, it is restarted.

pablogs9, 2022-08-08:
Related to this: https://github.com/micro-ROS/micro-ROS-Agent/issues/157
We are working on it.

dr-mh, 2022-08-28:
I tried the updated code and now it seems that everything work! Thank you for the update!

closed 2022-08-28 by dr-mh