micro-ROS/micro_ros_setup #559: Custom CAN transport ignoring some messages in communication between STM and agent

Sorokonog, 2022-08-08:
## Issue template

- Hardware description: STM32F446RET6 custom board identical to 446 NUCLEO
- RTOS: FreeRTOS
- Installation type: As recomended
- Version or commit hash: humble
#### Steps to reproduce the issue
Im trying to implement custom transport for CAN (agent and stm part has been asseembled and works) and face an issue... Some of CAN frames ignoing by agent. Session established and some messaging starts, but no topic has been initialized. What is curious that in candump I can see messages with corret header "7E" which are not showing in debug log of agent.  Could anydoby take a look and make a suggestion of what im doing wrong. (Log of candump and agent debug will attach later).

#### Additional information
Code of agent:
`#include <uxr/agent/transport/custom/CustomAgent.hpp>
#include <uxr/agent/transport/endpoint/CustomEndPoint.hpp>

#include <poll.h>
#include <sys/socket.h>
#include <unistd.h>
#include <signal.h>
#include <netinet/in.h>
#include <arpa/inet.h>

#include <uxr/agent/transport/can/CanAgentLinux.hpp>
#include <uxr/agent/utils/Conversion.hpp>
#include <uxr/agent/logger/Logger.hpp>

#include <net/if.h>
#include <sys/ioctl.h>

#include <linux/can.h>
#include <linux/can/raw.h>

/**
 * This custom XRCE Agent example attempts to show how easy is for the user to define a custom
 * Micro XRCE-DDS Agent behaviour, in terms of transport initialization and closing, and also
 * regarding read and write operations.
 * For this simple case, an UDP socket is opened on port 8888. Additionally, some information
 * messages are being printed to demonstrate the custom behaviour.
 * As the endpoint is already defined, we are using the provided
 * `eprosima::uxr::IPv4EndPoint` by the library.
 * Other transport protocols might need to implement their own endpoint struct.
 */

int main(int argc, char** argv)
{
    eprosima::uxr::Middleware::Kind mw_kind(eprosima::uxr::Middleware::Kind::FASTDDS);
    
    struct pollfd poll_fd;
    struct sockaddr_can addr;
    struct ifreq ifr;

    /**
     * @brief Agent's initialization behaviour description.
     */
    eprosima::uxr::CustomAgent::InitFunction init_function = [&]() -> bool
    {
        //struct sockaddr_can addr;
        struct canfd_frame frame;
        struct can_frame rec_frame;

        bool rv = false;

        if ((poll_fd.fd = socket(PF_CAN, SOCK_RAW, CAN_RAW)) < 0) 
        {
		perror("socket creation: ");
		return false;
        }
        
        if (-1 != poll_fd.fd)
        {
            struct sockaddr_in address{};
[... 223 lines cut ...]
    while ((can_head != can_tail) && (bytes_recv < len)){
        buf[bytes_recv] = can_buffer[can_head];
        can_head = (can_head + 1) % CAN_DMA_BUFFER_SIZE;
        bytes_recv++;
    }
    return bytes_recv;
	}

void HAL_CAN_RxFifo0MsgPendingCallback(CAN_HandleTypeDef *hcan){

CAN_RxHeaderTypeDef RxHeader;
uint8_t data[8];

HAL_CAN_GetRxMessage(hcan, CAN_RX_FIFO0, &RxHeader, data);
	if (RxHeader.StdId == ROUTER_ID)
	{
		for(uint16_t i=0; i<RxHeader.DLC; i++)
			{
			can_buffer[can_tail] = data[i];
			can_tail++;
			}
	}
}

#endif //RMW_UXRCE_TRANSPORT_CUSTOM`

Sorokonog, 2022-08-11:
Closing issue. Got some unexpected behaviour in write function but managed to handle it. For anyone who will try the same transport correct code listed below. The code is not clear and should be brushed up. Also timeouts should be implemented on both sides but it works.

Agent side

```C
#include <uxr/agent/transport/custom/CustomAgent.hpp>
#include <uxr/agent/transport/endpoint/CustomEndPoint.hpp>

#include <poll.h>
#include <sys/socket.h>
#include <unistd.h>
#include <signal.h>
#include <netinet/in.h>
#include <arpa/inet.h>

#include <uxr/agent/transport/can/CanAgentLinux.hpp>
#include <uxr/agent/utils/Conversion.hpp>
#include <uxr/agent/logger/Logger.hpp>

#include <net/if.h>
#include <sys/ioctl.h>

#include <linux/can.h>
#include <linux/can/raw.h>

/**
 * This custom XRCE Agent example attempts to show how easy is for the user to define a custom
 * Micro XRCE-DDS Agent behaviour, in terms of transport initialization and closing, and also
 * regarding read and write operations.
 * For this simple case, an UDP socket is opened on port 8888. Additionally, some information
 * messages are being printed to demonstrate the custom behaviour.
 * As the endpoint is already defined, we are using the provided
 * `eprosima::uxr::IPv4EndPoint` by the library.
 * Other transport protocols might need to implement their own endpoint struct.
 */

int main(int argc, char** argv)
{
    eprosima::uxr::Middleware::Kind mw_kind(eprosima::uxr::Middleware::Kind::FASTDDS);
    
    struct pollfd poll_fd;
    struct sockaddr_can addr;
    struct ifreq ifr;

    /**
     * @brief Agent's initialization behaviour description.
     */
    eprosima::uxr::CustomAgent::InitFunction init_function = [&]() -> bool
    {
        //struct sockaddr_can addr;
        struct canfd_frame frame;
        struct can_frame rec_frame;

        bool rv = false;

        if ((poll_fd.fd = socket(PF_CAN, SOCK_RAW, CAN_RAW)) < 0) 
        {
		perror("socket creation: ");
		return false;
        }
        
        if (-1 != poll_fd.fd)
        {
            struct sockaddr_in address{};
            strcpy(ifr.ifr_name, "can0");
            ioctl(poll_fd.fd, SIOCGIFINDEX, &ifr);
            poll_fd.events = POLLIN;
            addr.can_family  = AF_CAN;
            addr.can_ifindex = ifr.ifr_ifindex;
            if (-1 != bind(poll_fd.fd,
[... 240 lines cut ...]
uint8_t data[8];

ret = HAL_CAN_GetRxMessage(hcan, CAN_RX_FIFO0, &RxHeader, data);

if (ret != HAL_OK)
{
	HAL_GPIO_WritePin(GPIOD, GPIO_PIN_2, 1); //TODO ERROR HANDLER
}

if (RxHeader.StdId == ROUTER_ID)
	{
		for(uint16_t i=0; i<RxHeader.DLC; i++)
			{
			if(can_tail == CAN_IT_BUFFER_SIZE)
			{
			    can_tail = 0;
			}
			can_buffer[can_tail] = data[i];
			can_tail++;
			}
	}
}

#endif //RMW_UXRCE_TRANSPORT_CUSTOM
```

Sorokonog, 2022-08-23:
Got some further difficulties... I've implemented timeouts in both agent's and client's sides and everithing work pretty well except reconnect with agent. The behaveour is as follows
1. The client - STM32 starts upfront. It starts to send session create packets and if agent is alive everithing goes well.
2. If client restarts during session everithing works aswell.
3. But if we restarts agent during session the client would keep sending a packet:
  can0  00F   [8]  7E 00 00 0D 00 81 00 00
  can0  00F   [8]  00 0B 01 05 00 EB 0B EE
  can0  00F   [4]  0B 80 5A 2A
which is not recognizable by client as session reconnect. Actually the agent is not responding for this packet at all even the agent reconize it as valid packet.
So the question is if client or agent has any kind procedure to restart session if it broken due to timeout?

Acuadros95, 2022-08-23:
Yes, check this approach: [micro-ros_reconnection_example](https://github.com/micro-ROS/micro_ros_arduino/blob/humble/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino)

The idea is to use the ping to check if the agent is reachable. If the connection is lost, entities are destroyed and created again once the agent is ready again.

Sorokonog, 2022-08-23:
Great thanks. Will follow the idea.

Sorokonog, 2022-08-23:
Hello again looks like rmw_uros_ping_agent(args) is not RTOS safe... Will try to write smthing similar on my own.

Acuadros95, 2022-08-24:
> Hello again looks like rmw_uros_ping_agent(args) is not RTOS safe

Could you elaborate on this?

Sorokonog, 2022-08-28:
Seems that it takes a lot of work to wrap ping into RTOS function. And since ping is just "empty" session i've made simple workaround. It counts unsuccess write attempts and if the number of such failed writes overwhelms a value i destroy and recreate entities. Also I added a watchdog which resets the STM then writes callback stops to be called.

closed 2022-09-07 by Sorokonog