micro-ROS/micro_ros_setup #562: Sphinx/docutils version requirement incompatible errors

robodrome, 2022-08-11:
## Issue template

- Hardware description: olimex-stm32-e407
- RTOS: zephyr
- Installation type: micro_ros_setup, docker ros foxy
- Version or commit hash: foxy

#### Steps to reproduce the issue
Follow the [First micro-ROS Application on Zephyr](https://micro.ros.org/docs/tutorials/core/first_application_rtos/zephyr/) tutorial using the **ROS2 Foxy docker** option:

```bash
docker run -it --net=host -v /dev:/dev --privileged ros:foxy
```

```bash
# Source the ROS 2 installation
source /opt/ros/$ROS_DISTRO/setup.bash

# Create a workspace and download the micro-ROS tools
mkdir microros_ws
cd microros_ws
git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup

# Update dependencies using rosdep
sudo apt update && rosdep update
rosdep install --from-paths src --ignore-src -y

# Install pip
sudo apt-get install python3-pip

# Build micro-ROS tools and source them
colcon build
source install/local_setup.bash
```

```bash
# Create step
ros2 run micro_ros_setup create_firmware_ws.sh zephyr olimex-stm32-e407
```
#### Expected behavior

No errors.

#### Actual behavior

I can't place the whole output due to maximum 65536 character limitation, so I'll just post the ERRORS in this step:

```bash
ros2 run micro_ros_setup create_firmware_ws.sh zephyr olimex-stm32-e407

```

```text
(...)

Collecting pycparser
  Downloading pycparser-2.21-py2.py3-none-any.whl (118 kB)
     |████████████████████████████████| 118 kB 16.8 MB/s 
Building wheels for collected packages: progress, cbor, future, intervaltree, sh
  Building wheel for progress (setup.py) ... done
  Created wheel for progress: filename=progress-1.6-py3-none-any.whl size=9615 sha256=5626a07b18d6f5de596780904701842224c65226d3adf5bab1275efff1558b63
  Stored in directory: /root/.cache/pip/wheels/bb/01/5a/c916509df9b12c6465864251dbe826def8e31a16fa7da54f08
  Building wheel for cbor (setup.py) ... done
  Created wheel for cbor: filename=cbor-1.0.0-cp38-cp38-linux_x86_64.whl size=57869 sha256=cee0a72cd088690ca970af55b80b307790dbd96eff6e7b220fa7feee8b3183be
  Stored in directory: /root/.cache/pip/wheels/31/3a/8a/9d1a5e6f83f620826578089812b8ac236b86eaf4fbc3b36384
  Building wheel for future (setup.py) ... done
  Created wheel for future: filename=future-0.18.2-py3-none-any.whl size=491058 sha256=b48468252ddcbe41ca2c8f7c3226c8a6d27e9311e1b71d0d0549371abda4adbd
  Stored in directory: /root/.cache/pip/wheels/8e/70/28/3d6ccd6e315f65f245da085482a2e1c7d14b90b30f239e2cf4
  Building wheel for intervaltree (setup.py) ... done
  Created wheel for intervaltree: filename=intervaltree-3.1.0-py2.py3-none-any.whl size=26102 sha256=cbd8bb134381276ab9bc2fab452b2fa6e7d4bc3c27b9f81321b1aec0f6b89360
[... 41 lines cut ...]
Requirement already satisfied, skipping upgrade: certifi>=2017.4.17 in /usr/local/lib/python3.8/dist-packages (from requests>=2.5.0->Sphinx) (2022.6.15)
Requirement already satisfied, skipping upgrade: pyparsing!=3.0.5,>=2.0.2 in /usr/local/lib/python3.8/dist-packages (from packaging->Sphinx) (3.0.9)
Requirement already satisfied, skipping upgrade: zipp>=0.5 in /usr/lib/python3/dist-packages (from importlib-metadata>=4.4; python_version < "3.10"->Sphinx) (1.0.0)
ERROR: sphinx-tabs 3.4.1 has requirement docutils~=0.18.0, but you'll have docutils 0.19 which is incompatible.
ERROR: sphinx-rtd-theme 0.5.2 has requirement docutils<0.17, but you'll have docutils 0.19 which is incompatible.
Installing collected packages: importlib-metadata, Sphinx
  Attempting uninstall: importlib-metadata
    Found existing installation: importlib-metadata 1.5.0
    Not uninstalling importlib-metadata at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'importlib-metadata'. No files were found to uninstall.
  Attempting uninstall: Sphinx
    Found existing installation: Sphinx 3.5.4
    Uninstalling Sphinx-3.5.4:
      Successfully uninstalled Sphinx-3.5.4
Successfully installed Sphinx-5.1.1 importlib-metadata-4.12.0
```

#### Additional information

It seems to be an issue with version dependencies/requirements related to:
- docutils
- sphinx
- breathe
- sphinx-rtd-theme
- sphinx-tabs

pablogs9, 2022-08-11:
Please use https://github.com/micro-ROS/micro_ros_zephyr_module

robodrome, 2022-08-11:
> Please use https://github.com/micro-ROS/micro_ros_zephyr_module

Thank you for your quick response!
Does that mean that microros should always be built as a zephyr module and that the micro_ros_setup approach is deprecated for zephyr? Or is this a more temporary/workaround sort of thing?

pablogs9, 2022-08-11:
micro_ros_setup is not deprecated but it will. The recommended approach is using the Zephyr module

robodrome, 2022-08-11:
> micro_ros_setup is not deprecated but it will. The recommended approach is using the Zephyr module

Ok. Thanks. Good to know. Perhaps write a remark on top of the [tutorial page](https://micro.ros.org/docs/tutorials/core/first_application_rtos/zephyr/)? I will close the issue. Thanks again.

closed 2022-08-11 by robodrome