micro-ROS/micro_ros_setup #563: Create firmware on humble does not get passed ws building

russkel, 2022-08-12:
## Issue template

- Hardware description: nucleo_f767zi
- RTOS: zephyr
- Installation type: micro_ros_setup
- Version or commit hash: humble
- Ubuntu jammy 22.04

#### Steps to reproduce the issue
Follow the tutorial steps (not using the docker image - https://micro.ros.org/docs/tutorials/core/first_application_rtos/zephyr/) and until I get to this step:

```
↳ ros2 run micro_ros_setup create_firmware_ws.sh zephyr nucleo_f767zi
Creating firmware for zephyr platform nucleo_f767zi
.......
=== ./ament/ament_cmake (git) ===
Cloning into '.'...
=== ./ament/ament_index (git) ===
Cloning into '.'...
=== ./ament/ament_lint (git) ===
Cloning into '.'...
=== ./ament/ament_package (git) ===
Cloning into '.'...
=== ./ament/googletest (git) ===
Cloning into '.'...
=== ./ament/uncrustify_vendor (git) ===
Cloning into '.'...
=== ./ros2/ament_cmake_ros (git) ===
Cloning into '.'...
List of repositories is empty
#All required rosdeps installed successfully
..............
=== ./ros2/common_interfaces (git) ===
Cloning into '.'...
=== ./ros2/example_interfaces (git) ===
Cloning into '.'...
=== ./ros2/libyaml_vendor (git) ===
Cloning into '.'...
=== ./ros2/rcl (git) ===
Cloning into '.'...
=== ./ros2/rcl_interfaces (git) ===
Cloning into '.'...
=== ./ros2/rcl_logging (git) ===
Cloning into '.'...
=== ./ros2/rcpputils (git) ===
Cloning into '.'...
=== ./ros2/rmw (git) ===
Cloning into '.'...
=== ./ros2/rmw_implementation (git) ===
Cloning into '.'...
=== ./ros2/rosidl (git) ===
Cloning into '.'...
=== ./ros2/rosidl_dds (git) ===
Cloning into '.'...
=== ./ros2/rosidl_defaults (git) ===
Cloning into '.'...
=== ./ros2/test_interface_files (git) ===
Cloning into '.'...
=== ./ros2/unique_identifier_msgs (git) ===
Cloning into '.'...
...........
=== ./eProsima/Micro-CDR (git) ===
Cloning into '.'...
=== ./eProsima/Micro-XRCE-DDS-Client (git) ===
Cloning into '.'...
=== ./uros/micro_ros_msgs (git) ===
Cloning into '.'...
=== ./uros/micro_ros_utilities (git) ===
Cloning into '.'...
=== ./uros/rcl (git) ===
Cloning into '.'...
=== ./uros/rclc (git) ===
Cloning into '.'...
=== ./uros/rcutils (git) ===
Cloning into '.'...
=== ./uros/rmw_microxrcedds (git) ===
Cloning into '.'...
=== ./uros/rosidl_typesupport (git) ===
Cloning into '.'...
=== ./uros/rosidl_typesupport_microxrcedds (git) ===
Cloning into '.'...
=== ./uros/tracetools (git) ===
Cloning into '.'...
Crosscompiled environment: cleaning path
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> gtest_vendor
Starting >>> ament_cppcheck
Starting >>> ament_pycodestyle
Finished <<< gtest_vendor [1.91s]                                                                                                                     
Starting >>> gmock_vendor
Finished <<< gmock_vendor [0.08s]                                                                                                   
--- stderr: ament_package                                                                                                       
/home/russ/.local/lib/python3.10/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< ament_package [3.57s]
Starting >>> ament_cmake_core
--- stderr: ament_lint                                                                                                                   
/home/russ/.local/lib/python3.10/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< ament_lint [3.63s]
Starting >>> ament_flake8
--- stderr: ament_cppcheck                                                                                                                 
/home/russ/.local/lib/python3.10/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< ament_cppcheck [4.14s]
--- stderr: ament_pycodestyle
/home/russ/.local/lib/python3.10/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< ament_pycodestyle [4.15s]
Starting >>> ament_pyflakes
--- stderr: ament_cmake_core                                                                                    
Traceback (most recent call last):
  File "/home/russ/work/microros_ws/firmware/dev_ws/ament/ament_cmake/ament_cmake_core/cmake/package_templates/templates_2_cmake.py", line 21, in <module>
    from ament_package.templates import get_environment_hook_template_path
ModuleNotFoundError: No module named 'ament_package'
CMake Error at ament_cmake_package_templates-extras.cmake:41 (message):
  execute_process(/usr/bin/python3.10
  /home/russ/work/microros_ws/firmware/dev_ws/ament/ament_cmake/ament_cmake_core/cmake/package_templates/templates_2_cmake.py
  /home/russ/work/microros_ws/firmware/dev_ws/build/ament_cmake_core/ament_cmake_package_templates/templates.cmake)
  returned error code 1
Call Stack (most recent call first):
  CMakeLists.txt:19 (include)

---
Failed   <<< ament_cmake_core [0.64s, exited with code 1]
Aborted  <<< ament_pyflakes [0.03s]
Aborted  <<< ament_flake8 [1.23s]                      

Summary: 6 packages finished [5.35s]
  1 package failed: ament_cmake_core
  2 packages aborted: ament_flake8 ament_pyflakes
  5 packages had stderr output: ament_cmake_core ament_cppcheck ament_lint ament_package ament_pycodestyle
  52 packages not processed
[ros2run]: Process exited with failure 1
```

#### Expected behavior
Workspaces build correctly.

I did notice the `ament` repositories have all been checked out as `rolling` instead of `humble`. Is this deliberate or a carry over from when humble was rolling?

pablogs9, 2022-08-12:
Please use https://github.com/micro-ROS/micro_ros_zephyr_module, micro_ros_setup is going to be deprecated.

russkel, 2022-08-13:
Thanks @pablogs9 I will give it a try.

FYI @natrad100

pablogs9, 2022-09-19:
Any update on this?

vigoncla, 2022-12-19:
I had the same problem. I m very noob with zephyr and with micro-ros. What i should do?

pablogs9, 2022-12-30:
Closing due to inactivity

closed 2022-12-30 by pablogs9

russkel, 2022-12-30:
@pablogs9 I am currently playing around with zephyr for some personal projects. I will have microros wired up soon, and I will report back then.