micro-ROS/micro_ros_setup #568: uros/rcl: RCL_COMMAND_LINE_ENABLED requires RCL_LOGGING_ENABLED?

gavanderhoorn, 2022-08-16:
(not sure whether this is the correct venue, but [micro-ROS/rcl](https://github.com/micro-ROS/rcl) doesn't have the tracker enabled)

Just wanted to ask whether `micro-ROS/rcl/rcl` requires [RCL_LOGGING_ENABLED](https://github.com/micro-ROS/rcl/blob/98bb690e17b147f194bd2c92786784814539702e/rcl/CMakeLists.txt#L6) if [RCL_COMMAND_LINE_ENABLED](https://github.com/micro-ROS/rcl/blob/98bb690e17b147f194bd2c92786784814539702e/rcl/CMakeLists.txt#L5) is enabled?

I'm running into unresolved symbol errors when loading a binary linked against a `libmicroros.a` which was built with `RCL_LOGGING_ENABLED:OFF` and `RCL_COMMAND_LINE_ENABLED:ON` (to get access to `rcl_parse_arguments(..)`).

Specifically: `rcl_log_levels_fini`, `rcl_log_levels_copy`, `rcl_log_levels_init`, `rcl_log_levels_shrink_to_size`, `rcl_get_zero_initialized_log_levels` and `rcl_log_levels_add_logger_setting`, which all appear to be used in [rcl/arguments.c](https://github.com/micro-ROS/rcl/blob/98bb690e17b147f194bd2c92786784814539702e/rcl/src/rcl/arguments.c) (`rcl_get_zero_initialized_log_levels()` [here](https://github.com/micro-ROS/rcl/blob/98bb690e17b147f194bd2c92786784814539702e/rcl/src/rcl/arguments.c#L310) for instance).

(transferred from #557)

pablogs9, 2022-08-22:
`[RCL_COMMAND_LINE_ENABLED](https://github.com/micro-ROS/rcl/blob/98bb690e17b147f194bd2c92786784814539702e/rcl/CMakeLists.txt#L5)` is not intended to be enabled in micro-ROS as far as the is no CLI handling in micro-ROS.

closed 2022-08-22 by pablogs9

gavanderhoorn, 2022-08-22:
I understand.

But it seems to be necessary to get access to `rcl_parse_arguments(..)`, which is needed to get remapping working.