micro-ROS/micro_ros_setup #570: Build micro-ROS firmware creation failed for udp ESP32

At-Az, 2022-08-18:


At-Az, 2022-08-18:
Yes you are right, The error occurs during build step. (edited first post)
I still got an error when building following your link. 

- fresh ubuntu, ROS2 install
- ESP-IDF V4.4.2 (build and flashed a "hello world" example and it works fine on same hardware)
- ROS2 setup script sources and not sourced gave same error
Any suggestions?
Thanks!
```
ros2box@ros2box-VM:~/uros_ws/components/micro_ros_espidf_component/examples/int32_publisher$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/ros2box/uros_ws/components/micro_ros_espidf_component/examples/int32_publisher/build
Executing "ninja all"...
[0/2] Re-checking globbed directories...
[1/2] Re-running CMake...
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/ros2box/uros_ws/components/micro_ros_espidf_component/examples/int32_publisher/sdkconfig
Loading defaults file /home/ros2box/uros_ws/components/micro_ros_espidf_component/examples/int32_publisher/sdkconfig.defaults...
-- App "int32_publisher" version: 3.0.0-1-gf426aa2
-- Adding linker script /home/ros2box/uros_ws/components/micro_ros_espidf_component/examples/int32_publisher/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/ros2box/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /home/ros2box/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/ros2box/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/ros2box/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/ros2box/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/ros2box/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/ros2box/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/ros2box/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /home/ros2box/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- Using Micro XRCE-DDS middleware
mkdir -p /home/ros2box/uros_ws/components/micro_ros_espidf_component/micro_ros_src/libmicroros; cd /home/ros2box/uros_ws/components/micro_ros_espidf_component/micro_ros_src/libmicroros; \
for file in $(find /home/ros2box/uros_ws/components/micro_ros_espidf_component/micro_ros_src/install/lib/ -name '*.a'); do \
	folder=$(echo $file | sed -E "s/(.+)\/(.+).a/\2/"); \
	mkdir -p $folder; cd $folder; /home/ros2box/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar x $file; \
	for f in *; do \
		mv $f ../$folder-$f; \
	done; \
	cd ..; rm -rf $folder; \
done ; \
/home/ros2box/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar rc -s libmicroros.a *.obj; cp libmicroros.a /home/ros2box/uros_ws/components/micro_ros_espidf_component; \
cd ..; rm -rf libmicroros; \
cp -R /home/ros2box/uros_ws/components/micro_ros_espidf_component/micro_ros_src/install/include /home/ros2box/uros_ws/components/micro_ros_espidf_component/include;
/home/ros2box/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar: *.obj: No such file or directory
cp: cannot stat 'libmicroros.a': No such file or directory
cp: cannot stat '/home/ros2box/uros_ws/components/micro_ros_espidf_component/micro_ros_src/install/include': No such file or directory
make: *** [libmicroros.mk:128: /home/ros2box/uros_ws/components/micro_ros_espidf_component/libmicroros.a] Error 1
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip main mbedtls mdns micro_ros_espidf_component mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa

-- Component paths: /home/ros2box/esp/esp-idf/components/app_trace /home/ros2box/esp/esp-idf/components/app_update /home/ros2box/esp/esp-idf/components/asio /home/ros2box/esp/esp-idf/components/bootloader /home/ros2box/esp/esp-idf/components/bootloader_support /home/ros2box/esp/esp-idf/components/bt /home/ros2box/esp/esp-idf/components/cbor /home/ros2box/esp/esp-idf/components/cmock /home/ros2box/esp/esp-idf/components/coap /home/ros2box/esp/esp-idf/components/console /home/ros2box/esp/esp-idf/components/cxx /home/ros2box/esp/esp-idf/components/driver /home/ros2box/esp/esp-idf/components/efuse /home/ros2box/esp/esp-idf/components/esp-tls /home/ros2box/esp/esp-idf/components/esp32 /home/ros2box/esp/esp-idf/components/esp_adc_cal /home/ros2box/esp/esp-idf/components/esp_common /home/ros2box/esp/esp-idf/components/esp_eth /home/ros2box/esp/esp-idf/components/esp_event /home/ros2box/esp/esp-idf/components/esp_gdbstub /home/ros2box/esp/esp-idf/components/esp_hid /home/ros2box/esp/esp-idf/components/esp_http_client /home/ros2box/esp/esp-idf/components/esp_http_server /home/ros2bo [... cut ...] df/components/wifi_provisioning /home/ros2box/esp/esp-idf/components/wpa_supplicant /home/ros2box/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ros2box/uros_ws/components/micro_ros_espidf_component/examples/int32_publisher/build
[0/2] Re-checking globbed directories...
ninja: error: '/home/ros2box/uros_ws/components/micro_ros_espidf_component/libmicroros.a', needed by 'int32_publisher.elf', missing and no known rule to make it
```

Acuadros95, 2022-08-19:
Looks like something failed on your micro-ROS library build.

Did you run `idf.py set-target esp32`? What is your output for that command? (Run `idf.py clean-microros` first)
You should get the library compiled as `libmicroros.a` on the `micro_ros_espidf_component` main folder.

At-Az, 2022-08-19:
Some progress here: re-run `idf.py clean-microros` then `idf.py set-target esp32`  and `idf.py build` (attached output) 
[set-target_ros2NotSourced.txt](https://github.com/micro-ROS/micro_ros_setup/files/9384761/set-target_ros2NotSourced.txt)
[build_ros2NotSourced.txt](https://github.com/micro-ROS/micro_ros_setup/files/9384762/build_ros2NotSourced.txt)
There was some warnings but a firmware binary was generated and I was able to use `idf.py flash` to flash the firmware. 
The command `idf.py monitor` failed though. It turns out virtualbox coulden't attach the virtual serial port to the gust VM anymore. I'm assuming the serial interface was disabled because wifi transport layer was configured(?). When the esp module is booted on DFU mode then the virtual serial port become available again and was able the re-flash the module with same firmware.
```
ros2box@ros2box-VM:~/uros_ws/components/micro_ros_espidf_component/examples/int32_publisher$ idf.py monitor
Executing action: monitor
No serial ports found. Connect a device, or use '-p PORT' option to set a specific port.
``` 

Started a micro-ROS Agent with docker command 
```
sudo docker run -it --rm --net=host microros/micro-ros-agent:humble udp4 --port 8888 -v6
[sudo] password for ros2box: 
[1660938028.386789] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1660938028.387141] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6 
```
Didn't look like the ROS2 node connected to the agent (rebooted module couple of times but same results)
I run `ros2 topic list` the /int32_publisher topic didn't show up
```
ros2box@ros2box-VM:~$ ros2 topic list
/parameter_events
/rosout
```
Created another micro-ROS workspace and built/run micro-ROS agent using commands: `ros2 run micro_ros_setup create_agent_ws.sh`  `ros2 run micro_ros_setup build_agent.sh` and `ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888`. Same results as the docker one. 
To verify that the module was actually running and not hang or just rebooting: Checked main router for attached devices and the module was registered and active. ping command was also working fine (stops only when I force module to reboot then picks up again)
Could it be that the build command didn't produce a "proper" binary?
--General question-- Is it possible to configure micro-ROS to use multiple hardware interfaces as transport layer simultaneously for redundancy (serial&CAN) or (serial&udp)? 
Will probably try to build another firmware with serial transport layer and report back.

Acuadros95, 2022-08-22:
The Agent looks fine, could you check your network traffic?

The board should sent the UDP packets to the configured IP and port. Also check that the defined port (8888) is accessible and your VM is not messing arround.

> --General question-- Is it possible to configure micro-ROS to use multiple hardware interfaces as transport layer simultaneously for redundancy (serial&CAN) or (serial&udp)?

No, that would be problematic on the Agent side, as you would need two agents or one that handles data from two interfaces.
Anyway, you can always use reliable communication and the ping functionality to check that your connection is OK.

At-Az, 2022-08-22:
It turns out I had an issue with VirtualBox network adapter setting. Ones that was fixed then the Agent was communicating with the node properly. 
Thank you very much for the support and patience :)

closed 2022-08-22 by At-Az

Cherni-Wale, 2023-05-24:
At Az can you mention how you fix it because i have the same problem as you , and i think the VirtualBox is responsible for it

At-Az, 2023-05-29:
Try a fresh install of VirtualBox and set an extra network adapter as "bridged" with the host adapter that is connected to the ROS network. Hope this helps.
[image]

Cherni-Wale, 2023-05-29:
I followed your advice, and it worked like a charm! I am truly amazed and incredibly grateful for the time you took to help me out.  
Once again, thank you At-Az