micro-ROS/micro_ros_setup #571: Create firmware step failed for freertos and olimex-stm32-e407

ravi-nanjaiah, 2022-08-18:
## Issue template

- Hardware description: olimex-stm32-e407
- RTOS: FreeRTOS
- Installation type: MICRO_ROS_SETUP
- Version or commit hash: Foxy

#### Steps to reproduce the issue
I'm trying to follow the tutorials to build a demo application as per instructions in https://micro.ros.org/docs/tutorials/core/first_application_rtos/freertos/
1. install ROS2 Foxy as per instructions in https://docs.ros.org/en/foxy/Installation/Ubuntu-Install-Debians.html
2. Install micro_ros_setup build system as per instructions
3. Follow the create firmware step: `ros2 run micro_ros_setup create_firmware_ws.sh freertos olimex-stm32-e407`

#### Expected behavior
Firmware preparation is complete and progress to configure application step
#### Actual behavior
Create firmware step fails with following error
`ImportError: cannot import name 'environmentfilter' from 'jinja2' (/home/nrr2kor/.local/lib/python3.8/site-packages/jinja2/__init__.py)
`
#### Additional information
Through out the create firmware step the above error message is multiple times.

'$ ros2 run micro_ros_setup create_firmware_ws.sh freertos olimex-stm32-e407
Creating firmware for freertos platform olimex-stm32-e407
.......
=== ./ament/ament_cmake (git) ===
Cloning into '.'...
=== ./ament/ament_index (git) ===
Cloning into '.'...
=== ./ament/ament_lint (git) ===
Cloning into '.'...
=== ./ament/ament_package (git) ===
Cloning into '.'...
=== ./ament/googletest (git) ===
Cloning into '.'...
=== ./ament/uncrustify_vendor (git) ===
Cloning into '.'...
=== ./ros2/ament_cmake_ros (git) ===
Cloning into '.'...
List of repositories is empty
#All required rosdeps installed successfully
..............
=== ./ros2/common_interfaces (git) ===
Cloning into '.'...
=== ./ros2/example_interfaces (git) ===
Cloning into '.'...
=== ./ros2/libyaml_vendor (git) ===
Cloning into '.'...
=== ./ros2/rcl (git) ===
Cloning into '.'...
=== ./ros2/rcl_interfaces (git) ===
Cloning into '.'...
=== ./ros2/rcl_logging (git) ===
Cloning into '.'...
=== ./ros2/rcpputils (git) ===
Cloning into '.'...
=== ./ros2/rmw (git) ===
Cloning into '.'...
=== ./ros2/rmw_implementation (git) ===
Cloning into '.'...
=== ./ros2/rosidl (git) ===
Cloning into '.'...
=== ./ros2/rosidl_dds (git) ===
Cloning into '.'...
=== ./ros2/rosidl_defaults (git) ===
Cloning into '.'...
=== ./ros2/test_interface_files (git) ===
Cloning into '.'...
=== ./ros2/unique_identifier_msgs (git) ===
Cloning into '.'...
[... 107 lines cut ...]
    self.print_commands()
  File "/home/nrr2kor/.local/lib/python3.8/site-packages/setuptools/dist.py", line 962, in print_commands
    cmdclass = ep.load()
  File "/home/nrr2kor/.local/lib/python3.8/site-packages/setuptools/_vendor/importlib_metadata/__init__.py", line 194, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/nrr2kor/.local/lib/python3.8/site-packages/sphinx/setup_command.py", line 20, in <module>
    from sphinx.application import Sphinx
  File "/home/nrr2kor/.local/lib/python3.8/site-packages/sphinx/application.py", line 43, in <module>
    from sphinx.registry import SphinxComponentRegistry
  File "/home/nrr2kor/.local/lib/python3.8/site-packages/sphinx/registry.py", line 24, in <module>
    from sphinx.builders import Builder
  File "/home/nrr2kor/.local/lib/python3.8/site-packages/sphinx/builders/__init__.py", line 26, in <module>
    from sphinx.util import import_object, logging, progress_message, rst, status_iterator
  File "/home/nrr2kor/.local/lib/python3.8/site-packages/sphinx/util/rst.py", line 21, in <module>
    from jinja2 import Environment, environmentfilter
ImportError: cannot import name 'environmentfilter' from 'jinja2' (/home/nrr2kor/.local/lib/python3.8/site-packages/jinja2/__init__.py)
'

Acuadros95, 2022-08-19:
Could not replicate this on a fresh docker.

What is your output for `pip3 list | grep Jinja2`?
Could you try again after downgrading Jinja2 version with `pip3 install jinja2==3.0.0`?

Related: https://github.com/readthedocs/readthedocs.org/issues/9038

ravi-nanjaiah, 2022-08-22:
Thanks @Acuadros95 
Downgrading Jinja2 version worked. Build was completed post downgrade.
Downgrading had to be done post the configure firmware step.

closed 2022-08-22 by ravi-nanjaiah