micro-ROS/micro_ros_setup #575: micro_ros_setup uses old driver of I2C. Important function missing.

MiguelMartinez-HAW, 2022-09-02:
## I2C Function missing (i2c_master_write_read_device)

- Hardware description: ESP32
- RTOS: FreeRTOS
- Installation type: micro_ros_setup
- Version or commit hash: foxy

#### Steps to reproduce the issue
Following the steps described in the [official documentation](https://micro.ros.org//docs/tutorials/core/first_application_rtos/freertos/) lead to this part: 
`ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32`
which creates the folder `firmware` in `microros_ws` which contains `esp-idf` and its components. The folder `driver` contains the files `i2c.c` and `i2c.h`. 
Before implementing micro-ros, I had developed software for the ESP32 using the latest stable version of esp-idf (4.4.2). 

#### Expected behavior vs Actual behavior
When porting my code to use micro-ros, it turns out that one of the basic functions of I2C `i2c_master_write_read_device` is not available in the esp-idf version used by micro_ros_setup. I checked the file comments and the i2c.c file installed by micro_ros_setup has a date of 2019 whereas the current one is from 2021. Needless to say that my code doesn't work when I try to use micro-ros.

#### Solutions I tried
I tried overwriting the old driver files with the new files but because of its dependencies, it affected other files that suddenly couldn't find certain macros. I went on to replace every file that had an issue with a "missing definition" or function. It's a rabbit hole with too many dependencies and replacing all possible files didn't solve the issue either.

#### Question
Before I attempt to rewrite my code with a legacy version of i2c.c, I'd like to ask if there is a simple way to update the I2C driver that `micro_ros_setup` uses. If you know of any other solution, I'm open to hearing (reading) it.

Thanks everyone for your help.

pablogs9, 2022-09-05:
I guess that you can modify the IDF version here: https://github.com/micro-ROS/micro_ros_setup/blob/67a7cdb5fdea4bf09e396680ebf2b6d72397746e/config/freertos/esp32/create.sh#L6

As you mention, `micro_ros_setup` is just a tool for creating boilerplate projects, if you need the newest version, move to the micro-ROS module for ESP-IDF.

Could you provide some more details about your project?

pablogs9, 2022-09-05:
In any case I'm closing since this is not related to micro-ROS

closed 2022-09-05 by pablogs9

MiguelMartinez-HAW, 2022-09-05:
> I guess that you can modify the IDF version here:
> 
> https://github.com/micro-ROS/micro_ros_setup/blob/67a7cdb5fdea4bf09e396680ebf2b6d72397746e/config/freertos/esp32/create.sh#L6

Hi, thank you for your suggestion. Unfortunately, when I update the version in create.sh file and then run
`ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32`
Followed by 
`ros2 run micro_ros_setup configure_firmware.sh sensor_data_publisher --transport serial`
and `ros2 run micro_ros_setup build_firmware.sh`
I get this error :
```microros_ws/firmware/toolchain/esp-idf/export.sh: line 16: IDF_EXPORT_QUIET: unbound variable```

> As you mention, `micro_ros_setup` is just a tool for creating boilerplate projects, if you need the newest version, move to the micro-ROS module for ESP-IDF.

If I understood correctly from [this](https://github.com/micro-ROS/micro_ros_setup/issues/544#issuecomment-1232865789), my problem with the micro-ROS module lies in using Foxy instead of Humble. So at the moment, I cannot use it until we update to Humble.

> Could you provide some more details about your project?

It's a mobile robot. We want to use the ESP32 to handle multiple distance sensors through I2C, an IMU, lights, motors and an RC receiver. Ideally, multitasking should be implemented with the help of FreeRTOS and the robot communicates to a laptop running ROS2-Foxy.

pablogs9, 2022-09-05:
Let's move to the issue with Foxy and ESP-IDF. As far as I know, Foxy shall be working properly with ESP-IDF, I'll check.