micro-ROS/micro_ros_setup #576: when I choose uart&udp transport， UDP is not normal, but uart is normal.

navy-to-haijun, 2022-09-10:
- Hardware description: [STMH750-ART-Pi](https://github.com/RT-Thread-Studio/sdk-bsp-stm32h750-realthread-artpi).
- RTOS: [RT-Thread](https://github.com/RT-Thread/rt-thread).
- Installation type: using micro_ros_setup build  static micro-ROS library.
 micro-ROS  Version : galactic
transport: UART、DUP(WIFI)

#### Steps to reproduce the issue
1. Building static micro-ROS library [link](https://micro.ros.org/docs/tutorials/advanced/create_custom_transports/)
2. Writing UART transport;
3. According to the [link](https://github.com/micro-ROS/micro-ROS-demos/blob/humble/rclc/int32_publisher/main.c), using int32_publisher demo to test micro-ROS. Communication is normal.
4. According to the [link](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/master/src/c/profile/transport/ip/udp/udp_transport_posix_nopoll.c), writing UDP transport. publisher is normal.
5.  Then I use int32_subscriber demo, UART is normal. UDP can build node subscriber and executor , then micro-ROS no longer subscribes to ROS2 data.   

#### Additional information

**PC agent:**
```bash
$ ros2 run micro_ros_agent micro_ros_agent udp4 --port 9999 -v6
[1662820128.982161] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 9999
[1662820128.982472] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1662820376.867852] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 2C F4 52 89 81 00 FC 01
[1662820376.868295] info     | Root.cpp           | create_client            | create                 | client_key: 0x2CF45289, session_id: 0x81
[1662820376.868483] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x2CF45289, address: 192.168.31.73:704
[1662820376.868738] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2CF45289, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1662820376.874013] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2CF45289, len: 52, data: 
0000: 81 80 00 00 01 07 2A 00 00 0A 00 01 01 03 00 00 1B 00 00 00 00 01 01 24 13 00 00 00 6D 69 63 72
0020: 6F 5F 72 6F 73 5F 72 74 74 5F 6E 6F 64 65 00 00 00 00 00 00
[1662820376.893328] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x2CF45289, participant_id: 0x000(1)
[1662820376.893437] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2CF45289, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1662820376.893465] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2CF45289, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1662820376.899540] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2CF45289, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1662820376.901077] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2CF45289, len: 96, data: 
0000: 81 80 01 00 01 07 56 00 00 0B 00 02 02 03 00 00 48 00 00 00 21 00 00 00 72 74 2F 6D 69 63 72 6F
0020: 5F 72 6F 73 5F 72 74 74 5F 6E 6F 64 65 5F 73 75 62 73 63 72 69 62 65 72 00 00 01 00 1C 00 00 00
0040: 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 49 6E 74 33 32 5F 00 00 01 00 00
[1662820376.901209] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x2CF45289, topic_id: 0x000(2), participant_id: 0x000(1)
[1662820376.901403] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2CF45289, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00
[1662820376.901428] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2CF45289, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1662820376.907318] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2CF45289, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1662820376.907693] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2CF45289, len: 24, data: 
0000: 81 80 02 00 01 07 10 00 00 0C 00 04 04 03 00 00 02 00 00 00 00 00 00 01
[1662820376.907926] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x2CF45289, subscriber_id: 0x000(4), participant_id: 0x000(1)
[1662820376.908044] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2CF45289, len: 14, data: 
0000: 81 80 02 00 05 01 06 00 00 0C 00 04 00 00
[1662820376.908071] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2CF45289, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1662820376.913629] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2CF45289, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1662820376.914885] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2CF45289, len: 40, data: 
0000: 81 80 03 00 01 07 1D 00 00 0D 00 06 06 03 00 00 0F 00 00 00 00 02 01 24 03 00 01 00 0A 00 00 00
0020: 00 00 00 00 04 00 00 00
[1662820376.915346] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x2CF45289, datareader_id: 0x000(6), subscriber_id: 0x000(4)
[1662820376.915423] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2CF45289, len: 14, data: 
0000: 81 80 03 00 05 01 06 00 00 0D 00 06 00 00
[1662820376.915452] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2CF45289, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1662820376.921738] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2CF45289, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1662820377.026724] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2CF45289, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 03 00 80
[1662820377.027128] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2CF45289, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
```

**PC ROS2 publisher**
```
$ ros2 topic list -t
/micro_ros_rtt_node_subscriber [std_msgs/msg/Int32]
/parameter_events [rcl_interfaces/msg/ParameterEvent]
/rosout [rcl_interfaces/msg/Log]

$ ros2 topic pub  /micro_ros_rtt_node_subscriber std_msgs/msg/Int32 data:\ 1
publisher: beginning loop
publishing #1: std_msgs.msg.Int32(data=1)

publishing #2: std_msgs.msg.Int32(data=1)

publishing #3: std_msgs.msg.Int32(data=1)
```
**STMH750-ART-Pi**

**UDP transport don't receive data**. But, UART transport can  receive data as follows:
```
msh />Received: 1
Received: 1
Received: 1
Received: 1
```

pablogs9, 2022-09-12:
I do not understand what is the issue. I do not understand which is the expected behavior or which is the failure here.

Please try to explain your problem clearly and provide code and steps for replicating the issue.

navy-to-haijun, 2022-09-13:
Sorry，I didn't express the issue clearly.
Expected behavior: micros-ROS implements subscription a topic by UDP trainsport.
Failure: When ROS2 publishes a topic, it seems that micros-ROS does not subscribe to the corresponding topic.

### UDP trainsport code: 
**Head File:**
```c
#ifndef MICRO_ROS_RTT
#define MICRO_ROS_RTT

#include <rtthread.h>
#include <sys/time.h>

#include <uxr/client/transport.h>
#include <rmw_microros/rmw_microros.h>

#if defined MICROROS_UDP

bool   rtt_udp_transport_open(struct uxrCustomTransport * transport);
bool   rtt_udp_transport_close(struct uxrCustomTransport * transport);
size_t rtt_udp_transport_write(struct uxrCustomTransport* transport, const uint8_t * buf, size_t len, uint8_t * err);
size_t rtt_udp_transport_read(struct uxrCustomTransport* transport, uint8_t* buf, size_t len, int timeout, uint8_t* err);
int clock_gettime(clockid_t unused, struct timespec *tp);

struct micro_ros_agent_locator {
    char* address;
    int port;
};

static inline void set_microros_udp_transports(char * agent_ip, uint32_t agent_port){
    static struct micro_ros_agent_locator locator;
    locator.address = agent_ip;
    locator.port = agent_port;

    rmw_uros_set_custom_transport(
        false,
        (void *) &locator,
        rtt_udp_transport_open,
        rtt_udp_transport_close,
        rtt_udp_transport_write,
        rtt_udp_transport_read
    );
}
    #endif

#endif

```
**Source File**
```c
#include <rtthread.h>
#include "micro_ros_rtt.h"
#include <stdio.h>
#include <stdbool.h>
#include <sys/time.h>
#include <sys/socket.h>
#include <netdb.h>

#include <rtdbg.h>

#if defined MICROROS_UDP

#define micro_rollover_useconds 4294967295

int clock_gettime(clockid_t unused, struct timespec *tp)
{
    (void)unused;
    static uint32_t rollover = 0;
    static uint32_t last_measure = 0;

[... 242 lines cut ...]
[1663074309.889615] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x763D219F, len: 24, data: 
0000: 81 80 02 00 01 07 10 00 00 0C 00 04 04 03 00 00 02 00 00 00 00 00 00 01
[1663074309.889814] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x763D219F, subscriber_id: 0x000(4), participant_id: 0x000(1)
[1663074309.889947] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x763D219F, len: 14, data: 
0000: 81 80 02 00 05 01 06 00 00 0C 00 04 00 00
[1663074309.889976] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x763D219F, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1663074309.903946] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x763D219F, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1663074309.905269] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x763D219F, len: 40, data: 
0000: 81 80 03 00 01 07 1D 00 00 0D 00 06 06 03 00 00 0F 00 00 00 00 02 01 24 03 00 01 00 0A 00 00 00
0020: 00 00 00 00 04 00 00 00
[1663074309.905754] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x763D219F, datareader_id: 0x000(6), subscriber_id: 0x000(4)
[1663074309.905849] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x763D219F, len: 14, data: 
0000: 81 80 03 00 05 01 06 00 00 0D 00 06 00 00
[1663074309.905877] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x763D219F, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1663074309.912257] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x763D219F, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1663074310.018049] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x763D219F, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 03 00 80
[1663074310.018401] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x763D219F, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80

```

pablogs9, 2022-09-14:
The micro-ROS side seems to be correct, but the ROS 2 pub and the agent seem to be not discovering each other.

Are the agent and the publisher running on the very same machine?
Which version of ROS 2 and micro-ROS Agent are you using?

navy-to-haijun, 2022-09-14:
> 
Thank you for your reply.
The agent and the publisher run on the same machine.
version of ROS 2 : galactic
micro-ROS Agent：`git clone -b  galactic https://github.com/micro-ROS/micro_ros_setup.git`, then building micro-ROS-Agent. So the version of micro-ROS Agent is also galactic. 

In addition, In the same environment and code, if micro-ROS is used as a publishing node, ROS 2 can subscribe to the corresponding topic.

pablogs9, 2022-09-14:
Have you changed the default middleware in the ROS 2 side to Fast DDS or are you using the default Cyclone DDS middleware implementation? https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/ros2.html

navy-to-haijun, 2022-09-14:
> Have you changed the default middleware in the ROS 2 side to Fast DDS or are you using the default Cyclone DDS middleware implementation? https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/ros2.html

Check RMW MIDDLEWARE : `ros2 doctor --report`
```bash
/opt/ros/galactic/lib/python3.8/site-packages/ros2doctor/api/__init__.py: 143: UserWarning: Fail to call PackageReport class functions.

   NETWORK CONFIGURATION
inet         : 172.17.0.1
inet4        : ['172.17.0.1']
ether        : 02:42:96:80:b9:0a
netmask      : 255.255.0.0
device       : docker0
flags        : 4099<UP,BROADCAST,MULTICAST> 
mtu          : 1500
broadcast    : 172.17.255.255
ether        : 54:ee:75:af:4d:33
device       : enp3s0
flags        : 4099<UP,BROADCAST,MULTICAST> 
mtu          : 1500
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<UP,LOOPBACK,RUNNING> 
mtu          : 65536
inet         : 192.168.31.130
inet4        : ['192.168.31.130']
ether        : 84:ef:18:0f:6b:b7
inet6        : ['fe80::c468:968f:9406:e4fd']
netmask      : 255.255.255.0
device       : wlp2s0
flags        : 4163<UP,BROADCAST,RUNNING,MULTICAST> 
mtu          : 1500
broadcast    : 192.168.31.255

   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-5.15.0-46-generic-x86_64-with-glibc2.29
release          : 5.15.0-46-generic
processor        : x86_64

   RMW MIDDLEWARE
middleware name    : rmw_cyclonedds_cpp

   ROS 2 INFORMATION
distribution name      : galactic
distribution type      : ros2
distribution status    : active
release platforms      : {'rhel': ['8'], 'ubuntu': ['focal']}

   TOPIC LIST
topic               : none
publisher count     : 0
subscriber count    : 0
```
from  the report, RMW MIDDLEWARE is Cyclone DDS.

pablogs9, 2022-09-14:
Please change it to Fast DDS and check if the issue is still happening.

navy-to-haijun, 2022-09-14:
> Please change it to Fast DDS and check if the issue is still happening.

Thank you for your help.  When I add  `export RMW_IMPLEMENTATION=rmw_fastrtps_dynamic_cpp` in the terminal before run ROS 2, subscribing nodes work fine.

closed 2022-09-14 by pablogs9