micro-ROS/micro_ros_setup #577: Multiple clients to single agent connection

Sorokonog, 2022-09-12:
Hello! I'm creating CAN-client and everything goes well with single client (stm32) and single agent runs on raspberry (Linux). But with 3 clients the agent can communicate with only one of them (I suppose the first one which created session). Which way I should follow to connect them all?

Sorokonog, 2022-09-13:
```C
#include <uxr/agent/transport/custom/CustomAgent.hpp>
#include <uxr/agent/transport/endpoint/CustomEndPoint.hpp>

#include <sys/time.h>

#include <poll.h>
#include <sys/socket.h>
#include <unistd.h>
#include <signal.h>
#include <netinet/in.h>
#include <arpa/inet.h>

#include <uxr/agent/transport/can/CanAgentLinux.hpp>
#include <uxr/agent/utils/Conversion.hpp>
#include <uxr/agent/logger/Logger.hpp>

#include <net/if.h>
#include <sys/ioctl.h>

#include <linux/can.h>
#include <linux/can/raw.h>

/**
 * This custom XRCE Agent example attempts to show how easy is for the user to define a custom
 * Micro XRCE-DDS Agent behaviour, in terms of transport initialization and closing, and also
 * regarding read and write operations.
 * For this simple case, an UDP socket is opened on port 8888. Additionally, some information
 * messages are being printed to demonstrate the custom behaviour.
 * As the endpoint is already defined, we are using the provided
 * `eprosima::uxr::IPv4EndPoint` by the library.
 * Other transport protocols might need to implement their own endpoint struct.
 */
int64_t time_ms(void);

int main(int argc, char** argv)
{
    eprosima::uxr::Middleware::Kind mw_kind(eprosima::uxr::Middleware::Kind::FASTDDS);
    
    struct pollfd poll_fd;
    struct sockaddr_can addr;
    struct ifreq ifr;

    /**
     * @brief Agent's initialization behaviour description.
     */
    eprosima::uxr::CustomAgent::InitFunction init_function = [&]() -> bool
    {
        //struct sockaddr_can addr;
        struct canfd_frame frame;
        struct can_frame rec_frame;

        bool rv = false;

        if ((poll_fd.fd = socket(PF_CAN, SOCK_RAW, CAN_RAW)) < 0) 
        {
		perror("socket creation: ");
		return false;
        }
        
        if (-1 != poll_fd.fd)
        {
            struct sockaddr_in address{};
            strcpy(ifr.ifr_name, "can0");
            ioctl(poll_fd.fd, SIOCGIFINDEX, &ifr);
            poll_fd.events = POLLIN;
            addr.can_family  = AF_CAN;
            addr.can_ifindex = ifr.ifr_ifindex;
            if (-1 != bind(poll_fd.fd,
                (struct sockaddr *)&addr,
[... 160 lines cut ...]
        custom_agent.start();

        int n_signal = 0;
        sigset_t signals;
        sigwait(&signals, &n_signal);

        /**
         * Stop agent, and exit.
         */
        custom_agent.stop();
        return 0;
    }
    catch (const std::exception& e)
    {
        std::cout << e.what() << std::endl;
        return 1;
    }
}

int64_t time_ms(void)
{
    struct timeval tv;
    gettimeofday(&tv, NULL);
    return (int64_t)tv.tv_sec * 1000 + tv.tv_usec / 1000;
}

Acuadros95, 2022-09-13:
Are you setting different `can_id` on each client frames?

Check that this value is different for each client:
`source_endpoint->set_member_value<uint16_t>("ID",frame.can_id);`

You can check our transport implementations for CAN FD:
- [Client](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/master/src/c/profile/transport/can/can_transport_posix.c)
- [Agent](https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/master/src/cpp/transport/can/CanAgentLinux.cpp)

Check the Agent `bool CanAgent::recv_message` method, the approach is quite similar: [detail](https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/master/src/cpp/transport/can/CanAgentLinux.cpp#L174).

Sorokonog, 2022-09-13:
> Are you setting different `can_id` on each client frames?
> Check that this value is different for each client: `source_endpoint->set_member_value<uint16_t>("ID",frame.can_id);
I set differnet can_id for each client, but its remain the same inside each client's CAN frame. Basically can_id is ID of STM32 board sending through CAN. Its 1 for all CAN frames from first board, 2 for second board and etc.

> You can check our transport implementations for CAN FD:
Looked through it and can't find any clue.

Acuadros95, 2022-09-13:
> but its remain the same inside each client's CAN frame

What does this mean? Does the agent get a different `frame.can_id` for each client?

Sorokonog, 2022-09-13:
> 

Yes. Its different for each client.

Sorokonog, 2022-09-13:
[image]

Sorokonog, 2022-09-14:
> You can manually modify your client key on micro-ROS side, check this example: https://github.com/micro-ROS/micro-ROS-demos/blob/humble/rclc/configuration_example/custom_transports/main.c#L134

Great! Thanks! That works. 

Still have strange clients framerate distributions: 1st one has ~70 HZ on ros2 topic hz, 2nd ~ 20HZ, 3rd ~40HZ. But they are identical and has one chain CAN-bus connection. Will ellaborate the reasons of such behaviour.

Thanks a lot for your help!

Final DefaultTask for Freertos for whom it may concern:

```C
void StartDefaultTask(void *argument)
{
  /* USER CODE BEGIN StartDefaultTask */

	rcl_publisher_t publisher;
	rcl_subscription_t subscriber;
	rclc_executor_t executor;
	rclc_support_t support;
	rcl_allocator_t allocator;
	rcl_node_t node;
	rcl_init_options_t init_options;

	rmw_uros_set_custom_transport(
			    true,
			    (void *) &hcan1,
			    cubemx_transport_open,
			    cubemx_transport_close,
			    cubemx_transport_write,
			    cubemx_transport_read);

	rcl_allocator_t freeRTOS_allocator = rcutils_get_zero_initialized_allocator();
	freeRTOS_allocator.allocate = microros_allocate;
	freeRTOS_allocator.deallocate = microros_deallocate;
	freeRTOS_allocator.reallocate = microros_reallocate;
	freeRTOS_allocator.zero_allocate =  microros_zero_allocate;

	//create init_options
	allocator = rcl_get_default_allocator();
	init_options = rcl_get_zero_initialized_init_options();
	rcl_init_options_init(&init_options, allocator);

	rmw_init_options_t* rmw_options = rcl_init_options_get_rmw_init_options(&init_options);
	rmw_uros_options_set_client_key(CLIENT_KEY, rmw_options);

	//support init
	rclc_support_init_with_options(&support, 0, NULL, &init_options, &allocator);

	char node_name[13];
	char topic_pub_name[18];
	char topic_sub_name[18];
	char joint_name[8];

	sprintf(node_name,"joint_node_%d", MY_CAN_ID);
	sprintf(topic_pub_name,"joint_state_pub_%d", MY_CAN_ID);
	sprintf(topic_sub_name,"joint_state_sub_%d", MY_CAN_ID);
	sprintf(joint_name,"joint_%d", MY_CAN_ID);

	// create node
	rclc_node_init_default(&node, node_name, "", &support);

	// create publisher
	rclc_publisher_init_default(
		&publisher,
		&node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(sensor_msgs, msg, JointState),
		topic_pub_name);

	rcl_ret_t rc = rclc_subscription_init_default(
		&subscriber, &node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(sensor_msgs, msg, JointState),
		topic_sub_name);

	executor = rclc_executor_get_zero_initialized_executor();
	rc = rclc_executor_init(&executor, &support.context, 1, &allocator);

	rc = rclc_executor_add_subscription(
		&executor, &subscriber, &js_in,
		&motor_controller_cb, ON_NEW_DATA);

	//MSG data filling

	js_out.position.size=1;
	js_out.position.capacity=1;
	js_out.position.data = malloc(js_out.position.capacity*sizeof(double));
	js_out.position.data[0] = 0;
	js_out.velocity.size=1;
	js_out.velocity.capacity=1;
	js_out.velocity.data = malloc(js_out.velocity.capacity*sizeof(double));
	js_out.velocity.data[0] = 0;
	js_out.effort.size=1;
	js_out.effort.capacity=1;
	js_out.effort.data = malloc(js_out.effort.capacity*sizeof(double));
	js_out.effort.data[0] = 0;
	js_out.name.capacity = 1;
	js_out.name.size = 1;
	js_out.name.data = (std_msgs__msg__String*) malloc(js_out.name.capacity*sizeof(std_msgs__msg__String));
	js_out.name.data[0].data = joint_name;

	js_in.position.size=1;
	js_in.position.capacity=1;
	js_in.position.data = malloc(js_in.position.capacity*sizeof(double));
	js_in.position.data[0] = 0;
	js_in.velocity.size=1;
	js_in.velocity.capacity=1;
	js_in.velocity.data = malloc(js_in.velocity.capacity*sizeof(double));
	js_in.velocity.data[0] = 0;
	js_in.effort.size=1;
	js_in.effort.capacity=1;
	js_in.effort.data = malloc(js_in.effort.capacity*sizeof(double));
	js_in.effort.data[0] = 0;
	js_in.name.capacity = 1;
	js_in.name.size = 1;
	js_in.name.data = (std_msgs__msg__String*) malloc(js_in.name.capacity*sizeof(std_msgs__msg__String));

	/* Infinite loop */

	  for(;;)
	  {

		js_out.position.data[0] = kalman_angle;
		js_out.velocity.data[0] = real_velocity_to_go;
		js_out.effort.data[0] = prev_effort;

		rcl_ret_t ros_ret = rcl_publish(&publisher, &js_out, NULL);
	    if (ros_ret != RCL_RET_OK)
	    {
	      printf("Error publishing (line %d)\n", __LINE__);
	    }

	    rclc_executor_spin_some(&executor, 1);
	    vTaskDelay(1);
	  }

  /* USER CODE END StartDefaultTask */
}
```

SiddharthZokarkar, 2023-05-22:
Hello @Sorokonog ,

I tried your custom agent for CAN-BUS  , but I observed a strange behaviour.

When i am trying to connect the microROS subscriber , it is connecting without any issue and also subscribing to the topic. But after some time it stops subscribing the topic data. When I restart the subscriber it start working again. I need to restart the subscriber atleast 1-2 times. The data is also not much just 5 Bytes.
Are you also facing this issue ? Any suggestions to solve this issue?

Thank you in advance!!

Sorokonog, 2023-05-22:
> Hello @Sorokonog ,
>
> I tried your custom agent for CAN-BUS  , but I observed a strange behaviour.
>
> When i am trying to connect the microROS subscriber , it is connecting without any issue and also [...]

Could you describe the environment in more detail? What's on your stm side? And what kind of subscriber do you use (console type, or its a kind of a programm)?

SiddharthZokarkar, 2023-05-22:
Hello @Sorokonog ,

I am running microros on Linux . The subscriber is a EP32 client. The publisher is another ESP32 client. 
One ESP32 is publishing 5 char and another ESP32 is subscribing those 5 characters. 

I made the custom transport for ESP32 , and tried your custom agent .
The custom agent is working fine for both the publisher and the subscriber but after some time, susbcriber stops subscribing the topic data  whereas there is no issue with the publisher(it is working fine).

SiddharthZokarkar, 2023-06-08:
Hello @Sorokonog 

I use ros2 topic echo and found that it is not loosing any data but the micro-ROS subscriber(another ESP32)  is connecting and subscribing the data for some time and then stops subscribing. Need to restart again. 

If possible can you please share your custom client code for writing and reading. It would be a great help for me.

Thank you so much!!

Sorokonog, 2023-07-31:
> Hello @Sorokonog
>
> I use ros2 topic echo and found that it is not loosing any data but the micro-ROS subscriber(another ESP32) is connecting and subscribing the data for some time and then stops [...]

Hello @SiddharthZokarkar!
I'm using ros2 topic echo/pub for reading and writing so no special client for it.

Sorokonog, 2023-12-26:
Hello again! I'm reopening this issue in purpose of not creating new one with similar difficulties.
The system is almost the same so I have multiple clients on different STM32 connected via CANFD. For now I'm testing 2 of them with raspberry Pi as agent holder. 
So the question is - should 2 different clients (with different client keys and different can-ID) creates one agent's session?

In my case such behavior leads to hanging one of client with seg.fault in agent after some time. Could you share any suggestions regarding the situation. 

[image]
[image]

Sorokonog, 2023-12-30:
Well it was not obvious but i've missed changing of destination_endpoint... So I've added it instead of router_id
`fdframe.can_id = destination_endpoint->get_member<uint16_t>("ID") + 20;`
where `20` is just a number for passing through canfilter. And now its fine.
@SiddharthZokarkar make the same change in your code and seems it will solve your problem aswell.

closed 2024-01-01 by Sorokonog