micro-ROS/micro_ros_setup #579: Updating Zephyr to 3.1, host build issue.

gseqBE, 2022-09-18:
Hello,
I've been trying to up the zephyr version on arm from Zephyr 2.6 to 3.1. to sdk 0.14.2
I've been successful in testing the below on a nucleo_h743zi, however [zephyr host](https://micro.ros.org/docs/tutorials/core/zephyr_emulator/) seems to fail for me.
The fail is where idl is failing to find the cstddef include file. 

I'd like to fix this before issuing a PR. The docker setup in the reproduction has our repo with a branch for the sdk update.
Thank you!

- Hardware description: 
Nucleo H743zi and host modes.

- RTOS: 
Zephyr

- Installation type: 
Micro_ros_setup on Humble

- Version or commit hash: 

#### Steps to reproduce the issue

```
docker run -it --net=host -v /dev:/dev --privileged ros:humble
Once in, issue the following commands:
source /opt/ros/$ROS_DISTRO/setup.bash
mkdir microros_ws
cd microros_ws/
# Repo based on humble with Zephyr 3.1 and SDK 0.14.2 support
git clone -b gds/zephyr-rtos-3.1.0-sdk-0.15-update https://github.com/boston-engineering/micro_ros_setup src/micro_ros_setup
sudo apt update && rosdep update
rosdep install --from-paths src --ignore-src -y
sudo apt-get install python3-pip
colcon build
source install/local_setup.bash
ros2 run micro_ros_setup create_firmware_ws.sh zephyr host
ros2 run micro_ros_setup configure_firmware.sh ping_pong --transport udp --ip 127.0.0.1 --port 8888
ros2 run micro_ros_setup build_firmware.sh```

#### Expected behavior
build_firmware.sh succeeds

#### Actual behavior
Failure to find cstdef header 
`In file included from /microros_ws/firmware/mcu_ws/uros/rosidl_typesupport/rosidl_typesupport_c/src/type_support_dispatch.hpp:25,
                 from /microros_ws/firmware/mcu_ws/uros/rosidl_typesupport/rosidl_typesupport_c/src/message_type_support_dispatch.cpp:17:
/microros_ws/firmware/mcu_ws/uros/rosidl_typesupport/rosidl_typesupport_c/include/rosidl_typesupport_c/type_support_map.h:18:10: fatal error: cstddef: No such file or directory
   18 | #include <cstddef>
      |          ^~~~~~~~~
compilation terminated.
gmake[6]: *** [CMakeFiles/rosidl_typesupport_c.dir/build.make:93: CMakeFiles/rosidl_typesupport_c.dir/src/message_type_support_dispatch.cpp.o] Error 1
gmake[6]: *** Waiting for unfinished jobs....
In file included from /microros_ws/firmware/mcu_ws/uros/rosidl_typesupport/rosidl_typesupport_c/src/type_support_dispatch.hpp:25,
                 from /microros_ws/firmware/mcu_ws/uros/rosidl_typesupport/rosidl_typesupport_c/src/service_type_support_dispatch.cpp:17:
/microros_ws/firmware/mcu_ws/uros/rosidl_typesupport/rosidl_typesupport_c/include/rosidl_typesupport_c/type_support_map.h:18:10: fatal error: cstddef: No such file or directory
   18 | #include <cstddef>
      |          ^~~~~~~~~
compilation terminated.
gmake[6]: *** [CMakeFiles/rosidl_typesupport_c.dir/build.make:107: CMakeFiles/rosidl_typesupport_c.dir/src/service_type_support_dispatch.cpp.o] Error 1
gmake[5]: *** [CMakeFiles/Makefile2:223: CMakeFiles/rosidl_typesupport_c.dir/all] Error 2
gmake[5]: *** Waiting for unfinished jobs....
gmake[4]: *** [Makefile:149: all] Error 2
---
Failed   <<< rosidl_typesupport_c [4.11s, exited with code 2]
Aborted  <<< ros2trace [3.60s]                                                                                                              
Aborted  <<< rosidl_generator_dds_idl [3.37s]                                                                            
Aborted  <<< libyaml_vendor [6.97s]                                   

Summary: 22 packages finished [26.2s]
  1 package failed: rosidl_typesupport_c
  3 packages aborted: libyaml_vendor ros2trace rosidl_generator_dds_idl
  26 packages had stderr output: libyaml_vendor microcdr microxrcedds_client rcl_logging_interface rcl_logging_noop rcutils rmw rmw_implementation_cmake ros2trace rosidl_adapter rosidl_cli rosidl_cmake rosidl_generator_c rosidl_generator_dds_idl rosidl_parser rosidl_runtime_c rosidl_typesupport_c rosidl_typesupport_interface rosidl_typesupport_introspection_c rosidl_typesupport_microxrcedds_c test_interface_files tracetools tracetools_launch tracetools_read tracetools_test tracetools_trace
  41 packages not processed
gmake[3]: *** [Makefile:57: colcon_compile] Error 2
gmake[2]: *** [CMakeFiles/microroslib_project.dir/build.make:86: microros_extensions/src/microroslib_project-stamp/microroslib_project-build] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:2439: CMakeFiles/microroslib_project.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2`

#### Additional information

pablogs9, 2022-09-19:
Hello, the recommended way for using micro-ROS and Zephyr is using [micro-ROS Zephyr component](https://github.com/micro-ROS/micro_ros_zephyr_module). In this other repo, we are testing against Zephyr v3.1 a building successfully.

Can you take a look there?

gseqBE, 2022-09-20:
Thanks. Closing and and a few questions under micro_ros_zephyr_module.
FYI, it would be nice to have a blurb statement, under the [zephyr microros page](https://micro.ros.org/docs/tutorials/core/first_application_rtos/zephyr/), so it is known.

closed 2022-09-20 by gseqBE