micro-ROS/micro_ros_setup #580: menuconfig tool creates "flags.make:8: *** missing separator. Stop." error when building.

agentcaptain, 2022-09-19:
## Issue template

- Hardware description: ESP32 
- RTOS: FreeRTOS
- Installation type: micro_ros_setup, basic examples programmed through VM Ubuntu 22.04
- Version or commit hash: Humble ROS2

I've encountered an issue when trying to compile and configure a micro_ros example and would be very grateful for any help with this, I'm not so familiar with the issue reporting section on github so please excuse me if this post is a bit long winded. 

#### Steps to reproduce the issue
Fresh VM using Ubuntu 22.04 and ROS2 Humble, when following the basic micros ros tutorial found at https://micro.ros.org/blog/2020/08/27/esp32.

When running the command "ros2 run micro_ros_setup build_firmware.sh menuconfig" the following command "ros2 run micro_ros_setup build_firmware.sh" then produces the error found below. However, if the "ros2 run micro_ros_setup build_firmware.sh menuconfig" command is not executed and "ros2 run micro_ros_setup build_firmware.sh" is ran instead the example compiles without any issues.

In any case the menuconfig tool makes the following build process fail, even if the menu is exited without saving, this seems to be an issue even if I change micro controller and if serial is used rather than UDP. There is an error found in with flags.make every time, does anyone have any suggestions?

Many thanks!   

#### Expected behaviour
**Commands:**
**ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32
ros2 run micro_ros_setup configure_firmware.sh int32_publisher --transport serial #-t udp -i 192.168.1.120 -p 8888
ros2 run micro_ros_setup build_firmware.sh**

**Terminal output  (Working output without running menuconfig):** 

[100%] Linking CXX executable int32_publisher.elf
/usr/bin/cmake -E cmake_link_script CMakeFiles/int32_publisher.elf.dir/link.txt --verbose=1
/home/verk/Desktop/microros_ws/firmware/toolchain/espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -mlongcalls -Wno-frame-address CMakeFiles/int32_publisher.elf.dir/project_elf_src.c.obj -o int32_publisher.elf  esp-idf/xtensa/libxtensa.a esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/efuse/libefuse.a esp-idf/driver/libdriver.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/pthread/libpthread.a esp-idf/espcoredump/libespcoredump.a esp-idf/perfmon/libperfmon.a esp-idf/esp32/libesp32.a esp-idf/esp_common/libesp_common.a esp-idf/soc/libsoc.a esp-idf/esp_eth/libesp_eth.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_netif/libesp_netif.a esp-idf/esp_event/libesp_event.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/lwip/liblwip.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/freertos/libfreertos.a esp-idf/vfs/libvfs.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/asio/libasio.a esp-idf/cbor/libcbor.a esp-idf/coap/libcoap.a esp-idf/console/libconsole.a esp-idf/nghttp/libnghttp.a esp-idf/esp-tls/libesp-tls.a esp-idf/esp_adc_cal/libesp_adc_cal.a esp-idf/esp_gdbstub/libesp_gdbstub.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/esp_https_ota/libesp_https_ota.a esp-idf/protobuf-c/libprotobuf-c.a esp-idf/protocomm/libprotocomm.a esp-idf/mdns/libmdns.a esp-idf/esp_local_ctrl/libesp_local_ctrl.a esp-idf/sdmmc/libsdmmc.a esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a esp-idf/esp_websocket_client/libesp_websocket_client.a esp-idf/expat/libexpat.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/fatfs/libfatfs.a esp-idf/freemodbus/libfreemodbus.a esp-idf/jsmn/libjsmn.a esp-idf/json/libjson.a esp-idf/libsodium/liblibsodium.a esp-idf/mqtt/libmqtt.a esp-idf/openssl/libopenssl.a esp-idf/spiffs/libspiffs.a esp-idf/ulp/libulp.a esp-idf/unity/libunity.a esp-idf/wifi_provisioning/libwifi_provisioning.a esp-idf/main/libmain.a -Wl,--cref -Wl,--Map=/home/verk/Desktop/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build/int32_publisher.map -fno-rtti -fno-lto esp-idf/asio/libasio.a esp-idf/cbor/libcbor.a esp-idf/coap/libcoap.a esp-idf/esp_adc_cal/libesp_adc_cal.a esp-idf/esp_gdbstub/libesp_gdbstub.a esp-idf/esp_https_ota/libesp_https_ota.a esp-idf/esp_local_ctrl/libesp_local_ctrl.a esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a esp-idf/esp_websocket_client/libesp_websocket_client.a esp-idf/expat/libexpat.a esp-idf/fatfs/libfatfs.a esp-idf/sdmmc/libsdmmc.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/freemodbus/libfreemodbus.a esp-idf/jsmn/libjsmn.a esp-idf/libsodium/liblibsodium.a esp-idf/mqtt/libmqtt.a esp-idf/openssl/libopenssl.a esp-idf/spiffs/libspiffs.a esp-idf/unity/libunity.a esp-idf/wifi_provisioning/libwifi_provisioning.a esp-idf/protocomm/libprotocomm.a esp-idf/protobuf-c/libprotobuf-c.a esp-idf/mdns/libmdns.a esp-idf/console/libconsole.a esp-idf/json/libjson.a libmicroros.a esp-idf/xtensa/libxtensa.a esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/efuse/libefuse.a esp-idf/driver/libdriver.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/pthread/libpthread.a esp-idf/espcoredump/libespcoredump.a esp-idf/perfmon/libperfmon.a esp-idf/esp32/libesp32.a esp-idf/esp_common/libesp_common.a esp-idf/soc/libsoc.a esp-idf/esp_eth/libesp_eth.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_netif/libesp_netif.a esp-idf/esp_event/libesp_event.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/lwip/liblwip.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/freertos/libfreertos.a esp-idf/vfs/libvfs.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/nghttp/libnghttp.a esp-idf/esp-tls/libesp-tls.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/ulp/libulp.a esp-idf/mbedtls/mbedtls/library/libmbedtls.a esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a esp-idf/mbedtls/mbedtls/library/libmbedx509.a /home/verk/Desktop/microros_ws/firmware/toolchain/esp-idf/components/esp_wifi/lib/esp32/libcoexist.a /home/verk/Desktop/microros_ws/firmware/toolchain/esp-idf/components/esp_wifi/lib/esp32/libcore.a /home/verk/Desktop/microros_ws/firmware/toolchain/esp-idf/components/esp_wifi/lib/esp32/libespnow.a /home/verk/Desktop/microros_ws/firmware/toolchain/esp-idf/components/esp_wifi/lib/esp32/libmesh.a /home/verk/Desktop/microros_ws/firmware/toolchain/esp-idf/components/esp_wifi/lib/esp32/libnet80211.a /home/verk/Desktop/microros_ws/firmware/toolchain/esp-idf/components/esp_wifi/lib/esp32/libpp.a /home/verk/Desktop/microros_ws/firmware/toolchain/esp-idf/components/esp_wifi/lib/esp32/librtc.a /home/verk/Desktop/microros_ws/firmware/toolchain/esp-idf/components/esp_wifi/lib/esp32/libsmartconfig.a /home/verk/Desktop/microros_ws/firmware/toolchain/esp-idf/components/esp_wifi/lib/esp32/libphy.a esp-idf/xtensa/libxtensa.a esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/efuse/libefuse.a esp-idf/driver/libdriver.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/pthread/libpthread.a esp-idf/espcoredump/libespcoredump.a esp-idf/perfmon/libperfmon.a esp-idf/esp32/libesp32.a esp-idf/esp_common/libesp_common.a esp-idf/soc/libsoc.a esp-idf/esp_eth/libesp_eth.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_netif/libesp_netif.a esp-idf/esp_event/libesp_event.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/lwip/liblwip.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/freertos/libfreertos.a esp-idf/vfs/libvfs.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/nghttp/libnghttp.a esp-idf/esp-tls/libesp-tls.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/ulp/libulp.a esp-idf/mbedtls/mbedtls/library/libmbedtls.a esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a esp-idf/mbedtls/mbedtls/library/libmbedx509.a /home/verk/Desktop/microros_ws/firmware/toolchain/esp-idf/components/esp_wifi/lib/esp32/libcoexist.a /home/verk/Desktop/microros_ws/firmware/toolchain/esp-idf/components/esp_wifi/lib/esp32/libcore.a /home/verk [... cut ...]
echo 'source /home/verk/Desktop/microros_ws/install/local_setup.bash' >> ~/.bashrc

#Install USB drivers for VM
sudo apt-get install minicom -y
sudo apt install build-essential flex bison libssl-dev libelf-dev dwarves libncurses-dev -y

sudo usermod -a -G dialout verk
sudo usermod -a -G tty verk

ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32

sudo apt update

x-www-browser https://micro.ros.org/blog/2020/08/27/esp32/ &
x-www-browser https://micro.ros.org/docs/tutorials/core/first_application_rtos/freertos/ &

pablogs9, 2022-09-20:
Please use the ESP IDF component for micro-ROS: https://github.com/micro-ROS/micro_ros_espidf_component

agentcaptain, 2022-09-20:
> Please use the ESP IDF component for micro-ROS: https://github.com/micro-ROS/micro_ros_espidf_component

Thanks Pablo I'll try that method.

formigola90, 2022-11-09:
Hi @agentcaptain and @pablogs9. I encountered the same problem, but I think I found a hint that can help to solve the issue.
I followed [this](https://medium.com/@SameerT009/connect-esp32-to-ros2-foxy-5f06e0cc64df) tutorial trying to flash the int32_publisher to a ESP32 board. As described by @agentcaptain, after executing the _menuconfig_ command in order to change the wifi ssid and password, the build command failed. I did a little testing and noticed that if the WIFI SSID contains the character "-"  the build fails (my wifi ssid contains it), without it works. I don't know if there are other critical characters but it seems an encoding issue. Hope that this can help.

clive0106, 2023-01-30:
Has anyone worked out a fix for this issue?

I have the same problem... grepping the build output for 'Error' gives me..

  ros2 run micro_ros_setup build_firmware.sh | grep Error
  WARNING: using an unsupported version of tool cmake found in PATH: 3.22.1
  esp-idf/main/CMakeFiles/__idf_main.dir/flags.make:8: *** missing separator.  Stop.
  make[1]: *** [CMakeFiles/Makefile2:3835: esp-idf/main/CMakeFiles/__idf_main.dir/all] Error 2
  make: *** [Makefile:139: all] Error 2

Could it possibly be a problem with the version of cmake?

Also, if running menuconfig is triggering the problem, is there another way to set up the wifi ssid and password after the firmare is built?

clive0106, 2023-01-31:
Whatever process is creating the flags.make file seems to be corrupting it.. I am finding messages like 'Make[1]: Entering directory' in the file :(
[image]

clive0106, 2023-01-31:
These are the two files being corrupted by the make[1] messages..
./firmware/freertos_apps/microros_esp32_extensions/build/esp-idf/main/CMakeFiles/__idf_main.dir/flags.make
./firmware/freertos_apps/microros_esp32_extensions/build/CMakeFiles/int32_publisher.elf.dir/flags.make
Once these files have been hand edited, I appear to be able to build the firmware :)

pablogs9, 2023-01-31:
@clive0106 could you detail here your modification ?

clive0106, 2023-01-31:
I have a work around for this problem.
Two files are being corrupted by the build process, they are..

./firmware/freertos_apps/microros_esp32_extensions/build/esp-idf/main/CMakeFiles/__idf_main.dir/flags.make
./firmware/freertos_apps/microros_esp32_extensions/build/CMakeFiles/int32_publisher.elf.dir/flags.make

Extra text is being inserted into the files and it is breaking the firmware build.
So..
Before building the firmware using "ros2 run micro_ros_setup build_firmware.sh"
These two files need editing.. I am using sublime text and have word wrap enabled for ease of viewing.
This example is editing ./firmware/freertos_apps/microros_esp32_extensions/build/CMakeFiles/int32_publisher.elf.dir/flags.make though you will need to edit both of them,
This is what part of the corrupted file looks like
[image]
Then search for make[1], and in my case, the corrupt text is..
make[1]: Entering directory '/home/clive/ros/microros_ws/firmware/freertos_apps/microros_esp32_extensions'
Delete this text, it should now look like..
[image]
There is a second corruption in the file, here..
[image]
And here is the fixed version
[image]
Save the edited file, then fix the second file.
Then, you should be good to build the firmware using "ros2 run micro_ros_setup build_firmware.sh"
A proper fix would be for someone who understands how these files are generated, to stop the corrupt text being written in the first place, but that is not me :)
Hope this helps

Caian, 2023-02-24:
> I have a work around for this problem. Two files are being corrupted by the build process, they are.. [...]

Also happening to me and this fixed the issue, thanks!

I don't know if it is related to ROS version, but I'm working inside docker, image `osrf/ros:humble-desktop-full`

clive0106, 2023-02-27:
> Also happening to me and this fixed the issue, thanks!
> 
> I don't know if it is related to ROS version, but I'm working inside docker, image `osrf/ros:humble-desktop-full`

I dont think that fixing the problem should be that hard, I will have a look once I fix my current problem :)

clive0106, 2023-05-12:
@lonebots What is the error message that you are getting?

lonebots, 2023-05-12:
hey @clive0106,  
I was following the exact same tutorial as mentioned in this issue, the error occurred after setting the **menuconfig** by running `ros2 run micro_ros_setup build_firmware.sh menuconfig` when this was set the following command to build to the firmware `ros2 run micro_ros_setup build_firmware.sh` gave me the following error in **flags.make** file line number **8** 

```bash
cd /home/jishnu/testros/workspace/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build/esp-idf/wifi_provisioning && /home/jishnu/testros/workspace/microros_ws/firmware/toolchain/espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -I/home/jishnu/testros/workspace/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build/config -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/wifi_provisioning/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/wifi_provisioning/src -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/wifi_provisioning/proto-c -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/protocomm/proto-c -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/newlib/platform_include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/freertos/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/heap/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/log/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/lwip/include/apps -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/lwip/include/apps/sntp -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/lwip/lwip/src/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/lwip/port/esp32/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/lwip/port/esp32/include/arch -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/soc/esp32/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/soc/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/esp_rom/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/esp_common/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/xtensa/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/xtensa/esp32/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/esp32/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/driver/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/driver/esp32/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/esp_ringbuf/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/efuse/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/efuse/esp32/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/vfs/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/esp_wifi/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/esp_wifi/esp32/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/esp_event/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/esp_netif/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/esp_eth/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/tcpip_adapter/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/app_trace/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/protocomm/include/common -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/protocomm/include/security -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/protocomm/include/transports -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/nvs_flash/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/spi_flash/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/mbedtls/port/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/mbedtls/mbedtls/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/protobuf-c/protobuf-c -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/mdns/include -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/console -I/home/jishnu/testros/workspace/microros_ws/firmware/toolchain/esp-idf/components/json/cJSON -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -W [... cut ...]
make[1]: Leaving directory '/home/jishnu/testros/workspace/microros_ws/firmware/freertos_apps/microros_esp32_extensions'
"

C_FLAGS = -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.1\" -DESP_PLATFORM

```

* I tried removing it and rerun the build command but for me the error still prevails.

clive0106, 2023-05-12:
There are multiple instances 
You need to remove the 
make[1]: Entering directory '/home/jishnu/testros/workspace/microros_ws/firmware/freertos_apps/microros_esp32_extensions'
AND the 
make[1]: Leaving directory '/home/jishnu/testros/workspace/microros_ws/firmware/freertos_apps/microros_esp32_extensions'
And sometimes there are multiple files to fix..  It does work :)

lonebots, 2023-05-12:
Yeah @clive0106, I have tried running the build after removing these two lines from the flags.make file, but I don't know it didn't fix my issue. :(

clive0106, 2023-05-12:
Dont worry, we will get there.

Firstly, these were the files that I had to fix..
./firmware/freertos_apps/microros_esp32_extensions/build/esp-idf/main/CMakeFiles/__idf_main.dir/flags.make
./firmware/freertos_apps/microros_esp32_extensions/build/CMakeFiles/clive05_robot1.elf.dir/flags.make

Secondly, dont run anything other than
ros2 run micro_ros_setup build_firmware.sh

Thirdly, only remove the text in error, plus its newline.
This should be a fixed version of the text you pasted in above

[fixed.txt](https://github.com/micro-ROS/micro_ros_setup/files/11463713/fixed.txt)

lonebots, 2023-05-12:
Thank you so much @clive0106 ❤️ 
It was a mistake from my side that I didn't remote the additional lines while removing the problematic text. 
Now, it's perfectly completing the build. 😀