micro-ROS/micro_ros_setup #582: Demo crashes with RMW_IMPLEMENTATION set

Gleis1234, 2022-10-10:
## Issue template

- Hardware description: ACER TravelmateP with Ryzen Pro 5
- RTOS: Arduino Due
- Installation type: docker micro_ros_setup.git following demo
- Version or commit hash: 3.1.2

#### Steps to reproduce the issue
- ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0
- ros2 run micro_ros_demos_rclc string_subscriber

#### Expected behavior

#### Actual behavior
Topic /string_publisher is created; node prints messages
#### Additional information
When env. variable RMW_IMPLEMENTATION is not set, expected behavior.   
When env. variable RMW_IMPLEMENTATION=rmw_microxrcedds is set, app crashes with   
[ERROR] [1665421183.073143570] [rclc]: [rclc_init] Error in rcl_init: failed to resolve symbol 'rmw_feature_supported' in shared library '/home/guenther/microros_ws/install/rmw_microxrcedds/lib/librmw_microxrcedds.so', at ./src/functions.cpp:152, at ./src/rcl/init.c:218

pablogs9, 2022-10-11:
The missing symbol should be fixed here https://github.com/micro-ROS/rmw_microxrcedds/pull/267 thanks for the report.

But in any case, how are you connecting a micro-ROS node running in Linux with a micro-ROS Agent using a serial port?
The default `micro_ros_demos_rclc` package uses UDP try to run the agent with:

```
ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v6
```

Gleis1234, 2022-10-11:
@pablogs9 thank you for the answer. My configuration is as follows: ROS2 with Moveit running on a notebook with Linux. To drive some hardware I use an Arduino which is connected by USB. So my idea, following the demo docs, is to run the micro-ros agent on Linux with args serial and device address and a micro_ros client on the arduino.
Meanwhile it seems that my RMW installation has been broken somehow. Is there a way to repeat the RMW installation again?

pablogs9, 2022-10-11:
But where are you running the micro-ROS client right now in Arduino or in your Linux machine?

> Is there a way to repeat the RMW installation again?
Clean the environment and replicate all the steps

Gleis1234, 2022-10-11:
I have started the string_subscriber app on linux and have connected the arduino to have the USB-port available. On the arduino is the string_publisher app installed. But for testing I can send messages to the subscriber with ros2 topic pub. This works, when RMW_IMPLEMENTATION is not set. Otherwise the app crashes.
I will now try to repeat the whole micro_ros_setup.

pablogs9, 2022-10-11:
If you use `ros2 run micro_ros_demos_rclc string_subscriber` you are running **a micro-ROS app in Linux using UDP** not in Arduino using serial.

Gleis1234, 2022-10-11:
Have I to use a regular ros2 node as counterpart to the string_publisher? In the ping_pong demo it looks like I can use the same app for both sides.

pablogs9, 2022-10-11:
Sorry I cannot understand your use case.

This CMake Warning is not relevant you can just ignore it

Gleis1234, 2022-10-11:
It's really simple. Exchange messages between host and arduino. The procedure is described in demo "Teensy with arduino".
Do you have any example for connection client nodes on arduino with nodes on linux?

pablogs9, 2022-10-11:
Ok, to exchange messages between the host and Arduino you just need to have a micro-ROS firmware loaded in your Arduino, then run the micro-ROS Agent and use ROS 2 CLI to pub/sub.

You **do not need** to run `ros2 run micro_ros_demos_rclc string_subscriber`, instead you will need to do something like `ros2 topic echo ...`

Gleis1234, 2022-10-11:
That's what I tried. See above. ros2 node list fails with: failed to initialize rcl: error not set, at ./src/rcl/init.c:218.

pablogs9, 2022-10-11:
You do not need to set `RMW_IMPLEMENTATION=rmw_microxrcedds` when using `ros2 node list` commands

Gleis1234, 2022-10-11:
When do I need the RMW environment? I thought I need it for communication over serial line. Now I have an publisher running on arduino and an agent on linux. ros2 topic list does not show any topics, except standards.

pablogs9, 2022-10-11:
`rmw_microxrcedds` is the RMW implementation for Micro XRCE-DDS, the micro-ROS middleware. It runs on your microcontroller.

The micro-ROS Agent will translate between Micro XRCE-DDS and DDS.

DDS is the middleware used in ROS 2 so micro-ROS Agent will interface between micro-ROS and ROS middlewares.

**Please read the documentation: https://micro.ros.org/**

If you run the agent with the flag `-v6` you will see the communication log between the micro-ROS Agent and the Arduino, what are you seeing there?

Gleis1234, 2022-10-11:
I loaded up the publisher demo to my arduino, started the agent with Option -v6, but nothing is to be seen.
I build the arduino app with micro_ros_platformio without any errors.

pablogs9, 2022-10-11:
If the agent is not showing anything probably you are not connecting the board correctly, not setting the correct serial port, or not building/flashing correctly the firmware.

pablogs9, 2022-10-11:
debug the serial port to check if micro-ROS is sending bytes

Gleis1234, 2022-10-11:
Now I got an idea that I always was on the wrong way with the arduino app. I tried to take the demo app string_publisher and compiled it for the arduino hoping that micro_ros will transform it to the setup(), loop() structure of arduino.
Now I modified the app to the arduino structure. But there is one compilation error left in a micro_ros header file.
In file included from /home/guenther/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/stdlib.h:11:0,
                 from /home/guenther/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/7.2.1/cstdlib:75,
                 from /home/guenther/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/7.2.1/stdlib.h:36,
                 from /home/guenther/.platformio/packages/framework-arduino-sam/cores/arduino/Arduino.h:24,
                 from src/main.cpp:1:
/home/guenther/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/sys/unistd.h:147:9: error: conflicting declaration of C function 'int link(const char*, const char*)'
 int     _EXFUN(link, (const char *__path1, const char *__path2 ));

Who can help me with that.

pablogs9, 2022-10-11:
Sorry but this is out of the scope of micro-ROS. Do not hesitate to reopen if you have any other micro-ROS questions.

closed 2022-10-11 by pablogs9