micro-ROS/micro_ros_setup #584: Shared memory doesn't work with more than one sub/pub pair

Ruby-Phonics, 2022-10-14:
## Issue template

- Hardware description: ESP32S3
- RTOS: FreeRTOS
- Installation type: git cloned micro_ros_espidf_component and built with esp-idf
- Version or commit hash: 2c7c6ce (of micro_ros_espidf_component)

#### Steps to reproduce the issue
Using the following code and build flags
main.cpp:
`#include <stdio.h>
#include <string.h>
#include <unistd.h>
#include <math.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "esp_system.h"

#include <uros_network_interfaces.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#ifdef CONFIG_MICRO_ROS_ESP_XRCE_DDS_MIDDLEWARE
#include <rmw_microros/rmw_microros.h>
#endif

#include <micro_ros_utilities/type_utilities.h>
#include <micro_ros_utilities/string_utilities.h>

static const char *TAG = "main";

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Aborting.\n",__LINE__,(int)temp_rc);vTaskDelete(NULL);}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Continuing.\n",__LINE__,(int)temp_rc);}}

//--------------- Multithread pub/sub example ----------------------------
#include <std_msgs/msg/int32.h>
rcl_publisher_t publisher_1;
rcl_publisher_t publisher_2;
rcl_subscription_t subscriber_1;
std_msgs__msg__Int32 recvmsg;
std_msgs__msg__Int32 msg_tim;
void subscription1_callback(const void* msgin){        
        if(msgin != NULL){
                const std_msgs__msg__Int32 * msg = (const std_msgs__msg__Int32 *)msgin;
                printf("Received: %d\n",msg->data);}
        else{printf("callback Null \n");}
}
void thread_1(void * arg)
{
        std_msgs__msg__Int32 msg;
        msg.data = 0;
        while(1){
                RCSOFTCHECK(rcl_publish(&publisher_1, &msg, NULL));
                msg.data++;
                vTaskDelay(1000 / portTICK_PERIOD_MS);
        }
}

void timer_cb(rcl_timer_t * timer, int64_t last_call_time){
        //dummy timer
}
void thread_2(void * arg)
{
        std_msgs__msg__Int32 msg;
        msg.data = 0;
        while(1){
[... 166 lines cut ...]
                "-DERTPS_MAX_PUBLISHERS=2",
                "-DERTPS_MAX_SUBSCRIPTIONS=2",
                "-DERTPS_MAX_SERVICES=1",
                "-DERTPS_MAX_CLIENTS=1",
                "-DERTPS_MAX_HISTORY=10"
            ]
        },
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF"
            ]
        }
    }`

Using with shared memory set to off work as expected and it also works if only using one publisher.

#### Expected behavior
I expect to see published data from publisher1 printed in the monitor by the subscriber callback
#### Actual behavior
It fails silently without any prints or crash to the monitor.
#### Additional information
I have been debugging this issue for the past week and I can't seem to move forward. I have managed to get one publisher/subscriber pair to listen to work locally with shared memory but as soon as I add another subscriber or publisher (or a pair of them) I dont get any prints at all. I have tried several different things and verified that it works as expected when I use it with shared memory disabled and the communication is done via the agent, but as soon as I compile with shared memory feature enabled things stop working. 

I have played around with different priorities of the tasks as well as different QoS settings but I'm not getting anywhere. In the code above I had to add a dummy timer to my executor cause the subscriber callback wouldnt execute otherwise (despite being set to "ALWAYS").

Acuadros95, 2022-10-14:
Just to be clear: You are using the provided code and build options and it fails with the option `UCLIENT_PROFILE_SHARED_MEMORY` enabled?

We will take a look at this.

Ruby-Phonics, 2022-10-14:
Yes, I have tried multiple different combinations of pubs/subs, number of nodes etc and I always run into issues with that option enabled. I am happy to provide more information you might need, I have been dealing with this issue for the past week now so I have tried more things than are reasonable to put in one post.

Acuadros95, 2022-10-14:
You need to increase the `UCLIENT_SHARED_MEMORY_MAX_ENTITIES` parameter, as each publisher and subscriber creates a topic and a datawriter or datareader internally, so you need two shared memory entities for each one. 

For example, you can test two publishers and one subscriber with `UCLIENT_SHARED_MEMORY_MAX_ENTITIES=6`