micro-ROS/micro_ros_setup #586: micro-ROS agent stops on logger setup line

MarcinMlodzinski, 2022-11-02:
- Hardware description:  micro-ros node: STM32L4 Discovery kit IoT node (B-L475E-IOT01A2), agent: Raspberry Pi 4 Model B 2GB RAM
- RTOS: Zephyr
- Installation type: micro_ros_setup
- Version or commit hash: humble

#### Steps to reproduce the issue

1. Configure firmware to use ping-pong demo application and usb-serial transport.
2. Build and flash.
3. Plug microcontroller into Raspberry and run micro-ros agent.

#### Expected behavior
Micro-ros node starts properly and its topics are visible from Raspberry.

#### Actual behavior
Micro-ros agent stops on:
```
[1667242557.691466] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1667242557.692280] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
```

#### Additional information
I tried to unplug, then plug in the board and resetting it with dedicated pushbutton.
Raspberry runs ROS2 Humble on Ubuntu 22.04 server with kernel:
`Linux ubuntu 5.15.39-rt42-raspi #1 SMP PREEMPT_RT Sat Jul 16 08:34:58 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux`

pablogs9, 2022-11-03:
Could you provide which command are you using for instantiating the agent?

Also, have you checked if the STM32 is sending data through the USB-Serial?

MarcinMlodzinski, 2022-11-03:
> Could you provide which command are you using for instantiating the agent?

`ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-STMicroelectronics_STM32_STLink_066EFF514852897267173214-if02`
where /dev/serial/by-id/usb-STMicroelectronics_STM32_STLink_066EFF514852897267173214-if02 is present when I plug in the microcontroller.

> Also, have you checked if the STM32 is sending data through the USB-Serial?

command: `screen /dev/serial/by-id/usb-STMicroelectronics_STM32_STLink_066EFF514852897267173214-if02` gives nothing

In addition I'd say that with ESP32 on freertos everything works fine.

pablogs9, 2022-11-03:
It seems that you are not using the correct USB port. This board has two USBs, the one that you are connecting is the STLink interface that bridges a UART of the main MCU. Maybe micro-ROS is communicating using the USB-CDC available in the other port. It depends on your transport configuration.

Can you check that?

MarcinMlodzinski, 2022-11-03:
You're right - that board has two USB ports: one marked as USB STLINK (which i plugged into RPi) and another marked as USB OTG. Ubuntu doesn't see the device if it is connected to the USB OTG port (using command `sudo dmesg | grep tty`, which shows something if I connect microcontroller with USB STLINK). If I should check something in micro-ros configs, I need specific instructions where to look for it.

pablogs9, 2022-11-03:
You can switch between transports using the configure step: https://github.com/micro-ROS/micro_ros_setup/blob/cdd082b8152a8931ddab5adee33da036ce158fd1/config/zephyr/generic/configure.sh#L27-L34

BTW are you adjusting the power supply jumper when changing the USB or you are just connecting both of them?

In any case, I recommend you using the Zephyr module for micro-ROS https://github.com/micro-ROS/micro_ros_zephyr_module

MarcinMlodzinski, 2022-11-03:
> are you adjusting the power supply jumper when changing the USB or you are just connecting both of them?

I've tried both methods and some green diodes were flashing during those attempts to see the microcontroller from ubuntu, so I assume it was powered correctly.

MarcinMlodzinski, 2022-11-04:
I finally managed to make that micro-ROS Zephyr module flashing the board. Sadly it works in the same way as micro_ros_setup did. Do you have any other suggestions?

pablogs9, 2022-11-04:
Check your cables, check if you are able to send some data using the serial port...

Chick92, 2023-03-16:
I'm encountering the same problem. 
Hardware is RP2040 based, following these instructions - https://ubuntu.com/blog/getting-started-with-micro-ros-on-raspberry-pi-pico

I can get it to come up if i reset the board, or un plug and re plug the board in before starting the micro ros agent.

My target application does have the ability to power cycle the pico, so it's not a massive issue, but is problematic for other users that don't have this ability. 

Please advice as to what debug output etc you require to assist.

pablogs9, 2023-03-16:
@Chick92 you can handle reconnectiosn in micro-ROS code:
https://docs.vulcanexus.org/en/latest/rst/tutorials/micro/handle_reconnections/handle_reconnections.html
https://github.com/micro-ROS/micro_ros_arduino/blob/humble/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino

ashemah, 2024-01-22:
@Chick92 I think you might be running into the RPi Pico Errata E5 where it doesn't init the usb of the pico properly in certain circumstances. 

https://news.ycombinator.com/item?id=30354772