micro-ROS/micro_ros_setup #587: Agent not connecting on Docker 20.10.21

gseqBE, 2022-11-03:
## Issue template

- Hardware description: 
**Nucleo STM32H743zi**
- RTOS: 
**Zephyr 2.6**
- Installation type: 
**Micro_ros_setup, micro_ros_agent**
- Version or commit hash: 
**Humble**

#### Steps to reproduce the issue

- Start agent: `docker run -it --rm --net=host  microros/micro-ros-agent:humble udp4 --port 8888  -v6`
- Change IP of edge in microros_transports.h  to agent IP
- Edge shows waiting for connection.
- Wireshark shows edge requesting a connection to agent
[image]
- Agent shows:
[image]

#### Expected behavior
Agent should accept and connect to edge.

#### Actual behavior
Agent does not connect to edge.

#### Additional information

pablogs9, 2022-11-04:
Is 192.168.1.213 the IP of your agent?

This seems to be a network problem, not a micro-ROS problem since your micro-ROS client is sending packages correctly but they seems to not reach the agent.

rcolatobe, 2022-11-04:
Tested configuration on a different computer and could not reproduce the actual behavior - will continue to debug networking side.

closed 2022-11-04 by gseqBE