micro-ROS/micro_ros_setup #592: spdlog error when running create_agent.sh in dustynv/ros container.

R-Liebert, 2022-11-14:
## Issue

- Hardware description: NVIDIA Jetson Nano 2GB
- RTOS: Arduino Nano RP2040 Connect
- Installation type: micro_ros_setup
- Version or commit hash: foxy

#### Steps to reproduce the issue
Open docker container with dustynv/ros:galactic-ros-base-l4t-r32.7.1 then follow steps for micro_ros_setup until your ready to run build_agent.sh.

#### Expected behavior
Agent to be built.
 
#### Actual behavior
Error requesting spdlog=1.9.2

#### Additional information
I went into the cmake directories mentioned in the error, removed spdlog directories and got the latest ones. Now it can't find spdlogConfig.cmake. 

I've made a docker image of the point where it stops rliebert/ros2foxy_uros:brokenspdlog.

If there is any other way convenient way of getting ROS2 Foxy with micro-ros going on the Jetson Nano I would appreciate being pointed in the right direction.

pablogs9, 2022-11-15:
Is possible to replicate this issue using `ros:galactic` docker?

R-Liebert, 2022-11-17:
I follow the same routine for galactic(using container for nvidia jetson nano as base) and it's not the same error, but still failing to build the agent. Now the issue is similar to micrp_ros_agent issue #50, so I assume it can be solved with the flag. I'll close this request.

closed 2022-11-17 by R-Liebert