micro-ROS/micro_ros_setup #597: micro-ros-agent: [ros2run]: Segmentation fault when trying to set

Ecuashungo, 2022-11-25:
## Segmentation Fault in micro-ros-agent

- Hardware description: The micro-ros-agent was compiled on a raspberry pi CM4 where ROS2 Humble is installed.
- RTOS: I'm trying to setup a connection between the raspberry pi and a PX4 flight controller running NuttX
- Installation type: I have used the [current main branch](https://github.com/PX4/PX4-Autopilot) of the px4-firmware and confirmed that the microdds_client works (I have connected my development computer running ubuntu 20.04 and ros2 galactic and ran the same ros2 testing nodes using the same serial connection). 
- Version or commit hash: ros2 humble

#### Steps to reproduce the issue
I have compiled the micro-ros-agent according to the instructions found [here](https://github.com/micro-ROS/micro_ros_setup#building-micro-ros-agent). When running the agent it starts up just fine and as soon as the client side is launched I get the following error:

```
ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyAMA1 -b 921600 -v 6 
[1669365763.761759] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1669365763.762643] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1669365794.016024] info     | Root.cpp           | create_client            | create                 | client_key: 0xAAAABBBB, session_id: 0x81
[1669365794.017208] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0xAAAABBBB, address: 1
[1669365794.017602] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAAAABBBB, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1669365794.120910] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAAAABBBB, len: 108, data: 
0000: 81 80 00 00 01 05 62 00 00 0A 00 11 01 02 00 00 53 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 70 78 34 5F 6D 69 63 72 6F 5F 78 72 63 65
0040: 5F 64 64 73 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63 69 70 61 6E 74 3E
0060: 3C 2F 64 64 73 3E 00 00 00 00 00 00
[1669365794.121227] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAAAABBBB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[ros2run]: Segmentation fault
```
On the client side in the NuttX shell I get the following error:
```
ERROR [microdds_client] create entities failed: participant: 255
```

I have tested the client side with other agents where it worked like charm, so I believe that the problem is the agent running on my raspberry pi CM4. What are steps I can take to debug this problem?

pablogs9, 2022-11-25:
Could you provide instructions for replicating this issue?

Ecuashungo, 2022-11-25:
On the Raspberry pi I ran those commands in my ros2 workspace after having sourced the main ros2 installation (humble).
```
git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git
rosdep update && rosdep install --from-paths src --ignore-src -y
colcon build
source install/local_setup.bash 
ros2 run micro_ros_setup create_agent_ws.sh 
ros2 run micro_ros_setup build_agent.sh
source install/local_setup.bash 
ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyAMA1 -b 921600 -v 6
```

on the px4 side I have executed the f[ollowing command](https://docs.px4.io/main/en/modules/modules_system.html#microdds-client) in the NuttX-shell: 
```
microdds_client start -t serial -d /dev/ttyS2 -b 921600
```
And the two devices are connected through a uart connection using 3 cables: GND, TX and RX.

The serial connection works, I can access the data through `screen`.

If you need more detailed information or logfiles (do they exist?) I'm happy to share those as well.

pablogs9, 2022-11-28:
Sorry but I'm sure which library is using "microdds_client". Can you ensure that it is using the [latest version of Micro XRCE-DDS](https://github.com/eProsima/Micro-XRCE-DDS-Client/releases)?

Also, why are you using micro-ROS Agent and not Micro XRCE-DDS Agent? They are basically the same, but just curious.

Ecuashungo, 2022-11-28:
It is using the latest version of Micro XRCE-DDS ([this commit of the px4 fork](https://github.com/PX4/Micro-XRCE-DDS-Client/commit/4248559f3b111155c783e524e461ccc83e768103)).

Thanks for pointing out the Micro XRCE-DDS Agent. I was not aware that there is a difference between the two. I have now tried to setup everything with the Micro XRCE-DDS Agent (v2.2.1) and it works. 
The only problem is that I get a warning (`WARN  [timesync] RTT too high for timesync: 17 ms
`) on the px4/client side but I don't think it has anything to do with the micro XRCE-DDS (issue is being discussed [here](https://github.com/PX4/PX4-Autopilot/issues/20634)).

beniaminopozzan, 2022-11-28:
@Ecuashungo , could you try (I don't have the hardware to do that myself) to modify `SuperBuild.cmake` of micro-ROS-agent in order to pull the `master` branch of Micro-XRCE-DDS-Agent of  instead of the `ros2` one?
https://github.com/micro-ROS/micro-ROS-Agent/blob/0e41ea652238bd0123f6249d69b533d282882dd6/micro_ros_agent/cmake/SuperBuild.cmake#L28-L29