micro-ROS/micro_ros_setup #600: using micro-ROS libraries in Vitis on MicroBlaze platform

kislevro, 2022-12-14:
Hello everyone,
I am using freeRTOS on MicroBlaze platform and I want to run micro-ROS on top of the freeRTOS.
I saw documentations on integrating micro-ROS with different firmware, but I wanted to use Vitis to include those micro-ROS libraries.

I tried to [create micro-ROS app on freeRTOS](https://micro.ros.org/docs/tutorials/core/first_application_rtos/freertos/) by [creating MicroBlaze custom micro-ROS library](https://micro.ros.org/docs/tutorials/advanced/create_custom_static_library/), but I got stuck in the CMake and meta files because I wanted Vitis to handle it.

 How can I use micro-ROS in freeRTOS based project on Vitis?

Thank you!

pablogs9, 2022-12-14:
Hello, the latest documentation on how to port micro-ROS is: https://docs.vulcanexus.org/en/latest/rst/tutorials/micro/custom_platforms/custom_platforms.html

If you need professional support please contact: support@eProsima.com

closed 2022-12-14 by pablogs9

kislevro, 2022-12-14:
Hello @pablogs9, I already saw this tutorial, but I didn't understand how to integrate micro-ROS with Xilinx's Vitis.

I am a beginner in micro-ROS and Vitis so maybe I failed to link the tutorial. I would appreciate if you could focus me on the context between the micro-ROS and Vitis in the tutorial.

pablogs9, 2022-12-14:
Sorry, but we do not support Vitis now. If you need us to check the possible integrations and evaluate this port please contact: [support@eProsima.com](mailto:support@eProsima.com)

kislevro, 2022-12-14:
Thank you @pablogs9 for the reply. 
I saw several articles that suggest FPGA-ROS integration, in particular on Xilinx processors (that Vitis is the software development) [[1](https://arxiv.org/pdf/2205.03929.pdf)].

How can I adapt the FPGA integration with ROS to micro-ROS?