micro-ROS/micro_ros_setup #601: Micro Ros agent build stuck at 9%

AhmadSameh, 2022-12-22:
Hardware: Raspberry pi 3
Os: Ubuntu 22.04 server
Ros-distro: humble

Hello,
I have a problem trying to build the agent.
When I run `micro_ros_setup` build_agent.sh
The msgs package is installed without any problems.
However the agent gets stuck at 9%.
There's a message that shows up during the building process saying 
'''
INFO: task kworker/1:2:120 blocker for more than x seconds.
           Tainted: G                  C               5.15.0-1021-raspi #23-Ubuntu
'''
I would appreciate the help as this is urgent and is delaying my work.

pablogs9, 2022-12-22:
Could you check your system memory usage during build using `top` or `htop`?

AhmadSameh, 2022-12-22:
@pablogs9 
The memory is almost full with 833M/905M
Does that mean that it's a hardware problem and I should use a different raspberry pi with a higher ram?

pablogs9, 2022-12-22:
This means that your hardware is not able to build the package because of a lack of RAM memory. Try configuring colcon for using fewer resources. Or try to building this package manually: https://github.com/micro-ROS/micro-ROS-Agent

AhmadSameh, 2022-12-22:
@pablogs9 thanks for the help, i just tested it on a device with higher memory and it now works.

closed 2022-12-22 by pablogs9