micro-ROS/micro_ros_setup #602: MicroROS fails to build on Jetson Nano

usaotearoa, 2023-01-05:
## Issue template

- Hardware description: Jetson Nnao 4GB (B01) run in docker from 
- RTOS: Target device Teensy 4.1
- Installation type: 
- Version or commit hash: 

#### Steps to reproduce the issue

#### Expected behavior
Finish compiling

#### Actual behavior
Fails compiling
#### Additional information
Inside docker container (dustynv/ros:foxy-pytorch-l4t-r32.7.1)

Followed tutorial below. Have done this multiple times on another PC, no issues and also without problems on this same Jetson inside a docker. Not sure if the problem come from changes in the git repo cloned following the process.
Please let me know if there is any more information I can provide to get this solved.
Thank you!

> https://micro.ros.org/docs/tutorials/core/teensy_with_arduino/

```
root@ros2dev:~# source /opt/ros/foxy/install/setup.bash
root@ros2dev:~# mkdir microros_ws
root@ros2dev:~# cd microros_ws
root@ros2dev:~/microros_ws# git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
Cloning into 'src/micro_ros_setup'...
remote: Enumerating objects: 3385, done.
remote: Counting objects: 100% (1180/1180), done.
remote: Compressing objects: 100% (507/507), done.
remote: Total 3385 (delta 843), reused 957 (delta 666), pack-reused 2205
Receiving objects: 100% (3385/3385), 794.48 KiB | 2.43 MiB/s, done.
Resolving deltas: 100% (2341/2341), done.
root@ros2dev:~/microros_ws# sudo apt update && rosdep update
Hit:1 http://packages.ros.org/ros2/ubuntu bionic InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports bionic InRelease                                       
Hit:3 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease                               
Hit:4 https://apt.kitware.com/ubuntu bionic InRelease                                            
Hit:5 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease    
Hit:6 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease
Reading package lists... Done                      
Building dependency tree       
Reading state information... Done
All packages are up to date.
reading in sources list data from /etc/ros/rosdep/sources.list.d
Warning: running 'rosdep update' as root is not recommended.
  You should run 'sudo rosdep fix-permissions' and invoke 'rosdep update' again without sudo.
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Skip end-of-life distro "dashing"
Skip end-of-life distro "eloquent"
Add distro "foxy"
Skip end-of-life distro "galactic"
Skip end-of-life distro "groovy"
Add distro "humble"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Skip end-of-life distro "kinetic"
Skip end-of-life distro "lunar"
Add distro "melodic"
Add distro "noetic"
Add distro "rolling"
updated cache in /root/.ros/rosdep/sources.cache
root@ros2dev:~/microros_ws# sudo rosdep fix-permissions
Recursively changing ownership of ros home directory '/root/.ros' to 'root:root' (current user)...
Done.
root@ros2dev:~/microros_ws# rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Warning: running 'rosdep update' as root is not recommended.
  You should run 'sudo rosdep fix-permissions' and invoke 'rosdep update' again without sudo.
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Skip end-of-life distro "dashing"
Skip end-of-life distro "eloquent"
Add distro "foxy"
Skip end-of-life distro "galactic"
Skip end-of-life distro "groovy"
Add distro "humble"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Skip end-of-life distro "kinetic"
Skip end-of-life distro "lunar"
Add distro "melodic"
Add distro "noetic"
Add distro "rolling"
updated cache in /root/.ros/rosdep/sources.cache
root@ros2dev:~/microros_ws# sudo apt-get install python3-pip
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-pip is already the newest version (9.0.1-2.3~ubuntu1.18.04.5).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@ros2dev:~/microros_ws# colcon build
Starting >>> micro_ros_setup
Finished <<< micro_ros_setup [5.96s]                  

Summary: 1 package finished [8.45s]
root@ros2dev:~/microros_ws# source install/local_setup.bash
root@ros2dev:~/microros_ws# ros2 run micro_ros_setup create_agent_ws.sh
...
=== ./uros/drive_base (git) ===
Cloning into '.'...
=== ./uros/micro-ROS-Agent (git) ===
Cloning into '.'...
=== ./uros/micro_ros_msgs (git) ===
Cloning into '.'...
#All required rosdeps installed successfully
root@ros2dev:~/microros_ws# ros2 run micro_ros_setup build_agent.sh
Building micro-ROS Agent
Starting >>> micro_ros_msgs
[Processing: micro_ros_msgs]                             
Finished <<< micro_ros_msgs [45.1s]                       
Starting >>> micro_ros_agent
[Processing: micro_ros_agent]                                  
[Processing: micro_ros_agent]                                  
[Processing: micro_ros_agent]                                       
[Processing: micro_ros_agent]micro_ros_agent:build 4% - 2min 0.0s]   
[Processing: micro_ros_agent]                                      
[Processing: micro_ros_agent]                                       
[Processing: micro_ros_agent]                                       
[Processing: micro_ros_agent]                                       
[Processing: micro_ros_agent]                                       
[Processing: micro_ros_agent]                                       
[Processing: micro_ros_agent]                                       
[Processing: micro_ros_agent]                                       
[Processing: micro_ros_agent]                                       
[Processing: micro_ros_agent]                                       
[Processing: micro_ros_agent]                                       
[Processing: micro_ros_agent]                                       
[Processing: micro_ros_agent]                                       
[Processing: micro_ros_agent]                                       
--- stderr: micro_ros_agent                                          
Cloning into 'xrceagent'...
Switched to a new branch 'ros2'
Cloning into 'spdlog'...
HEAD is now at eb322062 Bump version to 1.9.2
CMake Warning (dev) at /usr/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (tinyxml2)
  does not match the name of the calling package (TinyXML2).  This can lead
  to problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindTinyXML2.cmake:40 (find_package_handle_standard_args)
  /opt/ros/foxy/install/share/fastrtps/cmake/fastrtps-config.cmake:51 (find_package)
  CMakeLists.txt:153 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at /root/microros_ws/install/micro_ros_msgs/share/micro_ros_msgs/cmake/ament_cmake_export_targets-extras.cmake:18 (message):
  Failed to find exported target names in
  '/root/microros_ws/install/micro_ros_msgs/share/micro_ros_msgs/cmake/micro_ros_msgs__rosidl_generator_cExport.cmake'
Call Stack (most recent call first):
  /root/microros_ws/install/micro_ros_msgs/share/micro_ros_msgs/cmake/micro_ros_msgsConfig.cmake:41 (include)
  CMakeLists.txt:51 (find_package)

make[2]: *** [micro_ros_agent-prefix/src/micro_ros_agent-stamp/micro_ros_agent-configure] Error 1
make[1]: *** [CMakeFiles/micro_ros_agent.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< micro_ros_agent [9min 24s, exited with code 2]

Summary: 1 package finished [10min 12s]
  1 package failed: micro_ros_agent
  1 package had stderr output: micro_ros_agent
```

pablogs9, 2023-01-05:
Which version of CMake do you have installed?

usaotearoa, 2023-01-05:
root@ros2dev:/# cmake --version
cmake version 3.25.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).

usaotearoa, 2023-01-05:
I have deleted all docker files and re-tried the entire process from the very beginning with the following steps.
And it still fails:

```
docker system prune
docker system prune -a
```

verified that no container and no images were left
```
docker ps
docker images
```
-> nothing shows up

```
git clone https://github.com/dusty-nv/jetson-containers.git
cd jetson-containers
./scripts/docker_run.sh -c dustynv/ros:foxy-pytorch-l4t-r32.7.1
```

then when it completed, imediately took a backup of the "Clean" image
`docker ps`
to get the container-id

`docker container commit --author "usaotearoa" -m 06_01_2023_0802 <container-id> usaotearoa/jetson-ros-containers:foxy_clean`

stopped the origin container 
`docker stop <origin container-id>`

then started the "clean" container
`sudo docker run -d -it --net=host -v /home/administrator/Documents/git_clones/:/root/git_clones/ --privileged usaotearoa/jetson-ros-containers:foxy_clean`

then opened a bash shell inside the container and attached
```
docker ps (copied <clean container-id>)
docker exec <clean container-id> bash
docker attach <clean container-id>
```

inside I followed the tutorial exactly:
`source /opt/ros/foxy/setup.bash`
this might have been source /opt/ros/foxy/install/setup.bash

```
mkdir microros_ws
cd microros_ws
git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
```

```
sudo apt update && rosdep update
rosdep install --from-paths src --ignore-src -y
```

`sudo apt-get install python3-pip`

```
colcon build
source install/local_setup.bash
```

```
ros2 run micro_ros_setup create_agent_ws.sh
ros2 run micro_ros_setup build_agent.sh
```

create agent was ok, no errors, then the build_agent fails.
Below the entire output:

[... 627 lines cut ...]
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `std::locale fmt::v6::internal::locale_ref::get<std::locale>() const'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `typeinfo for fmt::v6::format_error'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > fmt::v6::internal::grouping_impl<char>(fmt::v6::internal::locale_ref)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `int fmt::v6::internal::format_float<long double>(long double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v6::basic_string_view<char>, spdlog::level::level_enum, fmt::v6::basic_string_view<char>)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `int fmt::v6::internal::snprintf_float<long double>(long double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `fmt::v6::format_error::~format_error()'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `fmt::v6::internal::basic_data<void>::hex_digits'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `fmt::v6::internal::basic_data<void>::digits'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `int fmt::v6::internal::format_float<double>(double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `fmt::v6::internal::basic_data<void>::zero_or_powers_of_10_64'
collect2: error: ld returned 1 exit status
make[5]: *** [micro_ros_agent] Error 1
make[4]: *** [CMakeFiles/micro_ros_agent.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [micro_ros_agent-prefix/src/micro_ros_agent-stamp/micro_ros_agent-build] Error 2
make[1]: *** [CMakeFiles/micro_ros_agent.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< micro_ros_agent [9min 47s, exited with code 2]

Summary: 1 package finished [10min 33s]
  1 package failed: micro_ros_agent
  1 package had stderr output: micro_ros_agent
```

usaotearoa, 2023-01-05:
fair to say I also modified the scripts/l4t_version.sh 
to return the matching string for the container release version "R32.7.1"
`L4T BSP Version:  L4T R32.7.1`

as it returned
`L4T BSP Version:  L4T R32.7.3`

done so by the below modification, commented original line. Snippet includes leading and following lines for context:
```
L4T_REVISION_MAJOR=${L4T_REVISION:0:1}
L4T_REVISION_MINOR=${L4T_REVISION:2:1}

#L4T_VERSION="$L4T_RELEASE.$L4T_REVISION"
L4T_VERSION="$L4T_RELEASE.7.1"

echo "L4T BSP Version:  L4T R$L4T_VERSION"

```

Without this modification it of course would not work, since the container available from dusty are only in R32.7.1
I don't believe this causes any issues.
However, the entire picture is a little bit bigger, as I was forced to update from R32.7.1 to R32.7.3 due to a regular update process via apt-get update && apt-get upgrade caused the kernel packages to be bumped to the next release...

usaotearoa, 2023-01-06:
Just did an entirely new SD card with new image for the Jetson Nano (4GB) B01.
Left it at R32.7.1 and followed above steps:

But unfortunately with the same result.
Good way to spend an entire day, downloading and building SW just to end up at the same problem...

Not sure where to go from here.

git clone https://github.com/dusty-nv/jetson-containers.git
```
cd jetson-containers
./scripts/docker_run.sh -c dustynv/ros:foxy-pytorch-l4t-r32.7.1
```

then when it completed, imediately took a backup of the "Clean" image
`docker ps`
to get the container-id

`docker container commit --author "usaotearoa" -m 06_01_2023_1515 <container-id> usaotearoa/jetson-ros-containers:foxy_clean`

stopped the origin container 
`docker stop <origin container-id>`

then started the "clean" container
`sudo docker run -d -it --net=host -v /home/administrator/Documents/git_clones/:/root/git_clones/ --privileged usaotearoa/jetson-ros-containers:foxy_clean`

then opened a bash shell inside the container and attached
```
docker ps (copied <clean container-id>)
docker exec <clean container-id> bash
docker attach <clean container-id>
```

inside I followed the tutorial exactly:
`source /opt/ros/foxy/install/setup.bash`

```
mkdir microros_ws
cd microros_ws
git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
```

```
sudo apt update && rosdep update
rosdep install --from-paths src --ignore-src -y
```

`sudo apt-get install python3-pip`

```
colcon build
source install/local_setup.bash
```

```
ros2 run micro_ros_setup create_agent_ws.sh
ros2 run micro_ros_setup build_agent.sh
```

create agent was ok, no errors, then the build_agent fails.
Below the entire output:

```
root@rosdev:/# source /opt/ros/foxy/install/setup.bash 
root@rosdev:/# mkdir microros_ws
root@rosdev:/# cd microros_ws/
root@rosdev:/microros_ws# git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
Cloning into 'src/micro_ros_setup'...
remote: Enumerating objects: 3385, done.
remote: Counting objects: 100% (1180/1180), done.
[... 623 lines cut ...]
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `typeinfo for fmt::v6::format_error'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > fmt::v6::internal::grouping_impl<char>(fmt::v6::internal::locale_ref)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `int fmt::v6::internal::format_float<long double>(long double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v6::basic_string_view<char>, spdlog::level::level_enum, fmt::v6::basic_string_view<char>)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `int fmt::v6::internal::snprintf_float<long double>(long double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `fmt::v6::format_error::~format_error()'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `fmt::v6::internal::basic_data<void>::hex_digits'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `fmt::v6::internal::basic_data<void>::digits'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `int fmt::v6::internal::format_float<double>(double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `fmt::v6::internal::basic_data<void>::zero_or_powers_of_10_64'
collect2: error: ld returned 1 exit status
make[5]: *** [micro_ros_agent] Error 1
make[4]: *** [CMakeFiles/micro_ros_agent.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [micro_ros_agent-prefix/src/micro_ros_agent-stamp/micro_ros_agent-build] Error 2
make[1]: *** [CMakeFiles/micro_ros_agent.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< micro_ros_agent [10min 47s, exited with code 2]
                                       
Summary: 1 package finished [11min 35s]
  1 package failed: micro_ros_agent
  1 package had stderr output: micro_ros_agent
root@rosdev:/microros_ws# 
```

usaotearoa, 2023-01-06:
Is there a way to increase verbosity during the build process?
I have tried multiple things so far and spend well over 2 days troubleshooting this issue.
1. deleted all docker container
2. tried a different ROS feature-set (such as desktop, pytorch, base)
3. completely did a new SD image and new fresh Ubuntu on the Jetson.

However, nothing seems to work. It keeps on failing during the build process.

I remember about 3 weeks back, the first time I've tried it on the Jetson inside a docker it worked flawlessly. There were only some hickups, as I wasn't familiar with using docker container / images.

Not sure if anything has changed since, but I have spent so far a total of 3 days of my life troubleshooting this. Time I will never receive back.

Please help me overcome this hurdle, as everything is hinging on this for my current project to work. The entire project I worked on over the course of 2022 with the Jetson at its core.