micro-ROS/micro_ros_setup #603: MicroROS fails to build on Raspberry Pi 4/Ubuntu

Chuckwisc, 2023-01-09:
Error running colcon build on humble micro_ros_agent and micro_ros_msgs
raspberry pi 4, arm64, ubuntu 22.04,  01/09/2023
cmake version 3.22.1

    cd <your_ws>
    git clone -b humble https://github.com/micro-ROS/micro_ros_setup src/micro_ros_setup
    sudo apt install python3-vcstool build-essential
    sudo apt update && rosdep update
    rosdep install --from-path src --ignore-src -y
    colcon build
    source install/setup.bash
    ros2 run micro_ros_setup create_agent_ws.sh
        worked fine to here, made *_ws/src/uros/micro-ROS-Agent *_ws/src/uros/micro_ros_msgs

  Is there another build method which should be used on RPi for micro_ros ?

layer@rpi2:~/rover_ws$ colcon build --packages-select micro_ros_msgs
Starting >>> micro_ros_msgs
--- stderr: micro_ros_msgs                            
/usr/bin/ld: /home/layer/ros2_humble/ros2-linux/lib/librosidl_runtime_c.so: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/micro_ros_msgs__rosidl_generator_c.dir/build.make:192: libmicro_ros_msgs__rosidl_generator_c.so] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:213: CMakeFiles/micro_ros_msgs__rosidl_generator_c.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: /home/layer/ros2_humble/ros2-linux/lib/librosidl_typesupport_cpp.so: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/micro_ros_msgs__rosidl_typesupport_cpp.dir/build.make:151: libmicro_ros_msgs__rosidl_typesupport_cpp.so] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:395: CMakeFiles/micro_ros_msgs__rosidl_typesupport_cpp.dir/all] Error 2
/usr/bin/ld: /home/layer/ros2_humble/ros2-linux/lib/librosidl_typesupport_fastrtps_cpp.so: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/micro_ros_msgs__rosidl_typesupport_fastrtps_cpp.dir/build.make:163: libmicro_ros_msgs__rosidl_typesupport_fastrtps_cpp.so] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:343: CMakeFiles/micro_ros_msgs__rosidl_typesupport_fastrtps_cpp.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< micro_ros_msgs [2.37s, exited with code 2]

Summary: 0 packages finished [6.12s]
  1 package failed: micro_ros_msgs
  1 package had stderr output: micro_ros_msgs
layer@rpi2:~/rover_ws$ colcon build --packages-select micro_ros_agent
Starting >>> micro_ros_agent
[Processing: micro_ros_agent]                            
[Processing: micro_ros_agent]                                    
--- stderr: micro_ros_agent                                      
HEAD is now at 3eb56b5 Release v2.3.0
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (tinyxml2)
  does not match the name of the calling package (TinyXML2).  This can lead
  to problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindTinyXML2.cmake:40 (find_package_handle_standard_args)
  /home/layer/ros2_humble/ros2-linux/share/fastrtps/cmake/fastrtps-config.cmake:51 (find_package)
  CMakeLists.txt:153 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

gmake[5]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libtinyxml2.so', needed by 'libmicroxrcedds_agent.so.2.3.0'.  Stop.
gmake[5]: *** Waiting for unfinished jobs....
gmake[4]: *** [CMakeFiles/Makefile2:83: CMakeFiles/microxrcedds_agent.dir/all] Error 2
gmake[3]: *** [Makefile:136: all] Error 2
gmake[2]: *** [CMakeFiles/xrceagent.dir/build.make:86: agent/src/xrceagent-stamp/xrceagent-build] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:85: CMakeFiles/xrceagent.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
---
Failed   <<< micro_ros_agent [1min 6s, exited with code 2]

Summary: 0 packages finished [1min 10s]
  1 package failed: micro_ros_agent
  1 package had stderr output: micro_ros_agent

pablogs9, 2023-01-10:
Why are you doing ` colcon build --packages-select micro_ros_msgs` instead of `ros2 run micro_ros_setup build_agent.sh`?

Chuckwisc, 2023-01-10:
All other ROS2 packages have built ok in my ROS2 install. I'm using the linorobot2 packages.
I am doing  colcon build --packages-select micro_ros_msgs  micro_ros_agent
just to isolate where the error was coming from within ros2 run micro_ros_setup build_agent.sh .

pablogs9, 2023-01-10:
Do you have installed `ros-humble-tinyxml2-vendor` and `libtinyxml2-dev`?

Chuckwisc, 2023-01-10:
Your comment about my ros2 install being the issue got me checking things.
I had installed ros2-humble-20220523-linux-jammy-**amd64**.tar.bz on the arm64 RPi.
Re-installing ROS2 with ros2-humble-20221123-linux-jammy-**arm64**.tar.bz
 and will try again.

Chuckwisc, 2023-01-10:
The arm64 ros2 on the RPi, instead of the amd64 ros2 version let the build work.
ros2 run micro_ros_setup build_agent.sh
works fine now, and the linorobot2 bringup also worked.

closed 2023-01-10 by Chuckwisc