micro-ROS/micro_ros_setup #608: Facing issues in Custom Transport "CAN-ISOTP"

SiddharthZo, 2023-01-31:
Hardware description: ESP32
RTOS: FreeRTOS
Installation type: micro_ros_setup
Version or commit hash: ROS2 foxy

I am trying to make a custom transport for CAN-ISOTP, used UDP packet based communication and using the inbuilt UDP agent.
I wrote the custom client code for CAN-ISOTP and it is woking well.

In my approach, the data sent by the custom client (via CAN Adapter) is not directly send to the custom agent but via a CAN-ISOTP python code.  The python code grabs all the CAN-ISOTP data from the custom client (running on ESP32) and then using UDP Socket communication, sends the data to the UDP agent. Both the python code and the UDP agent are running on the same ip and port.

I used this method as it is simple and no need to design an agent for CAN-ISOTP from scratch.

**Below attached is the block diagram of my approach:**
[image]

**I am getting the following results:** 

[image]

[image]

I am not able to understand what is the problem. Why the inbuilt or existing UDP agent is not doing further process like creating topics etc.  Please guide me regarding that.

Thank you so much!!

SiddharthZo, 2023-03-14:
Hello @pablogs9 ,

The micro-ROS agent is freezing after creating the participant. It is not doing the further procedure like creating the Topic, publisher and datawriter. My issue is similar to the issue #618 (The micro-ROS agent was freezing) . 

**My results:**
[image]
[image]

**I also increased the stack size but still it is freezing.** 

[image]

Kindly help me to know what I am doing wrong to fix this issue.

pablogs9, 2023-03-15:
It seems to be related to your transport implementation. Could you provide more details and code of the transports you are using?

SiddharthZokarkar, 2023-03-20:
Hello @pablogs9 

Thank you for your reply . Here is the code for the custom client and main.c . I am using the custom agent as the inbuilt UDP agent in the examples. 

**custom client**

```

#include <stdio.h>
#include <unistd.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <std_msgs/msg/int32.h>

#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <uxr/client/transport.h>
#include <rmw_microxrcedds_c/config.h>
#include <microros_transports.h>

#include <unistd.h>
#include <stdio.h>
#include <string.h>
#include <stdbool.h>

#ifdef ESP_PLATFORM
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include <driver/gpio.h>
#include <driver/can.h>
#include "isotp.h"
#endif

/* Alloc IsoTpLink statically in RAM */
static IsoTpLink g_link;

/* Alloc send and receive buffer statically in RAM */
static uint8_t g_isotpRecvBuf[500];
static uint8_t g_isotpSendBuf[500];
bool received_data_status = false;

int ret;
uint16_t out_size = 0;

int isotp_user_send_can(const uint32_t arbitration_id,
                        const uint8_t *data, const uint8_t size)
{
    can_message_t msg;
    memcpy(msg.data, data, size);
    msg.data_length_code = size;
    msg.identifier = arbitration_id;
    msg.flags = CAN_MSG_FLAG_EXTD;
    return can_transmit(&msg, 0);
}

/* required, return system tick, unit is millisecond */
uint32_t isotp_user_get_ms(void)
{
    return esp_timer_get_time() / 1000;
}

/* optional, provide to receive debugging log messages */
void isotp_user_debug(const char *message, ...)
{

    va_list argp;
    va_start(argp, message);
    printf(message, argp);
    va_end(argp);
[... 331 lines cut ...]
    ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
    wifi_init_sta();
#elif defined(RMW_UXRCE_TRANSPORT_CUSTOM)
    rmw_uros_set_custom_transport(  
        false,  
        (void *) &args,  
        custom_transport_open,  
        custom_transport_close,  
        custom_transport_write,  
        custom_transport_read);  
#endif

    // start microROS task  , old size 12*2048
    xTaskCreate(appMain, "uros_task", 16*4096, NULL, 5, NULL);
}

```

For getting an idea please refer the below picture

[image]

Please let me know if anything else is needed.

Thank you so much!!

pablogs9, 2023-03-21:
Try to enable framing at transport layer: 

```c
    rmw_uros_set_custom_transport(  
        true,  
        (void *) &args,  
        custom_transport_open,  
        custom_transport_close,  
        custom_transport_write,  
        custom_transport_read); 
```

https://micro-xrce-dds.docs.eprosima.com/en/latest/transport.html?highlight=framing#stream-framing-protocol

closed 2023-04-03 by SiddharthZokarkar