micro-ROS/micro_ros_setup #613: ros2 run micro_ros_setup create_agent_ws.sh failed

TianmingLi, 2023-02-16:
## Issue template

- Hardware description: vmware virtual machine
- RTOS: Linux
- Installation type: micro_ros_setup  
- Version or commit hash:  commit 58b1dd729f6cb3481d087d55c60a312e8b9f67a4 (HEAD -> humble, origin/humble, origin/HEAD)

#### Steps to reproduce the issue

ros2 run micro_ros_setup create_agent_ws.sh
#### Expected behavior

#### Actual behavior
Repo-file ros2.repos already present, overwriting!
[ros2run]: Process exited with failure 7
[ros2run]: Process exited with failure 7
#### Additional information

pablogs9, 2023-02-16:
Sorry, but we need more details (such as replication instructions) to address this issue.

TianmingLi, 2023-02-16:
> Sorry, but we need more details (such as replication instructions) to address this issue.

thanks for the reply. i found out the reason of this problem.

this is for the people who have same problems.

---
it casued by china GFW, it blocked ROS repo address.

so, if you are living china and have same problem. 

use tools `proxychains`

and command like this

```
proxychains ros2 run micro_ros_setup create_agent_ws.sh

```

closed 2023-02-16 by TianmingLi