micro-ROS/micro_ros_setup #619: Building Custom library fails

michaell12, 2023-03-19:
- Hardware description: Ultra 96 with microblaze
- RTOS: freeRTOS
- Installation type: micro_ros_setup
- Version or commit hash: foxy

I'm Trying to integrate microROS on microblaze, after creating a genereic firmware workspace, i build to get the libmicroros.a using the following toolchain:
[image]
I'm trying to make a ping pong application, so i created a project in vitis and add the custom library and the microros include from the build (in gcc linker -> Libraries)
Then i created a custom transport and tried to build a simple application
When i try to build the project, i get the following error:
```
Building target: micro_ros_microblaze.elf
Invoking: MicroBlaze gcc linker
mb-gcc -L/home/kislevro/microros_ws/firmware/build -Wl,-T -Wl,../src/lscript.ld -L/home/kislevro/workspace/design_Ultra96_MicroBlaze_wrapper/export/design_Ultra96_MicroBlaze_wrapper/sw/design_Ultra96_MicroBlaze_wrapper/freertos10_xilinx_microblaze_0/bsplib/lib -mlittle-endian -mxl-barrel-shift -mxl-pattern-compare -mcpu=v11.0 -mno-xl-soft-mul -Wl,--no-relax -Wl,--gc-sections -o "micro_ros_microblaze.elf"  ./src/freertos_hello_world.o  ./micro_ros_microblaze_utils/custom_memory_manager.o ./micro_ros_microblaze_utils/microros_allocators.o ./micro_ros_microblaze_utils/microros_time.o ./micro_ros_microblaze_utils/uart_transport.o   -lmicroros -Wl,--start-group,-lxil,-lfreertos,-lgcc,-lc,--end-group -fno-pie
/tools/Xilinx/Vitis/2021.1/gnu/microblaze/lin/x86_64-oesdk-linux/usr/bin/microblaze-xilinx-elf/microblaze-xilinx-elf-ld.real: /home/kislevro/microros_ws/firmware/build/libmicroros.a(libmicrocdr-basic.c.obj): in function `ucdr_serialize_bool':
/home/kislevro/microros_ws/firmware/mcu_ws/eProsima/Micro-CDR/src/c/types/basic.c:271: undefined reference to `GLOBAL_OFFSET_TABLE'
collect2.real: fatal error: ld terminated with signal 11 [Segmentation fault], core dumped
compilation terminated.
make: * [makefile:39: micro_ros_microblaze.elf] Error 1
```
[image]

Is there something wrong maybe when i build the .a file?

michaell12, 2023-04-03:
Fixed by adding the the colcon.meta file the following args:
`"microcdr": {
            "cmake-args": [
                "-DUCDR_PIC=OFF"
            ]
        },`

Acuadros95, 2023-07-31:
Closing as solved by author

closed 2023-07-31 by Acuadros95