micro-ROS/micro_ros_setup #622: Error building micro-ROS component on Raspberry Pi 4 with ESP32

liamhan0905, 2023-03-26:
## Issue template

- Hardware description: 
   - Raspberry Pi 4
   - esp32 WROOM

- RTOS: freertos on esp32
- Installation type: tried `micro_ros_setup` but got error (mentioned below). Tried installing ESP-IDF directly and then using [micro-ROS component for ESP-IDF](https://github.com/micro-ROS/micro_ros_espidf_component). (More errors mentioned below)
- Version or commit hash: 
   - installed ROS 2 `humble` on Rapsberry Pi
   - ESP-IDF (v5.0)

## Description

I first tried installing `micro_ros_setup`. The [build process](https://github.com/micro-ROS/micro_ros_setup/blob/humble/README.md#building
) was okay but when [creating micro-ROS firmware](https://github.com/micro-ROS/micro_ros_setup/blob/humble/README.md#creating-micro-ros-firmware) by running the following command,
```
ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32
```
I get the following error
```
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
ERROR: tool xtensa-esp32-elf does not have versions compatible with platform linux-arm64
[ros2run]: Process exited with failure 1
```
It seems like `micro_ros_setup` is not compatible with raspberry pi 4.

So instead, I installed the ESP-IDF following this [link](https://docs.espressif.com/projects/esp-idf/en/release-v5.0/esp32/get-started/linux-macos-setup.html). I was able to flash the `hello_world` example. The version I installed was v5.0 since it states that `micro-ROS component has been tested in ESP-IDF v4.3, v4.4, and v5.0 with ESP32` shown [here](https://github.com/micro-ROS/micro_ros_espidf_component).
```
ubuntu@ubuntu:~$ idf.py --version
ESP-IDF v5.0.1-397-g3050ea656f
```
I wanted to ask 
1. Can `micro_ros_setup` work for raspberry pi 4? Or is it not compatible for linux-arm64?
2. I think the `Example` section can be elaborated. In the example shown [here](https://github.com/micro-ROS/micro_ros_espidf_component/#example), why isn't `components` directory created under the `int32_publisher` directory? Under the `Usage` [section](https://github.com/micro-ROS/micro_ros_espidf_component/#usage), it states to clone the repo directly in the components folder. Please correct me if I'm wrong. 

    I copied the `int32_publisher` example from the `micro_ros_espidf_component`([link](https://github.com/micro-ROS/micro_ros_espidf_component)) into the home directory, created a components directory instead `int32_publisher` directory and then git cloned `micro_ros_espidf_component` inside it.
      ```
      ubuntu@ubuntu:~/int32_publisher/components$ ls
      micro_ros_espidf_component
      ```
      Running `idf.py build` command inside the `int32_publisher` directory,  generates the error below
      ```
      MakeFiles/__idf_main.dir/main.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj   -c ../main/main.c
      ../main/main.c:10:10: fatal error: uros_network_interfaces.h: No such file or directory
         10 | #include <uros_network_interfaces.h>
            |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      compilation terminated.
      [844/851] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/scheme_softap.c.objninja: build stopped: subcommand failed.
      HINT: Please make sure that the header name is correct. Also please check if you've specified all component dependencies with 'idf_component_register(REQUIRES ...)'. If the component is not present then it should be added by the IDF Component Manager. For more information run 'idf.py docs -sp api-guides/build-system.html'.
      Also, please check if the header file has been removed, renamed or relocated - refer to the migration guide for more information.
      ninja failed with exit code 1, output of the command is in the /home/ubuntu/int32_publisher/build/log/idf_py_stderr_output_651110 and /home/ubuntu/int32_publisher/build/log/idf_py_stdout_output_651110
      ```
      I googled this error but didn't get anything helpful.

pablogs9, 2023-03-28:
You can run your `idf.py build` directly into the example folder since it is prepared to find the micro-ROS component two folders above.

Copying the micro-ROS module into the `components` folder is the way for integrating it into your own IDF project because it is the default way of doing it: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#example-project

liamhan1, 2023-04-03:
If I run `idf.py build` directly into the example folder under `micro_ros_espidf_component`, I get the following error. So is it possible to use Raspberry Pi 4 as the host? Everything worked well if I used my desktop but running into errors if I use Raspberry Pi 4. 
```
ubuntu@ubuntu:~/micro_ros_espidf_component/examples/int32_publisher$ get_idf
Setting IDF_PATH to '/home/ubuntu/esp/esp-idf'
Detecting the Python interpreter
Checking "python3" ...
Python 3.8.10
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Using a supported version of tool cmake found in PATH: 3.16.3.
However the recommended version is 3.24.0.
Adding ESP-IDF tools to PATH...
Using a supported version of tool cmake found in PATH: 3.16.3.
However the recommended version is 3.24.0.
Checking if Python packages are up to date...
Constraint file: /home/ubuntu/.espressif/espidf.constraints.v5.0.txt
Requirement files:
 - /home/ubuntu/esp/esp-idf/tools/requirements/requirements.core.txt
Python being checked: /home/ubuntu/.espressif/python_env/idf5.0_py3.8_env/bin/python
Python requirements are satisfied.
Added the following directories to PATH:
  /home/ubuntu/esp/esp-idf/components/esptool_py/esptool
  /home/ubuntu/esp/esp-idf/components/espcoredump
  /home/ubuntu/esp/esp-idf/components/partition_table
  /home/ubuntu/esp/esp-idf/components/app_update
  /home/ubuntu/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin
  /home/ubuntu/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin
  /home/ubuntu/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin
  /home/ubuntu/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin
  /home/ubuntu/.espressif/python_env/idf5.0_py3.8_env/bin
  /home/ubuntu/esp/esp-idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

ubuntu@ubuntu:~/micro_ros_espidf_component/examples/int32_publisher$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/ubuntu/micro_ros_espidf_component/examples/int32_publisher/build
Executing "ninja all"...
ninja: error: '/home/ubuntu/micro_ros_espidf_component/libmicroros.a', needed by 'int32_publisher.elf', missing and no known rule to make it
ninja failed with exit code 1, output of the command is in the /home/ubuntu/micro_ros_espidf_component/examples/int32_publisher/build/log/idf_py_stderr_output_700773 and /home/ubuntu/micro_ros_espidf_component/examples/int32_publisher/build/log/idf_py_stdout_output_700773
[0/2] Re-checking globbed directories...ubuntu@ubuntu:~/micro_ros_espidf_component/examples/int32_publisher$ 

```

pablogs9, 2023-04-04:
Regarding the first error, have you installed the required dependencies https://github.com/micro-ROS/micro_ros_espidf_component#dependencies ?

Regarding

> Installing ESP-IDF tools
> Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
> ERROR: tool xtensa-esp32-elf does not have versions compatible with platform linux-arm64
> [ros2run]: Process exited with failure 1

It is not related to micro-ROS

liamhan1, 2023-04-04:
> Regarding the first error, have you installed the required dependencies https://github.com/micro-ROS/micro_ros_espidf_component#dependencies ?
Yes, all the dependencies is installed

I followed the build and then ran `ros2 run micro_ros_setup component esp_idf` from the `uros_ws` directory. Tree shown below. And then tried building the `int32_publisher` example.

```
ubuntu@ubuntu:~/uros_ws$ tree -d -L 1
.
├── build
├── install
├── log
├── micro_ros_espidf_component
└── src
```

So it is possible to use raspberry pi 4 with esp32?

MadsRossen, 2024-11-21:
In relation to this i got the same error, when building the esp-idf tools on Raspberry PI 5 running Ubuntu 24.04.

I tried to build it following [this](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/linux-macos-setup.html) installation tutorial from espressif, and it worked! 

Then, to make it work using the `ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32` command, i edited the file [create.sh](https://github.com/micro-ROS/micro_ros_setup/blob/jazzy/config/freertos/esp32/create.sh).

Commenting line 14. 23 and 25 and adding `./esp-idf/install.sh` to this: 
```bash
# python3 esp-idf/tools/idf_tools.py install
./esp-idf/install.sh

echo "Installing ESP-IDF virtualenv"
dpkg -s python3-pip > /dev/null
if [[ $? -ne 0 ]]; then
    echo "Error: python3-pip package must be installed before continuing..."
    exit 1
fi
pip3 install virtualenv --break-system-packages
# python3 esp-idf/tools/idf_tools.py install-python-env

# eval $(python3 $FW_TARGETDIR/toolchain/esp-idf/tools/idf_tools.py export --prefer-system)
```
This worked for me.

ola1182006, 2025-03-18:
[image]

ola1182006, 2025-03-18:
Help me out. Having problem integrating the software with esp32. This error comes up