micro-ROS/micro_ros_setup #631: Micro_ros udp transport build issue on esp32

Apurva2510, 2023-04-14:
## Issue template

- Hardware description: esp-wroom-32
- RTOS: freertos
- Installation type: micro_ros_setup
- Version or commit hash: humble

#### Steps to reproduce the issue
ros2 run micro_ros_setup configure_firmware.sh motor_subscriber -t udp -i [YOUR IP] -p 8888

ros2 run micro_ros_setup build_firmware.sh

#### Expected behavior
100% build

#### Actual behavior
[image]

#### Additional information
When I try to build it with serial Transport the build is successful with some stderr outputs but when I try udp it does not seem to build

pablogs9, 2023-04-14:
Hello @Apurva2510 the recommended approach for using micro-ROS in ESP32 is using the component for ESP-IDF. Could you give a try using this repo: https://github.com/micro-ROS/micro_ros_espidf_component

Apurva2510, 2023-04-14:
> Hello @Apurva2510 the recommended approach for using micro-ROS in ESP32 is using the component for ESP-IDF. Could you give a try using this repo: https://github.com/micro-ROS/micro_ros_espidf_component

@pablogs9 OK. I will try that approach.

Apurva2510, 2023-04-14:
Update: @pablogs9 The repo that you mentioned worked for the udp transport. Thanks a lot for that advise!

Acuadros95, 2023-04-17:
Closing as its solved

closed 2023-04-17 by Acuadros95