micro-ROS/micro_ros_setup #636: micro ros agent build fails ROS2 humble

FrGe2016, 2023-05-05:
## Issue template

- Hardware description:  PC with ROS2 Humble
- RTOS: 
- Installation type:  ros2 run micro_ros_setup build_agent.sh
- Version or commit hash:  humble

#### Steps to reproduce the issue
 I installed the package according to the instructions at https://micro.ros.org/docs/tutorials/core/teensy_with_arduino/

# Source the ROS 2 installation
source /opt/ros/foxy/setup.bash
# Create a workspace and download the micro-ROS tools
mkdir microros_ws
cd microros_ws
git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
# Update dependencies using rosdep
sudo apt update && rosdep update
rosdep install --from-paths src --ignore-src -y
# Install pip
sudo apt-get install python3-pip

# Build micro-ROS tools and source them
colcon build
source install/local_setup.bash

# Download micro-ROS agent packages
ros2 run micro_ros_setup create_agent_ws.sh

# Build step
ros2 run micro_ros_setup build_agent.sh

#### Expected behavior 
Should build without errors

#### Actual behavior 
Build fails with theses messages

ros2 run micro_ros_setup build_agent.sh
Building micro-ROS Agent
Starting >>> micro_ros_msgs
Finished <<< micro_ros_msgs [7.26s]                  
Starting >>> micro_ros_agent
[Processing: micro_ros_agent]                         
--- stderr: micro_ros_agent                           
Cloning into 'xrceagent'...
Switched to a new branch 'ros2'
HEAD is now at 9311555 Release v2.4.0
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (tinyxml2)
  does not match the name of the calling package (TinyXML2).  This can lead
  to problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindTinyXML2.cmake:40 (find_package_handle_standard_args)
  /opt/ros/humble/share/fastrtps/cmake/fastrtps-config.cmake:51 (find_package)
  CMakeLists.txt:153 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

In file included from /home/francois/microros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/Agent.cpp:18:
/home/francois/microros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/include/agent/Agent.hpp:18:10: fatal error: uxr/agent/AgentInstance.hpp: No such file or directory
   18 | #include <uxr/agent/AgentInstance.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/francois/microros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/main.cpp:15:
/home/francois/microros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/include/agent/Agent.hpp:18:10: fatal error: uxr/agent/AgentInstance.hpp: No such file or directory
   18 | #include <uxr/agent/AgentInstance.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
compilation terminated.
gmake[5]: *** [CMakeFiles/micro_ros_agent.dir/build.make:76: CMakeFiles/micro_ros_agent.dir/src/main.cpp.o] Error 1
gmake[5]: *** Waiting for unfinished jobs....
gmake[5]: *** [CMakeFiles/micro_ros_agent.dir/build.make:90: CMakeFiles/micro_ros_agent.dir/src/agent/Agent.cpp.o] Error 1
In file included from /home/francois/microros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_manager.cpp:18:
/home/francois/microros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/include/agent/graph_manager/graph_manager.hpp:28:10: fatal error: fastdds/dds/domain/DomainParticipant.hpp: No such file or directory
   28 | #include <fastdds/dds/domain/DomainParticipant.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/francois/microros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_typesupport.cpp:18:
/home/francois/microros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/include/agent/graph_manager/graph_typesupport.hpp:28:10: fatal error: fastdds/dds/domain/DomainParticipant.hpp: No such file or directory
   28 | #include <fastdds/dds/domain/DomainParticipant.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
compilation terminated.
gmake[5]: *** [CMakeFiles/micro_ros_agent.dir/build.make:104: CMakeFiles/micro_ros_agent.dir/src/agent/graph_manager/graph_manager.cpp.o] Error 1
gmake[5]: *** [CMakeFiles/micro_ros_agent.dir/build.make:118: CMakeFiles/micro_ros_agent.dir/src/agent/graph_manager/graph_typesupport.cpp.o] Error 1
gmake[4]: *** [CMakeFiles/Makefile2:137: CMakeFiles/micro_ros_agent.dir/all] Error 2
gmake[3]: *** [Makefile:146: all] Error 2
gmake[2]: *** [CMakeFiles/micro_ros_agent.dir/build.make:86: micro_ros_agent-prefix/src/micro_ros_agent-stamp/micro_ros_agent-build] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:111: CMakeFiles/micro_ros_agent.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
---
Failed   <<< micro_ros_agent [36.3s, exited with code 2]

Summary: 1 package finished [43.8s]
  1 package failed: micro_ros_agent
  1 package had stderr output: micro_ros_agent
[ros2run]: Process exited with failure 2

#### Additional information

pablogs9, 2023-05-08:
@Acuadros95 PTAL

Acuadros95, 2023-05-08:
Tested sucesfully on a fresh `ros:humble` docker.

How did you install ROS 2? Do you have FastDDS installed?

```bash
root@mydevice:/home/microros_ws# apt list --installed | grep fastrtps
ros-humble-fastrtps-cmake-module/jammy,now 2.2.0-2jammy.20230426.000247 amd64 [installed,automatic]
ros-humble-fastrtps/jammy,now 2.6.4-1jammy.20230425.234452 amd64 [installed,automatic]
ros-humble-rmw-fastrtps-cpp/jammy,now 6.2.2-1jammy.20230426.040350 amd64 [installed,automatic]
ros-humble-rmw-fastrtps-shared-cpp/jammy,now 6.2.2-1jammy.20230426.033343 amd64 [installed,automatic]
ros-humble-rosidl-typesupport-fastrtps-c/jammy,now 2.2.0-2jammy.20230426.010916 amd64 [installed,automatic]
ros-humble-rosidl-typesupport-fastrtps-cpp/jammy,now 2.2.0-2jammy.20230426.005843 amd64 [installed,automatic]
```

Acuadros95, 2023-07-25:
Closing due to inactivity

closed 2023-07-25 by Acuadros95