micro-ROS/micro_ros_setup #645: microros failed in "rclc_node_init_default"

wyingda, 2023-05-27:
## Issue template
- Hardware description: Olimex board with STM32
- RTOS: FreeRtos
- Installation type: micro_ros_setup serial
- Version or commit hash: foxy

#### Steps to reproduce the issue
Starting with https://micro.ros.org/docs/tutorials/core/first_application_rtos/freertos/.
source:micro_ros_setup
I am using a Linux laptop with Ubuntu 20.04 and build ROS 2 foxy from the sources.
[image]

1.I started the agent in a terminal.(
docker run -it --rm --net=host -v /dev/shm:/dev/shm -v /dev:/dev --privileged microros/micro-ros-agent:foxy serial --dev /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 -v6 and 
ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 -v6
)
2.I checked incoming ROS 2 topic in third window.

Error: there is not connection established between client and agent. 
#### Expected behavior

working with demo ping_pong

#### Actual behavior
[1685183904.873771] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1685183904.874110] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1685183913.488719] info     | Root.cpp           | create_client            | create                 | client_key: 0x08306BA8, session_id: 0x81
[1685183913.488861] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x08306BA8, address: 0
[1685183913.489005] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x08306BA8, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1685183913.681995] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x08306BA8, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1685183913.682401] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x08306BA8, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1685183913.874918] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x08306BA8, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1685183913.875032] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x08306BA8, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1685183914.067942] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x08306BA8, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1685183914.068092] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x08306BA8, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1685183914.260808] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x08306BA8, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1685183914.260940] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x08306BA8, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1685183914.453948] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x08306BA8, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1685183914.454137] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x08306BA8, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80

#### Additional information
I have verified the functionality of TX and RX.No problem with TX, RX and memory allocation.
I printed the information sent and received and found that agent not received this message
7e  0  0  18  0  80  0  0  0  0  1  10  0  58  52  43   45  1  0  1  f  8  30  6b  a8  81  0  fc  1  1a  fa.
Is it something wrong with my agent? Are you have any ideas?
I really appreciate your help. @pablogs9

wyingda, 2023-05-27:
[image]

closed 2023-06-16 by wyingda

orengeokami, 2024-05-13:
Until now, I still met this problem. But I resolved this by adding force reset after function rclc_node_init_default invoked with RCL_RET_ERROR:
some block might be this:

  	//create transport:
  	{
  		set_custom_transport:

		//USART-DMA
		/*
  		rmw_ret_t rmw_ret = \
  			rmw_uros_set_custom_transport(
  				true,
  				(void*)&huart2,
  				cubemx_transport_open,
  				cubemx_transport_close,
  				cubemx_transport_write,
  				cubemx_transport_read);
  				*/

		//USB-CDC
  		rmw_ret_t rmw_ret = \
  			rmw_uros_set_custom_transport(
  				true,
  		  	    NULL,
  		  	    cubemx_transport_open,
  		  	    cubemx_transport_close,
  		  	    cubemx_transport_write,
  		  	    cubemx_transport_read);

          if(rmw_ret != RMW_RET_OK)
          {
              goto set_custom_transport;
          }
  	};

  	//create allocator:
    {
		rcl_allocator_t freeRTOS_allocator = \
			rcutils_get_zero_initialized_allocator();

//		freeRTOS_allocator.allocate = microros_allocate;

//		freeRTOS_allocator.deallocate = microros_deallocate;

//		freeRTOS_allocator.reallocate = microros_reallocate;

//		freeRTOS_allocator.zero_allocate = microros_zero_allocate;

		//Modified Version.
		//integrated with FreeRTOS instead of using custom_memory_manager.c
		freeRTOS_allocator.allocate = __freertos_allocate;

		freeRTOS_allocator.deallocate = __freertos_deallocate;

		freeRTOS_allocator.reallocate = __freertos_reallocate;

		freeRTOS_allocator.zero_allocate = __freertos_zero_allocate;

		if(!rcutils_set_default_allocator(&freeRTOS_allocator)) {
			printf("Error on default allocators (line %d)\n", __LINE__);
			for(;;);
		}
    };

	allocator = rcl_get_default_allocator();

	osDelay(1000);

	//support_init:
    {
        support_init:

		rcl_ret_t ret;

		//connect to the micro-ros agent...
		ret = rclc_support_init(&support, 0, NULL, &allocator);

	    //if ROS-agent is not ready,
	    //program will stuck here, trying again and again...
        if(ret != RCL_RET_OK)
        {
            osDelay(1000);

            printf("Error on support init (line %d). Passing, Trying Again...\r\n", __LINE__);

            goto support_init;
        }
    };

	osDelay(1000);

	// create node
    {
		rcl_ret_t ret;

		ret = rclc_node_init_default(
				&node,
				"alphabot_micro_ros_node",
				"/",
				&support);

        if(ret != RCL_RET_OK)
        {
            printf("Error on create node (line %d). Failed, Reseting System...\r\n", __LINE__);

            //communication error!
            HAL_NVIC_SystemReset();
        }
    };