micro-ROS/micro_ros_setup #653: Multiple micro-ROS Publishers and multiple micro-ROS subscribers are not working properly with UDP(WIFI) Transport.

SiddharthZokarkar, 2023-07-11:
Hardware description: ESP32
    RTOS: FreeRTOS
    Installation type: micro_ros_setup
    Version - micro-ROS running on ROS2 Foxy

Hello @pablogs9 ,

I tested the behaviour of UDP (WIFI) transport by creating 1 micro-ROS publisher (producing 1 topic:  /publisher1) on one ESP32 and 20 micro-ROS subscribers on another ESP32 all are subscribing to the same TOPIC (/publisher1) .
The micro-ROS agent has created 20 micro-ROS subscribers but around 15 micro-ROS subscribers are working. 

The same code is referred for creating the publisher
https://github.com/micro-ROS/freertos_apps/blob/iron/apps/int32_publisher/app.c

The same code is referred for creating the subscribers
https://github.com/micro-ROS/freertos_apps/blob/iron/apps/int32_subscriber/app.c

Is there any way so that all the subscribers can work?

Kindly give your guidance.

Thank you so much!!

Acuadros95, 2023-07-11:
Hi @SiddharthZokarkar ,

You probably want to remove the `usleep(100000);` on the subscriber side or reduce its sleep time, as the executor will need to dispatch all the received messages.

You can also increase the `RMW_UXRCE_MAX_HISTORY` parameter, so you can store all the received messages before they are processed. More details here: [Link](https://docs.vulcanexus.org/en/latest/rst/tutorials/micro/memory_management/memory_management.html#middleware-related-memory).

Another option is to use best effort communication. Some messages may be dropped, but the latency will improve.

Acuadros95, 2023-07-31:
Closing due to inactivity, reopen if needed

closed 2023-07-31 by Acuadros95