micro-ROS/micro_ros_setup #659: Using micro-ROS with rclcpp

AlexandreGC35, 2023-08-30:
## Issue template

- Hardware description: Ubuntu 22.04 machine, compiling to flash on an Olimex_stm32_e407 board.
- RTOS: Zephyr
- Installation type: Follow the First micro-ROS Application on Zephyr guide from micro-ROS (link below), installing micro-ROS inside a docker container and creating the zephyr - olimex_stm32_e407 firmware with micro_ros_setup

https://micro.ros.org/docs/tutorials/core/first_application_rtos/zephyr/

- Version or commit hash: foxy version of micro_ros_setup

#### Steps to reproduce the issue
Hi, I want to use micro-ROS for building an application in c++, and needing rclcpp to compile. I have already figured out how to build and flash a C app using rcl/rclc, which I saw on the site are supported, but I can´t manage to do it with rclcpp. 
From what I read on the site and the existing git issues, rclcpp is not yet supported officially by micro-ROS but is still a WIP, so I understand why it´s more complicated, but I also saw that several attempts have been made at implementing this, so I assume some solution/workaround is possible atm.

My understanding of the situation is that, after creating the firmware described above and configuring  it for my application (made by copying a sample existing in the *firmware/zephyr_apps/apps/folder*  , where all examples are), I should modify the CMakeLists.txt file and/or the .meta file in my application folder to make it include this rclcpp package, as it is present by default in the ros2 installation in */opt/ros/foxy/include/* folder. But, as I am a fairly new user to this, I couldn´t find the solution as to how to do it.

What steps should I follow ? Is my understanding of the situation right ? Should I look elsewhere to include this package ? Any help is appreciated, thank you in advance :)

#### Expected behavior
The code of my micro-ROS application in c++ should compile entirely. It is for now just a simple publisher code in c++, just like the one described in this tutorial : 

https://docs.ros.org/en/foxy/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Cpp-Publisher-And-Subscriber.html

#### Actual behavior
It doesn´t build as it cannot find the "rclcpp/rclcpp.hpp" file because it is not included by default. I´m not including screenshots or logs because they are simple "cannot find rclcpp/rclcpp.hpp" reports, so unless I missed some critical info not much should come from it-

#### Additional information
As I am a fairly new user to micro-ROS and git, there may be some mistakes or elements missing in what I said. I am open to any request for more details, so don´t hesitate.

pablogs9, 2023-09-04:
Hello as you mention RCLCPP is not supported in micro-ROS nor is in our roadmap, the discussion of a C++ API is here: https://github.com/ros2/rclc/issues/126

BTW, if you are using Zephyr, the recommended option is: https://github.com/micro-ROS/micro_ros_zephyr_module

AlexandreGC35, 2023-09-04:
Hello, first of all thank you for your quick response and feedback!

As I searched quite a bit for a solution, I had already seen the C++ API discussion you linked. I understand now that such APi is not in your roadmap, and a solution should not be expected soon. 

I still have one question though, linked with the micro_ros_demos repository ( https://github.com/micro-ROS/micro-ROS-demos/tree/foxy ). In the ROS2 branches of this repo before galactic (including foxy, the one i´m working in) , there exists a RCLCPP/ folder with simple examples including RCLCPP, in micro-ROS. And the compilation of these examples doesn´t raise any errors linked to incompatibility with RCLCPP (lots of warnings though, which is expected), not to mention they can executed after compilation succesfully.
I you need any precisions or instructions to reproduce what I did with this repo, don´t hesitate to ask!

I´m just wondering what made this compilation with RCLCPP successful, and how was it possible in the first place with micro-ROS if RCLCPP is not supported? If possible I would like to reproduce this in my project.

Also thank you very much regarding the zephyr module. I didn´t use it because I thought it was still in modifcation to be made a pure module (I´ve also installed it before), but if this is the recommanded option I will try to migrate my project.