micro-ROS/micro_ros_setup #660: unable to enter the callback function when publisher's frequency increases or the receiver's data volume becomes larger, and the trigger becomes invalid.

Junming-Liang, 2023-09-03:
## Issue template

- Hardware description: STM32E407 development board
- RTOS: FreeRTOS
- Installation type: micro_ros_setup
- Version or commit hash: foxy

#### Steps to reproduce the issue

1. Add multiple subscribers and publishers in the system.
2. Use high frequency msg
3. Too long time in callbcak

#### Expected behavior
All subscribers should be able to enter their callback functions when there are publishers in the system.

#### Actual behavior
When using only subscribers, everything works fine. However, when adding publishers to the system, some subscribers cannot enter their callback functions. (This problem also occurs when I don't set up a publisher and receive topic messages at a high frequency)The following solutions have been attempted but did not fully resolve the issue:
1. Reduce the size of publishers and messages. With this approach, some improvement was observed, but still, not all messages could be received. Of the three subscribers set up, only one or two would be able to enter their callback functions.
4. Change the publisher to "best effort" mode. However, when the publisher is adjusted to "best effort" mode, the topic cannot be received in ROS.
5. Use Trigger. When Trigger is set to accept all messages or to enter the callback only when a specific message is received, none of the subscriber's callback functions can be entered.

#### Additional information
This is my source code app.c:
```c
#include <rcl/error_handling.h>
#include <rcl/rcl.h>
#include <rclc/executor.h>
#include <rclc/rclc.h>
// #include
// </home/liangjunming/program/mcu_ws/firmware/mcu_ws/install/std_msgs/include/std_msgs/msg/detail/header__struct.h>
#include "devastator_control_msgs/msg/vehicle_control_command.h"
#include "devastator_control_msgs/msg/vehicle_control_command_light.h"
#include "devastator_control_msgs/msg/vehicle_report_common.h"
#include "devastator_control_msgs/msg/vehicle_report_common_light.h"
#include "devastator_localization_msgs/msg/localization_estimate.h"
#include "devastator_localization_msgs/msg/localization_estimate_light.h"

#include "devastator_planning_msgs/msg/path_point.h"
// #include "devastator_planning_msgs/msg/target_trajectory.h"
#include "devastator_planning_msgs/msg/trajectory_point.h"
#include "mcu_planning/msg/planning_trajectory.h"
#include "mcu_planning/msg/trajectory_point.h"
#include <std_msgs/msg/header.h>
#include <stdio.h>
#include <time.h>
#include <unistd.h>
#ifdef ESP_PLATFORM
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#endif
#include "control_nodes/common/interfaces_c.h"
#include "control_nodes/controller_agent_c.h"
#include "control_nodes/controllers/lat_pid_controller_c.h"
#include "control_nodes_proto_c/control_conf.pb-c.h"
#include <math.h>
#define malloc(size) pvPortMalloc(size)
#define free(ptr) vPortFree(ptr)

#define STRING_BUFFER_LEN 6
#define MAX_trajectoryPOINTS 30
#define RCCHECK(fn)                                                            \
  {                                                                            \
    rcl_ret_t temp_rc = fn;                                                    \
    if ((temp_rc != RCL_RET_OK)) {                                             \
      printf("Failed status on line %d: %d. Aborting.\n", __LINE__,            \
             (int)temp_rc);                                                    \
      vTaskDelete(NULL);                                                       \
    }                                                                          \
  }
[... 383 lines cut ...]
[1693749390.291601] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0A173F68, len: 13
[1693749390.382624] debug    | DataReader.cpp     | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000000, len: 112 
[1693749390.382705] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x0A173F68, len: 124
[1693749390.383142] debug    | DataReader.cpp     | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000001, len: 56                                          
[1693749390.383184] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x0A173F68, len: 68
[1693749390.387651] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0A173F68, len: 13
[1693749390.389762] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0A173F68, len: 13
[1693749390.483427] debug    | DataReader.cpp     | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000000, len: 112 
[1693749390.483512] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x0A173F68, len: 124                                      
[1693749390.484381] debug    | DataReader.cpp     | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000001, len: 56  
[1693749390.484429] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x0A173F68, len: 68
[1693749390.488506] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0A173F68, len: 13
[1693749390.490671] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0A173F68, len: 13 
[1693749390.583185] debug    | DataReader.cpp     | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000000, len: 112                                    
[1693749390.583281] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x0A173F68, len: 124
[1693749390.583876] debug    | DataReader.cpp     | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000001, len: 56  
[1693749390.583928] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x0A173F68, len: 68
[1693749390.588221] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0A173F68, len: 13
[1693749390.590424] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0A173F68, len: 13                                        
[1693749390.683125] debug    | DataReader.cpp     | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000000, len: 112 
[1693749390.683206] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x0A173F68, len: 124
[1693749390.683721] debug    | DataReader.cpp     | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000001, len: 56  
                                                                                                                                               
```
Three publisher working but no one enter callback,Looking forward to your reply very much

Junming-Liang, 2023-09-04:
I found the reason why the published best effort topics could not be received. The solution is the same as for this issue：https://github.com/micro-ROS/micro_ros_setup/issues/236. It's because when I subscribed to the corresponding topics in ROS, I did not set the QOS to best effort.However, I still have not found a solution for the issue with the Trigger.

pablogs9, 2023-09-04:
Can we close this as solved?

Junming-Liang, 2023-09-04:
> Can we close this as solved?
Thank you very much for your reply. After setting the publisher correctly, my issue was actually resolved. Now I can subscribe and publish related topics normally. However, my question about the Trigger remains unresolved. After I set up the Trigger, there is still the issue that none of the topics can enter the callback function, whether using rclc_executor_trigger_one or rclc_executor_trigger_all. This is very frustrating for me.

pablogs9, 2023-09-04:
@JanStaschulat could you take a look at this issue about triggering the executor in RCLC?