micro-ROS/micro_ros_setup #663: add init_options for domain_id , the code can't reconnect again !

ncnynl, 2023-09-18:
## Issue template

- Hardware description: stm43f407 
- RTOS:  freertos 
- Installation type:   micro_ros_setup docker 
- Version or commit hash: galatic 

#### Steps to reproduce the issue

this is good code , not init_option 

```
// create init_options
RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

// create node
const char * node_name = "ailibot2";
RCCHECK(rclc_node_init_default(&node, node_name, "", &support));
```

this is bad code  with init_options .  it will cause can't reconnect again . 

```
init_options = rcl_get_zero_initialized_init_options();

RCCHECK(rcl_init_options_init(&init_options, allocator));

size_t domain_id = 1;
RCCHECK(rcl_init_options_set_domain_id(&init_options, domain_id));

rmw_init_options_t* rmw_options = rcl_init_options_get_rmw_init_options(&init_options);

// create init_options
RCCHECK(rclc_support_init_with_options(&support, 0, NULL, &init_options, &allocator));

// create init_options
// RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

// create node
const char * node_name = "ailibot2";
RCCHECK(rclc_node_init_default(&node, node_name, "", &support));
```

#### Expected behavior

When I disconnect the micro-ros-agent, it can automatically connect when I use the micro-ros-agent again.

#### Actual behavior

Good code can achieve reconnection, but bad code cannot reconnect. However, for the actual data output by dubug, the development board is connected to the agent, but cannot output data.

pablogs9, 2023-09-18:
Hello, Galactic is not a supported version of micro-ROS, could you confirm that this is replicable in humble or iron?

ncnynl, 2023-09-18:
Currently we do not have a humble environment for testing.  Maybe we will also build a humble environment later.

pablogs9, 2023-09-18:
Could you share the micro-ROS Agent log with the flag `-v6`?

ncnynl, 2023-09-18:
```

docker run -it --net=host -v /dev:/dev --privileged microros/micro-ros-agent:galactic serial --dev /dev/ailibot -v6

[1695030014.669471] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1695030014.669762] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
```

Only these contents, when I disconnect and reconnect

ncnynl, 2023-09-18:
My debug infomation show like ： 
```
[17:44:04:766] APP  State is AGENT_CONNECTED Continue!␍␊
[17:44:04:780] APP  State is AGENT_CONNECTED!␍␊
[17:44:04:782] APP  State is AGENT_CONNECTED Continue!␍␊
[17:44:04:795] APP  State is AGENT_CONNECTED!␍␊
[17:44:04:798] APP  State is AGENT_CONNECTED Continue!␍␊
[17:44:04:811] APP  State is AGENT_CONNECTED!␍␊
[17:44:04:815] APP  State is AGENT_CONNECTED Continue!␍␊
[17:44:04:827] APP  State is AGENT_CONNECTED!␍␊
[17:44:04:830] APP  State is AGENT_CONNECTED Continue!␍␊
[17:44:04:843] APP  State is AGENT_CONNECTED!␍␊
```

this is a normal information ， if i disconnected ， it will be ： 

```
[17:45:42:614] APP  State is WAITING_AGENT!␍␊
[17:45:42:882] APP  State is WAITING_AGENT!␍␊
[17:45:42:885] APP  State is WAITING_AGENT!␍␊
[17:45:42:893] APP  State is WAITING_AGENT!␍␊
[17:45:42:893] APP  State is WAITING_AGENT!␍␊
[17:45:42:901] APP  State is WAITING_AGENT!␍␊
[17:45:42:901] APP  State is WAITING_AGENT!␍␊
[17:45:42:901] APP  State is WAITING_AGENT!␍␊
[17:45:42:914] APP  State is WAITING_AGENT!␍␊
[17:45:42:914] APP  State is WAITING_AGENT!␍␊
[17:45:42:914] APP  State is WAITING_AGENT!␍␊
[17:45:42:914] APP  State is WAITING_AGENT!␍␊
```

pablogs9, 2023-09-18:
Does it work if you use `rclc_support_init` instead of `rclc_support_init_with_options` ?

ncnynl, 2023-09-18:
Yes, reconnect is correct if I use rclc_support_init

pablogs9, 2023-09-18:
I would recommend using a newer version of micro-ROS in order to ensure that this is not a old resolved issue

ncnynl, 2023-09-18:
Okay, I will set up a humble environment to test it.

ncnynl, 2023-09-20:
@pablogs9  I tested the code on humble today and found that this problem still exists, but this one can reconnect one more time than the galactic version. I replugged the USB, then the micro-ros-agent connected, disconnected, reconnected, disconnected again, and then couldn't connect again. The feedback is the same as galactic's problem.

ncnynl, 2023-09-20:
@pablogs9  On Galactic, there is no limit on reconnection. It is normal for me to disconnect and reconnect more than 5 or 6 times. Judging from the content printed by debug on the code, the code on the MCU is running normally, but the agent cannot obtain it normally. Unable to reconnect, all that appear are stuck in the connected state:

```
docker run -it --net=host -v /dev:/dev --privileged microros/micro-ros-agent:humble serial --dev /dev/ailibot -v6

[1695030014.669471] info | TermiosAgentLinux.cpp | init | running... | fd: 3
[1695030014.669762] info | Root.cpp | set_verbose_level | logger setup | verbose_level: 6
```

ncnynl, 2023-09-20:
@pablogs9  I may have found the answer to the problem. Without the rcl_init_options_fini(&init_options); line of code, after adding this, the current behavior on humble is normal. Can successfully reconnect multiple times in a row

pablogs9, 2023-09-20:
Are you taking into account to set `RMW_UXRCE_ENTITY_CREATION_DESTROY_TIMEOUT` to 0 and `UCLIENT_MAX_SESSION_CONNECTION_ATTEMPTS` to a low value as stated here: 

https://github.com/micro-ROS/micro_ros_arduino/blob/6f004dc1c88b3a0cdfa86b60b52bf0c335b69dc0/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino#L44-L48

ncnynl, 2023-09-20:
I didn't set these two values. I don't know where I can set them :( After I added rcl_init_options_fini, it became normal. The galactic and humble codes are all normal.

pablogs9, 2023-09-20:
So can we close as solved?

closed 2023-09-20 by ncnynl