micro-ROS/micro_ros_setup #664: Problems with include path of libraries in microros

N-K1998, 2023-09-18:
## Issue template

- Hardware description: STM
- RTOS: freeRTOS
- Installation type: microros_setup
- Version or commit hash: iron

#### Steps to reproduce the issue
I have compiled Microros as a static lib and added it using the compile flags. I have also added the folder libmicroros/include as an include path. However, the code does not find any of the headers because the folders are organized in the following way:
library_name/library_name/header_file.h. That's because the includes are /library_name/headear_file.h. In other words, I am forced to add each folder path as an include path. Is there a way to make it simpler? In the previous version of Microros, the folder organization was /library_name/headear_file.h, and the new version's code still follows that standard.

pablogs9, 2023-09-18:
In other packages, we have script lines that move the folder to the logical location: 

https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/3819f6819cee1ff01c3da58ecbc8bfcbf4b0a2a5/microros_static_library/library_generation/library_generation.sh#L77-L87

Please take a look.

It would be great if you could contribute back to this repo the solution.

N-K1998, 2023-09-18:
Basically, is it sufficient to take the folders and put them back into the main include one?

pablogs9, 2023-09-18:
That should be enought

N-K1998, 2023-09-18:
Ok, I try and then I report.

NikoSalamini, 2023-09-19:
This is the script that i have used:
```
#!/bin/bash
INCLUDE_MICROROS_PACKAGES=$(ls YOUR_UROS_WS/firmware/build/include)
for var in ${INCLUDE_MICROROS_PACKAGES}; do
        if [ -d "YOUR_UROS_WS/firmware/build/include/${var}/${var}" ]; then
                rsync -r YOUR_UROS_WS/firmware/build/include/${var}/${var}/* YOUR_UROS_WS/firmware/build/include/${var}
                rm -rf YOUR_UROS_WS/firmware/build/include/${var}/${var}
        fi
done
```