micro-ROS/micro_ros_setup #666: ROS2 nodes do not see microros custom agent

N-K1998, 2023-09-26:
## Issue template

- Hardware description: STM
- Linux: ubuntu 22.04
- Installation type: micro_ros_agent
- Version or commit hash: iron

I have defined a custom transport (custom agent for Microros and I want my ros2 nodes to use that transport on Linux. I am executing the command `ros2 run micro_ros_agent micro_ros_agent` to run the custom agent and `ros2 run cpp_pubsub listener` of the cpp_pubsub example provided by ROS2. The CMake of the ROS2 application is the following:
```
cmake_minimum_required(VERSION 3.8)
project(cpp_pubsub)

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

# find dependencies
find_package(ament_cmake REQUIRED)
find_package(rclcpp REQUIRED)
find_package(std_msgs REQUIRED)
# uncomment the following section in order to fill in
# further dependencies manually.
# find_package(<dependency> REQUIRED)

add_executable(talker src/publisher_member_function.cpp)
add_executable(listener src/subscriber_member_function.cpp)
ament_target_dependencies(talker rclcpp std_msgs)
ament_target_dependencies(listener rclcpp std_msgs)

install(TARGETS
          talker
          listener
            DESTINATION lib/${PROJECT_NAME})

if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  # the following line skips the linter which checks for copyrights
  # comment the line when a copyright and license is added to all source files
  set(ament_cmake_copyright_FOUND TRUE)
  # the following line skips cpplint (only works in a git repo)
  # comment the line when this package is in a git repo and when
  # a copyright and license is added to all source files
  set(ament_cmake_cpplint_FOUND TRUE)
  ament_lint_auto_find_test_dependencies()
endif()

ament_package()
```
This file is placed inside microros_ws/src/uros/cpp_pubsub. Inside the uros folder, there is also the micro-ROS-agent folder containing the files for the agent. How can I force the ros2 nodes to use the custom transport?

pablogs9, 2023-09-26:
Hello @N-K1998 I do not fully understand your issue. Please answer these questions:

- Have you implemented successfully a custom transport on micro-ROS Client side? If yes, provide details.
- Have you implemented successfully a custom transport on micro-ROS Agent side? If yes, provide details.
- Are your micro-ROS Client and Agent communicating? If yes, provide the micro-ROS Agent logs with the flag `-v6`

N-K1998, 2023-09-26:
I am sorry, I cannot provide you with the details of the custom transport but it works. The custom agent can communicate with microros clients. The question is if it is possible for an agent to be able to communicate with a normal ROS2 node, not a microros client, through the DDS.

pablogs9, 2023-09-26:
If you need to implement custom transport for DDS, you need to ask in [Fast DDS repo](https://github.com/eProsima/Fast-DDS) as far as it is not related to micro-ROS or Micro XRCE-DDS.

If the micro-ROS part is working successfully, let me know if we can close this issue.

N-K1998, 2023-09-26:
Yes, the microros part works correctly but the XRCE-DDS cannot communicate with the standard DDS global data space. Is there a possibility that the agent is running on a different DDS than the one belonging to ROS2 nodes?

pablogs9, 2023-09-26:
Well, this log shows that DDS communication is happening, check the `[** <<DDS>> **]`.

Are your ROS2 node using `ROS_DOMAIN_ID=0`?

pablogs9, 2023-09-26:
As far as I understand this is a DDS issue between the agent and your ROS 2 node. 

Are both entities running on the same computer?
Are you using dockers somehow?
Is possible to share a Wireshark captura of the RTPS traffic on the network?

N-K1998, 2023-09-26:
The problem was on the topic the listener was subscribed to. Thank you for your support.

closed 2023-09-26 by pablogs9