micro-ROS/micro_ros_setup #668: How to get microros thread support for freeRTOS

N-K1998, 2023-09-30:
## Issue template

- Hardware description: generic
- RTOS: freeRTOS
- Installation type: static lib
- Version or commit hash: iron

I have generated firmware by using "ros2 run micro_ros_setup create_firmware_ws.sh generate_lib". Then I enabled the the option by using "Enable multithread support ON" in the CMakeLists.txt file. When generating the static lib I get the "#error XRCE multithreading not supported for this platform".

Is there a way to get microros thread support for platform generic running on freeRTOS?

pablogs9, 2023-10-02:
You need to configure in your toolchain the include path for FreeRTOS (required for [this](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/97175304425c5bee87c6fddd99de1ef8d0c394dc/include/uxr/client/profile/multithread/multithread.h#L33-L34)), also you need to define `PLATFORM_NAME_FREERTOS`

Sorokonog, 2023-11-28:
Sorry to interfere but let me add some hints for generic library generation if you have STM32 Cube IDE build.
1. For defining `PLATFORM_NAME_FREERTOS` you can add `-DPLATFORM_NAME_FREERTOS` into section of
# C defines of your main project Makefile
so it will looks like:
# C defines
```
C_DEFS =  \
-DUSE_HAL_DRIVER \
-DSTM32G474xx \
-DSTM32_THREAD_SAFE_STRATEGY=4 \
-DPLATFORM_NAME_FREERTOS
```
2. In terms of toolchain configuration you should add
```
include_directories(/project/Middlewares/Third_Party/FreeRTOS/Source/include)
include_directories(/project/Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F)
include_directories(/project/Core/inc) 
```
into /{YOUR PROJECT NAME}/micro_ros_stm32cubemx_utils/microros_static_library/library_generation/extra_packages/toolchain.cmake

Hope it will help.

pablogs9, 2023-11-28:
Would be nice to have this in the README.md @Sorokonog

Johannes-42, 2024-01-30:
I have a question related to this post:

I am new to ROS2, microROS and freeRTOS and I have the following **Hardware**:
- Olimex STM32 - E407 (With j-tag, USB and serial connection) (for freeRTOS and Zephyr)
- a Laptop to run ROS2 with Ubuntu 22.04 and i use Humble for ROS2.

**What I want**: I want a microRos application that drives my Robot. And I need a way to stop everything when an emergency happens. Therefore I think multithreading is the way to go. I think of one thread with higher priority that checks the whole time if the subscriber has new data. And if the stop command comes (I publish the stop command on the computer) it will change a shared memory value and all other threads will check for that value and stop the robot and everything he is doing until the value is flipped back. I do this to guarantee that my robot will be fast enough.

I found the following Examples for multithreading: [rclc multithreading example](https://github.com/micro-ROS/micro-ROS-demos/blob/feature/multithread/rclc/multithread_publisher_subscriber/main.c)
Here pthread is used. _Can i used pthread in my freeRTOS?_
 I can compile with pthread but it is not working.

I also found this Example:
[multithreading on ESP](https://github.com/micro-ROS/micro_ros_espidf_component/blob/iron/examples/multithread_publisher/main/main.c)

The problem I have is that when I set up microROS like in the tutorial from the Webpage  [microROS webpage](https://micro.ros.org/docs/tutorials/core/first_application_rtos/freertos/) I do not have cmake files in the app folder.
Therefore I do not understand where to put the "include_directories" or "-DPLATFORM_NAME_FREERTOS" mentioned above.
I added the "-DUCLIENT_PROFILE_MULTITHREAD=ON" to my colcon.meta file but I am not sure if that makes anything. 
If I go to the Micro-XRCE-DDS-Client folder and change the value of UCLIENT_PROFILE_MULTITHREAD there to ON I see when compiling that I have the same Error as the person who started this question (#error XRCE multithreading not supported for this platform).

I made a github folder to solve this problem: https://github.com/Johannes-42/microROS-freeRTOS/tree/main
My first step would be to just get it running with multiple threads.
The idea would be to just copy it to the firmware/freertos_apps/apps/ folder and run it.
Could I please get some help here?

Johannes-42, 2024-02-08:
I updated my repository. Now the threads will start and will publish. But the timer will never call the callback. And if the timer is not used the threads will not publish.

Is there a way to see the output of the Olimex board but do not lose the connection to the micro_ros_agent? I saw that there are ways to get the output over USB but then that would interfere with the micro_ros_agent.