micro-ROS/micro_ros_setup #671: Build int32_publisher Firmware Error (ESP32-S3 on YD-ESP32-S3-N16R8 Board)

AlfonGio, 2023-10-20:
Hi, I am learning micro-ROS with ESP32 board by following the basic tutorial from this blog post [Connect ESP32 to ROS2 (Foxy +)](https://medium.com/@SameerT009/connect-esp32-to-ros2-foxy-5f06e0cc64df) which is quite similar as the original tutorial written in micro-ROS webpage.

While doing the build_firmware.sh step, I got the following error:

- Hardware description: 
   - Board: YD-ESP32-S3-N16R8 by VCC-GND
   - Host: Ubuntu 22.04 with ROS2 Humble
   - Interface: Serial USB
- RTOS: FreeRTOS
- Installation type: micro-ROS example project int32_publisher
- Version or commit hash: micro-ROS humble branch

#### Steps to reproduce the issue
1. Create firmware `ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32`
2. Configure the firmware 
    `ros2 run micro_ros_setup configure_firmware.sh int32_publisher -t udp -i 192.xxx.xxx.xxx -p xxxx`
3. Configure WIFI connection `ros2 run micro_ros_setup build_firmware.sh menuconfig`
4. Build the firmware `ros2 run micro_ros_setup build_firmware.sh`

#### Expected behavior
[image]

#### Actual behavior
[image]

There is an error which I am not sure. I have tried to search the internet and found someone with the same error but has not been solved yet.

#### Additional information
I am connecting the board via USB cable through host machine.

pablogs9, 2023-10-24:
Hello, the way to go in ESP32-based boards is to use micro-ROS IDF component. You can find it here: https://github.com/micro-ROS/micro_ros_espidf_component

pablogs9, 2023-10-24:
[micro-ROS component for ESP-IDF](https://github.com/micro-ROS/micro_ros_espidf_component#micro-ros-component-for-esp-idf) is a different approach, you do not need micro_ros_setup at all.

Just create an environment of ESP-IDF using their tooling: https://docs.espressif.com/projects/esp-idf/en/release-v5.0/esp32/get-started/linux-macos-setup.html#get-started-linux-macos-first-steps

And then add the micro-ROS module as explained in the [repo](https://github.com/micro-ROS/micro_ros_espidf_component#micro-ros-component-for-esp-idf).

AlfonGio, 2023-10-24:
Ah I get it now. Thank you..