micro-ROS/micro_ros_setup #673: can't build firmware after putting custom message

AhmadSameh, 2023-10-27:
## Issue template

- Hardware description: esp32
- RTOS: FreeRTOS
- Installation type:  micro_ros_setup,
- Version or commit hash: humble

#### Steps to reproduce the issue
1- followed the custom message tutorial on the micro-ros documentation
2- built the package using the command
`ros2 run micro_ros_setup build_firmware.sh`

#### Expected behavior
should compile just fine

#### Actual behavior
this error shows up:
```
Crosscompiled environment: cleaning path
Building firmware for freertos platform esp32
WARNING: using an unsupported version of tool cmake found in PATH: 3.27.6
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
/home/boi/work_stuff/ros2_course/microros_ws/firmware/toolchain/esp-idf/tools/check_python_dependencies.py:22: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Python requirements from /home/boi/work_stuff/ros2_course/microros_ws/firmware/toolchain/esp-idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /home/boi/work_stuff/ros2_course/microros_ws/firmware/toolchain/esp-idf/components/esptool_py/esptool
  /home/boi/work_stuff/ros2_course/microros_ws/firmware/toolchain/esp-idf/components/espcoredump
  /home/boi/work_stuff/ros2_course/microros_ws/firmware/toolchain/esp-idf/components/partition_table/
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

head: cannot open '/home/boi/work_stuff/ros2_course/microros_ws/firmware/APP' for reading: No such file or directory
Selected app:
/home/boi/work_stuff/ros2_course/microros_ws/install/micro_ros_setup/config/freertos/esp32/build.sh: line 23: pushd: /home/boi/work_stuff/ros2_course/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build: No such file or directory
[ros2run]: Process exited with failure 1

```
#### Additional information

pablogs9, 2023-10-30:
Does it works without using custom messages?

In anyn case, the recommended option for using ESP32 is using [micro-ROS component for ESP-IDF](https://github.com/micro-ROS/micro_ros_espidf_component#micro-ros-component-for-esp-idf)

AhmadSameh, 2023-10-30:
@pablogs9 when I wrote this problem I had expected that on a successful building, I would have a header file and then I would add it to the Arduino Library, but that doesn't seem to be the case.
So I apologize for the sudden change of questions, but how can I add a custom message to the micro-ros Arduino library.

pablogs9, 2023-10-30:
If you use the mentioned component, just add your packages here: https://github.com/micro-ROS/micro_ros_espidf_component/tree/iron/extra_packages