micro-ROS/micro_ros_setup #674: build_agent.sh failed for ROS2 Humble on RPi

jidnyesha-patil, 2023-11-02:
- Hardware description: Raspberri Pi 4
- RTOS: Ubuntu 22.04
- Installation type: micro_ros_setup
- Version or commit hash: humble

#### Steps to reproduce the issue
In ws, clone humble branch of micro-ros-setup
`git clone https://github.com/micro-ROS/micro_ros_setup.git`
Source the setup
`source /opt/ros/humble/setup.bash`
Update system
`rosdep update && rosdep install --from-paths src --ignore-src -y`
Build package
`colcon build`
Create agent 
`ros2 run micro_ros_setup create_agent_ws.sh`
Build agent
`ros2 run micro_ros_setup build_agent.sh`

#### Expected behavior
Build should complete without issues

#### Actual behavior

```
Building micro-ROS Agent
Starting >>> micro_ros_msgs
Finished <<< micro_ros_msgs [9.95s]                     
Starting >>> micro_ros_agent
--- stderr: micro_ros_agent                               
/home/rushi/SLAM_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_manager.cpp: In member function ‘void uros::agent::graph_manager::GraphManager::add_datawriter(const eprosima::fastrtps::rtps::GUID_t&, const string&, const string&, const eprosima::fastrtps::rtps::GUID_t&, const eprosima::fastdds::dds::DataWriterQos&)’:
/home/rushi/SLAM_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_manager.cpp:378:9: error: ‘rosidl_get_zero_initialized_type_hash’ was not declared in this scope; did you mean ‘rcutils_get_zero_initialized_hash_map’?
  378 |         rosidl_get_zero_initialized_type_hash(), participant_gid, qos_profile, false);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         rcutils_get_zero_initialized_hash_map
/home/rushi/SLAM_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_manager.cpp: In member function ‘void uros::agent::graph_manager::GraphManager::add_datareader(const eprosima::fastrtps::rtps::GUID_t&, const string&, const string&, const eprosima::fastrtps::rtps::GUID_t&, const eprosima::fastdds::dds::DataReaderQos&)’:
/home/rushi/SLAM_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_manager.cpp:420:9: error: ‘rosidl_get_zero_initialized_type_hash’ was not declared in this scope; did you mean ‘rcutils_get_zero_initialized_hash_map’?
  420 |         rosidl_get_zero_initialized_type_hash(), participant_gid, qos_profile, true);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         rcutils_get_zero_initialized_hash_map
gmake[2]: *** [CMakeFiles/micro_ros_agent.dir/build.make:104: CMakeFiles/micro_ros_agent.dir/src/agent/graph_manager/graph_manager.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/micro_ros_agent.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< micro_ros_agent [12.4s, exited with code 2]

Summary: 1 package finished [23.3s]
  1 package failed: micro_ros_agent
  1 package had stderr output: micro_ros_agent
[ros2run]: Process exited with failure 2
```

#### Additional information

A previous issue with similar description was closed without resolution due to inactivity.

pablogs9, 2023-11-02:
You are not cloning humble branch with `git clone https://github.com/micro-ROS/micro_ros_setup.git`, you are cloning `iron`..

Could you report back if checking out the correct branch it builds?

jidnyesha-patil, 2023-11-05:
Hello @pablogs9, it worked with the humble branch. Thank you for the response. 
No issues with the build now.

closed 2023-11-05 by jidnyesha-patil